arendst/Tasmota #8614: Windmeter boot issue

lacrimosa2k2, 2020-06-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
i have a pulse signal (windmeter) on gpio4 or gpio5 (i testet it on all avaible pins) and and the windmeter turns and comes pulse in, and then i restart the esp, it could not boot complete, it reboots so often that its looses the config.
when i dissconect the pulse cable from the esp and reboot and after boot complete add it, it runns fine.
the pulse ar not the error, when i disable (select none from the gpio) the esp reboots without problems and pulses are connected

more info from other user here
https://github.com/arendst/Tasmota/pull/8283

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: _____8.3
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [x ] IDE / Compiler used: _____ platform.io
- [x ] Flashing tools used: _____tasmotizer
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
damn i forgot to save it

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
15:00:22 APP: Restarting

ets Jan 8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 3456, room 16
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
00:00:00 CFG: Loaded from flash at F9, Count 51
00:00:00 Project tasmota Tasmota2 Version 8.3.1(tasmota)-2_7_1
00:00:00 WIF: Connecti 31o AP1 ***** Channel 11 BSSId ::::: in mode 11N as Tasmota2...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on Tasmota2 with IP address 192.168.1.37
15:00:28 RSL: INFO1 = {"Module":"Generic","Version":"8.3.1(tasmota)","FallbackTopic":"cmnd/DVES_BC58D4_fb/","GroupTopic":"tasmotas/cmnd/"}
15:00:28 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"Tasmota2","IPAddress":"192.168.1.37"}
15:00:28 RSL: INFO3 = {"RestartReason":"Software/System restart"}

Exception (0):
epc1=0x4021abec epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

'>>>stack>>>
`

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

malbinola, 2020-06-03:
Unfortunately i was not able to test it on GPIO4 or 5 for now, i do not have a free device to use for experiments. Windmeter is running fine on my Sonoff 4CH PRO R2 using GPIO3 (rx - with serial log disabled) as input pin with a 4k7 pull-up resistor. I've also tested GPIO2 with success but with some boot problems if reed switch is closed (low) on start, but this is due to ESP design.

Starting from what you said ("with the counter function it runs fine") I've compared the init process of counter sensor against windmeter. I think that there are two differences:
1. [PinMode](https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_68_windmeter.ino#L118)
In counter "no_pullup" flag controls whether to enable internal pull-up or not. Implementation in windmeter sets INPUT_PULLUP instead. Maybe this can be an issue with some GPIOs?

2. [AttachInterrupt](https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_68_windmeter.ino#L119)
New improved counter debouncing supports CHANGE trigger. I do not think this can be a problem. Maybe @to-scho can give us some advices.

lacrimosa2k2, 2020-06-03:
@malbinola when i connect it to gpio3 and restart it works when no pulses comes, when the windmeter rotate the esp crashes. i think when only one pulse per rotation and slow wind the issue doenst come up the time between two pulses ar loong, but i use a meter with 3 pulses per rotation and when a little bit more wind than crashes

device111, 2020-06-04:
@malbinola 
i have seen, that your variables are not volatile in ISR Routine. 
normaly all of your variables in ISR function should be volatile do prevent compiler for optimizing.
this could be the issue. give it a try...

arendst, 2020-06-04:
I see the windmeter interrupt routine tries to output debug information. This is a no-go. It takes too much time and will most likely result in the exceptions you encounter.

I will disable the debug message so you can try again.

device111, 2020-06-04:
yes, i have seen also. you can outsource this routine completly:
`  if (time_diff > Settings.windmeter_pulse_debounce * 1000) {
    WindMeter.counter_time = time;
    WindMeter.counter++;
    AddLog_P2(LOG_LEVEL_DEBUG, PSTR("WMET: Counter %d"), WindMeter.counter);
  }`

device111, 2020-06-04:
oh, Theo was faster...:-)

ascillato2, 2020-06-04:
Closing this issue as it has been fixed. Thanks.

closed 2020-06-04 by ascillato2