arendst/Tasmota #8617: MQTT stops functioning - web, console etc continues

ShonkyCH, 2020-06-04:
### PROBLEM DESCRIPTION
After a period of time, sometimes I think hours quickly, the MQTT connection seems lost. Device pings, web interface works fine, syslog to a remote server continues. Running "SetOption19 1" will bring it back.

Home Assistant ultimately decides device has gone away, but immediately brings it back online.

It does appear to temporarily lose the MQTT connection sometimes.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic):  Happens with multiple devices including:
Brilliant plug device looks identical to this: https://templates.blakadder.com/mirabella_genio_1002341.html
Also Genio downlights:
https://templates.blakadder.com/mirabella_genio_I002741.html
Sonof basic:
https://templates.blakadder.com/sonoff_basic.html
- [X] Tasmota binary firmware version number used: 8.3.1
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: tuya convert
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
20:37:49 CMD: Backlog Template; Module; GPIO 255
20:37:49 SRC: WebConsole from 192.168.99.102
20:37:49 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
20:37:49 SRC: Backlog
20:37:49 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
20:37:49 MQT: sonoff_dishwasher/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
20:37:50 SRC: Backlog
20:37:50 CMD: Group 0, Index 1, Command "MODULE", Data ""
20:37:50 MQT: sonoff_dishwasher/stat/RESULT = {"Module":{"18":"Generic"}}
20:37:50 SRC: Backlog
20:37:50 CMD: Group 0, Index 1, Command "GPIO", Data "255"
20:37:50 MQT: sonoff_dishwasher/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"21":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"52":"Led1"},"GPIO13":{"0":"None"},"GPIO14":{"90":"Button1n"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
20:37:59 WIF: Checking connection...
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
no rules
```
- [X] Provide the output of this command: `Status 0`:
```
20:39:24 CMD: status 0
20:39:24 SRC: WebConsole from 192.168.99.102
20:39:24 CMD: Group 0, Index 1, Command "STATUS", Data "0"
20:39:24 MQT: sonoff_dishwasher/stat/STATUS = {"Status":{"Module":18,"DeviceName":"Dishwasher Sonoff","FriendlyName":["Dishwasher Sonoff"],"Topic":"sonoff_dishwasher","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:39:24 MQT: sonoff_dishwasher/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"8T01:46:34","StartupUTC":"2020-05-27T08:52:50","Sleep":50,"CfgHolder":4617,"BootCount":51,"BCResetTime":"2020-04-09T21:25:43","SaveCount":580,"SaveAddress":"FA000"}}
20:39:24 MQT: sonoff_dishwasher/stat/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":4,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"412/699"}}
20:39:24 MQT: sonoff_dishwasher/stat/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"MqttLog":4,"SysLog":4,"LogHost":"192.168.99.6","LogPort":514,"SSId":["Shonky",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005AFF000000000000","00000200","00000000"]}}
20:39:24 MQT: sonoff_dishwasher/stat/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
20:39:24 MQT: sonoff_dishwasher/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff_dishwasher-3083","IPAddress":"192.168.99.195","Gateway":"192.168.99.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.99.10","Mac":"60:01:94:BB:2C:0B","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:39:24 MQT: sonoff_dishwasher/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.99.6","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BB2C0B","MqttUser":"DVES_USER","MqttCount":5,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:39:24 MQT: sonoff_dishwasher/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-06-04T10:39:24","Local":"2020-06-04T20:39:24","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+10:00","Sunrise":"13:49","Sunset":"05:47"}}
20:39:24 MQT: sonoff_dishwasher/stat/STATUS10 = {"StatusSNS":{"Time":"2020-06-04T20:39:24"}}
20:39:24 MQT: sonoff_dishwasher/stat/STATUS11 = {"StatusSTS":{"Time":"2020-06-04T20:39:24","Uptime":"8T01:46:34","UptimeSec":697594,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Shonky","BSSId":"FE:EC:DA:95:49:19","Channel":1,"RSSI":70,"Signal":-65,"LinkCount":3,"Downtime":"0T00:07:57"}}}
```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
syslog output
```

Jun  2 19:40:53 sonoff_dishwasher-3083 ESP-HTP: Main Menu
Jun  2 19:41:01 sonoff_dishwasher-3083 ESP-WIF: Checking connection...
Jun  2 19:41:08 sonoff_dishwasher-3083 ESP-MQT: sonoff_dishwasher/tele/HASS_STATE = {"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Module or Template":"Generic","RestartReason":"Power On","Uptime":"6T00:48:18","Hostname":"sonoff_dishwasher-3083","IPAddress":"192.168.99.195","RSSI":"72","Signal (dBm)":"-64","WiFi LinkCount":3,"WiFi Downtime":"0T00:07:57","MqttCount":4,"LoadAvg":19}
[... 207 lines cut ...]
Jun  4 16:22:19 sonoff_dishwasher-3083 ESP-WIF: Checking connection...
Jun  4 16:22:39 sonoff_dishwasher-3083 ESP-WIF: Checking connection...
Jun  4 16:22:58 sonoff_dishwasher-3083 ESP-WIF: Checking connection...
Jun  4 16:23:18 sonoff_dishwasher-3083 ESP-WIF: Checking connection...
Jun  4 16:23:35 sonoff_dishwasher-3083 ESP-MQT: sonoff_dishwasher/tele/STATE = {"Time":"2020-06-04T16:23:34","Uptime":"7T21:30:44","UptimeSec":682244,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"POWER":"ON","Wifi":{"AP":1,"SSId":"Shonky","BSSId":"FE:EC:DA:95:49:19","Channel":1,"RSSI":76,"Signal":-62,"LinkCount":3,"Downtime":"0T00:07:57"}}
Jun  4 16:23:39 sonoff_dishwasher-3083 ESP-WIF: Checking connection...
Jun  4 16:23:58 sonoff_dishwasher-3083 ESP-WIF: Checking connection...
Jun  4 16:24:19 sonoff_dishwasher-3083 ESP-WIF: Checking connection...
```

### TO REPRODUCE
Can't say for sure makes it happen but appears to happen after a few days usually

Running "SetOption19 1" will bring it back working temporarily as will of course a reboot.

### EXPECTED BEHAVIOUR
MQTT stays connected

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-06-04:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-06-04 by ascillato2

ShonkyCH, 2020-06-04:
Where is this duplicated please so I can follow there? Seems like a bug to me?

ascillato, 2020-06-04:
Just saw you found the chat. Great!

The issue is duplicated and addressed several times in old issues. If you want to dig further you can perform a search.

Summary> Issue not related to Tasmota but to the MQTT Broker Setup. That is why you were sent to the Support Chat.

mkyral, 2020-06-12:
For everyone with the same issue.
I've just upgraded Tasmota on Sonoff POW R2 from v 6.4 to 7.1, 7.2, 8.1 and latest and faced this as well. No issue with mqtt before.

After some searching I've found this: https://tasmota.github.io/docs/FAQ/#weaker-wi-fi-signal-after-upgrade and it helped.