arendst/Tasmota #8619: Telegram message notification support

ademirci, 2020-06-04:
**Have you looked for this feature in other issues and in the docs?**  
Thank you for this great software. Yes, I searched in the repsitory and can not see related result.
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No, it is not a problem
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I want to send a notification as Telegram message so that I can recieve to my phone anywhere.
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
There are IOT MQTT alternatives for notification but they are not straightforward.
**Additional context**  
_Add any other context or screenshots about the feature request here._
I noticed following arduino library in the github and maybe it is easy to integrate it to Tasmota
https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot

**(Please, remember to close the issue when the problem has been addressed)**

D0n1elT, 2020-06-04:
That would be Awesome!
I can imagine a command like `notify "..."` for rules or sensor values!!

kugelkopf123, 2020-06-04:
https://github.com/ArduinoHannover/Pushover
Another one that works.

ascillato, 2020-06-05:
Duplicated https://github.com/arendst/Tasmota/issues/1456

arendst, 2020-06-06:
Been working on this yesterday but failed to get a https connection to api.telegram.org

All mentionend libraries fail in this case as is also reported by their different issues.

Tried a very basic approach using all SSL provided libraries but still fail to make a connection.

If anyone manages to make a https connection using the current core version I'll be interessted to dig in it again.

s-hadinger, 2020-06-06:
Theo, I know very well the SSL part. I can try.

arendst, 2020-06-06:
This is the driver code currently not using any library other than the available BearSSL.

[xdrv_50_telegram.zip](https://github.com/arendst/Tasmota/files/4739761/xdrv_50_telegram.zip)

arendst, 2020-06-06:
You'll need this too

[tasmota.zip](https://github.com/arendst/Tasmota/files/4739764/tasmota.zip)

arendst, 2020-06-06:
And define USE_TELEGRAM in user_config_overrid.h

ademirci, 2020-06-06:
I am excited to see the result. Even we can send special format messages to Tasmota to enable or disable functions or rules over internet without the need for an IOT account. Also we can collect some sensor values periodically with another Telegram bot and setup our own IOT dashboard for free.

eku, 2020-06-06:
@ademirci well a plain XMPP/Jabber client would be enougth to reach public or private XMPP/Jabber chat server. 
@arendst I would rather invest the time in support for an open federal platform than support a proprietary solution like Telegram.

s-hadinger, 2020-06-06:
Telegram api is HTTP not XMPP which makes it far easier to integrate in Tasmota.

What do you mean by "open federal platform"?

eku, 2020-06-06:
@s-hadinger wrote:
> What do you mean by "open federal platform"?

Anyone can run a server and you do not need to have an account on the same server where the chat partner has one. Classic examples are Email and XMPP/Jabber, but also Mastodon.

s-hadinger, 2020-06-06:
At first glance, `api.telegram.org` uses a certificate signed by GoDaddy `Daddy Secure Certificate Authority - G2`. The server key is 2048 bits and the proposed cipher is `ECDHE-RSA-AES128-GCM-SHA256` which is exactly the cipher used with AWS IoT.

This is all good news, apart from adding the CA in Flash, it should run smoothly. Stay tuned.

If you're curious about this, try `openssl s_client -showcerts -connect api.telegram.org:443`

arendst, 2020-06-06:
Just did. Nice find!

s-hadinger, 2020-06-06:
And here is the CA ready to be munched by BearSSL (added at the end of `tasmota_ca.ino`:

```
#ifdef USE_TELEGRAM
// we add a separate CA for telegram

/*********************************************************************************************\
 * GoDaddy Daddy Secure Certificate Authority - G2, RSA 2048 bits SHA 256, valid until 20220523
 *
 * to convert do: "brssl ta GoDaddyCA.pem"
 * then copy and paste below, chain the generic names to the same as below
 * remove "static" and add "PROGMEM"
\*********************************************************************************************/

const unsigned char GoDaddyCAG2_DN[] PROGMEM = {
	0x30, 0x3E, 0x31, 0x21, 0x30, 0x1F, 0x06, 0x03, 0x55, 0x04, 0x0B, 0x13,
	0x18, 0x44, 0x6F, 0x6D, 0x61, 0x69, 0x6E, 0x20, 0x43, 0x6F, 0x6E, 0x74,
	0x72, 0x6F, 0x6C, 0x20, 0x56, 0x61, 0x6C, 0x69, 0x64, 0x61, 0x74, 0x65,
	0x64, 0x31, 0x19, 0x30, 0x17, 0x06, 0x03, 0x55, 0x04, 0x03, 0x13, 0x10,
	0x61, 0x70, 0x69, 0x2E, 0x74, 0x65, 0x6C, 0x65, 0x67, 0x72, 0x61, 0x6D,
	0x2E, 0x6F, 0x72, 0x67
};

const unsigned char GoDaddyCAG2_RSA_N[] PROGMEM = {
	0xB4, 0xA3, 0x16, 0x9E, 0x5C, 0x57, 0xC9, 0x89, 0x65, 0xED, 0xEA, 0x78,
	0x0B, 0xAE, 0x8A, 0x58, 0x2F, 0xAE, 0x5A, 0xC8, 0x6E, 0x49, 0x8D, 0xFC,
	0x57, 0xA5, 0x98, 0x88, 0x78, 0x2E, 0x0B, 0x3C, 0x40, 0x3C, 0x21, 0x2E,
	0x9A, 0x94, 0x98, 0x33, 0xA7, 0xE3, 0x42, 0xA7, 0x85, 0xFA, 0xD0, 0x73,
	0x84, 0x01, 0x1C, 0x72, 0x39, 0x37, 0x23, 0xB5, 0x56, 0x1D, 0x43, 0xA5,
	0x71, 0x14, 0x08, 0x24, 0xA5, 0x39, 0xCC, 0xDE, 0x58, 0x53, 0x94, 0x8E,
	0x2A, 0x42, 0xA7, 0x4E, 0x2D, 0x07, 0x32, 0x9E, 0xBA, 0x8B, 0xD3, 0x2A,
	0xA9, 0x9E, 0xC0, 0xE3, 0xCE, 0x9A, 0x10, 0x96, 0x45, 0x58, 0x7A, 0xC7,
	0x1E, 0x45, 0x14, 0x23, 0x92, 0xBB, 0x54, 0x82, 0x88, 0x94, 0x49, 0xB6,
	0xBE, 0x81, 0x21, 0x00, 0x29, 0x6D, 0xC9, 0xCE, 0x8B, 0x39, 0x3A, 0xDC,
	0x35, 0x15, 0xD9, 0xEB, 0x47, 0x9C, 0xEF, 0xBA, 0x09, 0x0E, 0x16, 0xE4,
	0xD9, 0xEB, 0x72, 0x30, 0xFA, 0x49, 0xAB, 0x98, 0x31, 0x7C, 0xB3, 0xAC,
	0x2B, 0x29, 0x91, 0x87, 0x08, 0x41, 0x72, 0x5E, 0x35, 0xC7, 0x87, 0x04,
	0x22, 0xF5, 0x48, 0x76, 0x30, 0x6D, 0x88, 0xDF, 0xF2, 0xA5, 0x29, 0x13,
	0x70, 0xB3, 0x87, 0x02, 0xD5, 0x6B, 0x58, 0xB1, 0xE8, 0x73, 0xC7, 0xE4,
	0xEF, 0x79, 0x86, 0xA4, 0x07, 0x5F, 0x67, 0xB4, 0x79, 0x8D, 0xA4, 0x25,
	0x01, 0x82, 0x8C, 0xE0, 0x30, 0x17, 0xCB, 0x4B, 0x5C, 0xFB, 0xEB, 0x4C,
	0x12, 0x51, 0xB9, 0xC9, 0x04, 0x1F, 0x7E, 0xD2, 0xF8, 0xBA, 0xF5, 0x35,
	0x8D, 0x8A, 0x1C, 0x37, 0x82, 0xF0, 0x15, 0x73, 0x00, 0x6E, 0x3D, 0x1C,
	0x76, 0x8B, 0x01, 0x74, 0x81, 0x3D, 0xE4, 0x2C, 0xA7, 0xCC, 0x2F, 0x66,
	0xDC, 0x44, 0xA8, 0x27, 0x3F, 0xEA, 0xD0, 0xA7, 0xA8, 0xF1, 0xCB, 0xEA,
	0xDA, 0x07, 0x38, 0xBD
};

const unsigned char GoDaddyCAG2_RSA_E[] PROGMEM = {
	0x01, 0x00, 0x01
};

const br_x509_trust_anchor GoDaddyCAG2_TA[1] PROGMEM = {
	{
		{ (unsigned char *)TA0_DN, sizeof TA0_DN },
		0,
		{
			BR_KEYTYPE_RSA,
			{ .rsa = {
				(unsigned char *)GoDaddyCAG2_RSA_N, sizeof GoDaddyCAG2_RSA_N,
				(unsigned char *)GoDaddyCAG2_RSA_E, sizeof GoDaddyCAG2_RSA_E,
			} }
		}
	}
};

#endif // USE_TELEGRAM
```

The downside is that the certificate is valid until May 2022, so we will need to update regularly.

s-hadinger, 2020-06-06:
I will not be available this afternoon to test the whole, but here is how it should work:

```
#include "WiFiClientSecureLightBearSSL.h"
BearSSL::WiFiClientSecure_light *tlsClient;

[...]
  tlsClient = new BearSSL::WiFiClientSecure_light(1024,1024);
  tlsClient->setTrustAnchor(&GoDaddyCAG2_TA);
```

Then pass `tlsClient` to the Telegram lib.

arendst, 2020-06-06:
Thx I'll give it a try today

s-hadinger, 2020-06-06:
Oh I forgot, you need to set `#define USE_MQTT_TLS_FORCE_EC_CIPHER` to force `ECDHE-RSA-AES128-GCM-SHA256`

s-hadinger, 2020-06-06:
I'm trying now,  actually here is the CA:

```
#ifdef USE_TELEGRAM
// we add a separate CA for telegram

/*********************************************************************************************\
 * GoDaddy Daddy Secure Certificate Authority - G2, RSA 2048 bits SHA 256, valid until 20220523
 *
 * to convert do: "brssl ta GoDaddyCA.pem"
 * then copy and paste below, chain the generic names to the same as below
 * remove "static" and add "PROGMEM"
\*********************************************************************************************/

const unsigned char GoDaddyCAG2_DN[] PROGMEM = {
	0x30, 0x3E, 0x31, 0x21, 0x30, 0x1F, 0x06, 0x03, 0x55, 0x04, 0x0B, 0x13,
	0x18, 0x44, 0x6F, 0x6D, 0x61, 0x69, 0x6E, 0x20, 0x43, 0x6F, 0x6E, 0x74,
	0x72, 0x6F, 0x6C, 0x20, 0x56, 0x61, 0x6C, 0x69, 0x64, 0x61, 0x74, 0x65,
	0x64, 0x31, 0x19, 0x30, 0x17, 0x06, 0x03, 0x55, 0x04, 0x03, 0x13, 0x10,
	0x61, 0x70, 0x69, 0x2E, 0x74, 0x65, 0x6C, 0x65, 0x67, 0x72, 0x61, 0x6D,
	0x2E, 0x6F, 0x72, 0x67
};

const unsigned char GoDaddyCAG2_RSA_N[] PROGMEM = {
	0xB4, 0xA3, 0x16, 0x9E, 0x5C, 0x57, 0xC9, 0x89, 0x65, 0xED, 0xEA, 0x78,
	0x0B, 0xAE, 0x8A, 0x58, 0x2F, 0xAE, 0x5A, 0xC8, 0x6E, 0x49, 0x8D, 0xFC,
	0x57, 0xA5, 0x98, 0x88, 0x78, 0x2E, 0x0B, 0x3C, 0x40, 0x3C, 0x21, 0x2E,
	0x9A, 0x94, 0x98, 0x33, 0xA7, 0xE3, 0x42, 0xA7, 0x85, 0xFA, 0xD0, 0x73,
	0x84, 0x01, 0x1C, 0x72, 0x39, 0x37, 0x23, 0xB5, 0x56, 0x1D, 0x43, 0xA5,
	0x71, 0x14, 0x08, 0x24, 0xA5, 0x39, 0xCC, 0xDE, 0x58, 0x53, 0x94, 0x8E,
	0x2A, 0x42, 0xA7, 0x4E, 0x2D, 0x07, 0x32, 0x9E, 0xBA, 0x8B, 0xD3, 0x2A,
	0xA9, 0x9E, 0xC0, 0xE3, 0xCE, 0x9A, 0x10, 0x96, 0x45, 0x58, 0x7A, 0xC7,
	0x1E, 0x45, 0x14, 0x23, 0x92, 0xBB, 0x54, 0x82, 0x88, 0x94, 0x49, 0xB6,
	0xBE, 0x81, 0x21, 0x00, 0x29, 0x6D, 0xC9, 0xCE, 0x8B, 0x39, 0x3A, 0xDC,
	0x35, 0x15, 0xD9, 0xEB, 0x47, 0x9C, 0xEF, 0xBA, 0x09, 0x0E, 0x16, 0xE4,
	0xD9, 0xEB, 0x72, 0x30, 0xFA, 0x49, 0xAB, 0x98, 0x31, 0x7C, 0xB3, 0xAC,
	0x2B, 0x29, 0x91, 0x87, 0x08, 0x41, 0x72, 0x5E, 0x35, 0xC7, 0x87, 0x04,
	0x22, 0xF5, 0x48, 0x76, 0x30, 0x6D, 0x88, 0xDF, 0xF2, 0xA5, 0x29, 0x13,
	0x70, 0xB3, 0x87, 0x02, 0xD5, 0x6B, 0x58, 0xB1, 0xE8, 0x73, 0xC7, 0xE4,
	0xEF, 0x79, 0x86, 0xA4, 0x07, 0x5F, 0x67, 0xB4, 0x79, 0x8D, 0xA4, 0x25,
	0x01, 0x82, 0x8C, 0xE0, 0x30, 0x17, 0xCB, 0x4B, 0x5C, 0xFB, 0xEB, 0x4C,
	0x12, 0x51, 0xB9, 0xC9, 0x04, 0x1F, 0x7E, 0xD2, 0xF8, 0xBA, 0xF5, 0x35,
	0x8D, 0x8A, 0x1C, 0x37, 0x82, 0xF0, 0x15, 0x73, 0x00, 0x6E, 0x3D, 0x1C,
	0x76, 0x8B, 0x01, 0x74, 0x81, 0x3D, 0xE4, 0x2C, 0xA7, 0xCC, 0x2F, 0x66,
	0xDC, 0x44, 0xA8, 0x27, 0x3F, 0xEA, 0xD0, 0xA7, 0xA8, 0xF1, 0xCB, 0xEA,
	0xDA, 0x07, 0x38, 0xBD
};

const unsigned char GoDaddyCAG2_RSA_E[] PROGMEM = {
	0x01, 0x00, 0x01
};

const br_x509_trust_anchor GoDaddyCAG2_TA PROGMEM = {
	{ (unsigned char *)GoDaddyCAG2_DN, sizeof GoDaddyCAG2_DN },
	0,
	{
		BR_KEYTYPE_RSA,
		{ .rsa = {
			(unsigned char *)GoDaddyCAG2_RSA_N, sizeof GoDaddyCAG2_RSA_N,
			(unsigned char *)GoDaddyCAG2_RSA_E, sizeof GoDaddyCAG2_RSA_E,
		} }
	}
};

#endif // USE_TELEGRAM
```

s-hadinger, 2020-06-06:
I'm progressing. Call is done, I'm now getting error 296 which is `SSL3_ALERT_HANDSHAKE_FAILURE`

I will continue tomorrow. The current code is here: https://github.com/s-hadinger/Tasmota/tree/telegram_dev

s-hadinger, 2020-06-06:
I got a first version working by disabling CA validation, otherwise it seems to hang.

First good news is TLS was indeed connected, second good news is that Telegram supports MFLN so we can keep 1KB buffers. Bad news is it crashed in the destructor of HTTPClient, but I know why.

Fun fact: after doing lots of tests by IP address got black-listed. I needed to switch to mobile.

```
GET Update Messages
Client initialized
connect(api.telegram.org,443)
connect(149.154.167.220,443)
_connectSSL: start connection
Connected! MFLNStatus = 1
[HTTPS] GET... code: 404

Exception (28):
epc1=0x4023a0e7 epc2=0x00000000 epc3=0x40248b9d excvaddr=0x000002c0 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffc40 end: 3fffffc0 offset: 0190
3ffffdd0:  00000001 3fff2fb8 3fff4b84 40208e70
```

s-hadinger, 2020-06-06:
Crash fixed. It seems to connect correctly, since I get a 404 error because of a fake bot token.

Theo, the code shouldn't retry automatically when there is an error, because it ends up hammering the api endpoint and makes the ip address black-listed.

Log is very verbose for now. You need for now to `#define USE_TELEGRAM` and do not define any TLS option, especially don't enable USE_MQTT_TLS_CA_CERT (for now)

s-hadinger, 2020-06-07:
Progressing. Connection works well and I need to clean up code.

However the current logic reconnects every second, which is problematic: the actual SSL connection takes around 1 second and freezes the entire Tasmota. Is long polling possible? Or should me relax the frequency?

Also send does not seem to work (yet).

```
connect(api.telegram.org,443)
connect(149.154.167.220,443)
_connectSSL: start connection
Connected! MFLNStatus = 1
[HTTPS] GET... code: 200
[HTTPS] Received payload telegram:
1BTC = {"ok":true,"result":[]}USD
Sent Update request messages up to : 0
no new messages
```

s-hadinger, 2020-06-07:
Hi have a dilemma, for TSL server validation:

1. validate the CA. Problem: the server certificate is bigger than 1024 bytes, so I need to increase the receive buffer from 1024 to 1536 bytes and eats up 512 more byte or RAM. Also, the certificate is only valid until 2022.
2. is fingerprint validation of server public key. I have no clue about the stability of their public key, it may break tomorrow or stay stable for 10 years.
3. don't check the server identity which is subject to man-in-the-middle attack.

I propose to implement both 1 and 2, it will depend on `#define USE_MQTT_TLS_CA_CERT` value.

s-hadinger, 2020-06-07:
Done. The TLS is fully working, both with CA validation or Fingerprint validation. If the fingerprint is wrong, the new one is shown in logs.

@arendst You can now work on the fun part :)

All needed changes are in this branch: https://github.com/s-hadinger/Tasmota/tree/telegram_dev

arendst, 2020-06-07:
Thx for the basics. Will see tomorrow.

arendst, 2020-06-08:
Just had a very quick look but failed to get anything running. Both my and your repository end in this:

```
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP2, SSId indebuurt_IoT, Channel 1, BSSId 74:83:C2:2A:F1:AC, RSSI -85, Encryption 1
00:00:04 WIF: Network 1, AP2, SSId indebuurt_IoT, Channel 11, BSSId 18:E8:29:CA:17:C1, RSSI -23, Encryption 1
00:00:04 WIF: Connecting to AP2 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11N as wemos4...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on wemos4 with IP address 192.168.2.223
18:25:27 NTP: UTC 2020-06-08T16:25:26, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
18:25:27 TLG: Started
18:25:28 MQT: Attempting connection...
18:25:28 MQT: Connected
18:25:28 MQT: tele/wemos4/LWT = Online (retained)
18:25:28 MQT: cmnd/wemos4/POWER = 
18:25:28 MQT: Subscribe to cmnd/wemos4/#
18:25:28 MQT: Subscribe to cmnd/tasmotas/#
18:25:28 MQT: Subscribe to cmnd/DVES_83BB10_fb/#
18:25:28 MQT: tele/wemos4/INFO1 = {"Module":"Sonoff Basic","Version":"8.3.1.2(tasmota)","FallbackTopic":"cmnd/DVES_83BB10_fb/","GroupTopic":"cmnd/tasmotas/"}
18:25:28 MQT: tele/wemos4/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos4","IPAddress":"192.168.2.223"}
18:25:28 MQT: tele/wemos4/INFO3 = {"RestartReason":{"Exception":4,"Reason":"Software Watchdog","EPC":["40003b53","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000"}}
18:25:28 MQT: stat/wemos4/RESULT = {"POWER":"OFF"}
18:25:28 MQT: stat/wemos4/POWER = OFF
GET Update Messages 
18:25:28 TLG: Cmnd bot1179906608:AAE-3UfPJt3u8d-gbGXslnFO96o3DahG6EM/getUpdates?offset=0

User exception (panic/abort/assert)
Abort called

>>>stack>>>

ctx: sys
sp: 3fff7ee0 end: 3fffffb0 offset: 0000
3fff7ee0:  f489b847 deadbeef 00000004 3fff6630  
3fff7ef0:  000000fe 00000000 00000000 00000000  
3fff7f00:  00000000 00000000 00000000 3fff7f30  
3fff7f10:  00000004 00000090 00000004 f489b847  
3fff7f20:  00000008 00000000 3fff5e4c 4024f63a  
3fff7f30:  3ffe892a 00000000 00000000 4024f64c  
3fff7f40:  3fff7c60 00000000 3fff5e4c 402085fd  
3fff7f50:  deadbeef 00000000 3fff5e4c 4020868c  
3fff7f60:  deadbeef 00000000 068508d8 40283e8c  
3fff7f70:  00000001 3fff54b8 40284128 40284128  
3fff7f80:  00000000 00000001 3fff81dc 00000100  
3fff7f90:  3fff82dc 00000003 3fff7fa4 3fff803c  
3fff7fa0:  4028d528 ffffffff 00000000 00000000  
3fff7fb0:  00000000 000000c2 00000000 0000001f  
3fff7fc0:  0000001f 000000ae 00000000 0000001f 
```

No worries. Will have a better look tomorrow.

s-hadinger, 2020-06-08:
You can try in `WiFiClientSecureLightBearSSL.cpp`:
```
#define DEBUG_TLS
#define DEBUG_ESP_SSL
```

arendst, 2020-06-09:
Frustrating but all I get is this:

```
00:00:01 HTP: Web server active on wemos4 with IP address 192.168.2.223
15:13:26 NTP: UTC 2020-06-09T13:13:25, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
15:13:26 TLG: Started
15:13:27 MQT: Attempting connection...
15:13:27 MQT: Connected
15:13:27 MQT: tele/wemos4/LWT = Online (retained)
15:13:27 MQT: cmnd/wemos4/POWER = 
15:13:27 MQT: Subscribe to cmnd/wemos4/#
15:13:27 MQT: Subscribe to cmnd/tasmotas/#
15:13:27 MQT: Subscribe to cmnd/DVES_83BB10_fb/#
15:13:27 MQT: tele/wemos4/INFO1 = {"Module":"Sonoff Basic","Version":"8.3.1.2(theo)","FallbackTopic":"cmnd/DVES_83BB10_fb/","GroupTopic":"cmnd/tasmotas/"}
15:13:27 MQT: tele/wemos4/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos4","IPAddress":"192.168.2.223"}
15:13:27 MQT: tele/wemos4/INFO3 = {"RestartReason":"External System"}
15:13:27 MQT: stat/wemos4/RESULT = {"POWER":"OFF"}
15:13:27 MQT: stat/wemos4/POWER = OFF
15:13:31 MQT: tele/wemos4/STATE = {"Time":"2020-06-09T15:13:31","Uptime":"0T00:00:09","UptimeSec":9,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-28,"LinkCount":1,"Downtime":"0T00:00:03"}}
15:13:32 APP: Boot Count 161
15:13:32 CFG: Saved to flash at F5, Count 290, Bytes 4096
GET Update Messages 
StackThunk before 27496, Fragmentation=2, Thunkstack=0, Free stack=3664, FreeContStack=1680
StackThunk after 27496, Fragmentation=2, Thunkstack=0, Free stack=3664, FreeContStack=1680
allocateBuffers before 27496, Fragmentation=2, Thunkstack=0, Free stack=3616, FreeContStack=1680
allocateBuffers ClientContext 24056, Fragmentation=2, Thunkstack=0, Free stack=3616, FreeContStack=1680
allocateBuffers after 21528, Fragmentation=2, Thunkstack=0, Free stack=3616, FreeContStack=1680
15:13:33 TLG: Cmnd bot1179906608:AAE-3UfPJt3u8d-gbGXslnFO96o3DahG6EM/getUpdates?offset=0
connect(api.telegram.org,443)
connect(149.154.167.220,443)
Before calling _connectSSL 21192, Fragmentation=1, Thunkstack=0, Free stack=3632, FreeContStack=1680
_connectSSL.start 21192, Fragmentation=1, Thunkstack=0, Free stack=3600, FreeContStack=1680
Thunk allocated 17584, Fragmentation=1, Thunkstack=0, Free stack=3600, FreeContStack=1680
_connectSSL: start connection
_connectSSL before DecoderContext allocation 17584, Fragmentation=1, Thunkstack=0, Free stack=3600, FreeContStack=1680
_connectSSL after DecoderContext allocation 16432, Fragmentation=1, Thunkstack=0, Free stack=3600, FreeContStack=1680
_connectSSL before PrivKey allocation 16432, Fragmentation=1, Thunkstack=0, Free stack=3600, FreeContStack=1680
_wait_for_handshake: start

User exception (panic/abort/assert)
Abort called

>>>stack>>>

ctx: sys
sp: 3fff7b38 end: 3fffffb0 offset: 0000
3fff7b38:  f2532668 deadbeef 00000004 3fff5e18  
3fff7b48:  000000fe 00000000 00000000 00000000  
3fff7b58:  00000000 00000000 00000000 3fff7b88 
```
It seems it stalls in ``_run_until`` called from ``_wait_for_handshake`` but I don't see why.

s-hadinger, 2020-06-09:
It seems to be a conditional compile issue. Get the latest commit to enable CA validation without a private key, and compile with both `#define USE_MQTT_TLS_CA_CERT` and `#define USE_MQTT_TLS_FORCE_EC_CIPHER`.

I'll check later how to make sure it compiles with the right options.

s-hadinger, 2020-06-09:
I found the issue, the Stack Thunk was not correctly adjusted. Now the branch is good: https://github.com/s-hadinger/Tasmota/tree/telegram_dev

arendst, 2020-06-11:
Initial support for Telegram messaging allows for sending commands to a Tasmota device and receiving response. It also allows sending (user)text from Tasmota to Telegram.

As comms between the two is slow (every message needs at least 1 second to complete connection and transfer) it's more like a gadget than a feature.

Anyway. Enabling Telegram support needs at least uncomment of ``define USE_TELEGRAM`` and uncomment of either ``define USE_TELEGRAM_FINGERPRINT`` OR ``define USE_MQTT_TLS_CA_CERT``

Default, once enabled in user_config_override it does nothing. It needs to know your bots token and chat id. Enter them with command ``TmToken`` and ``TmChatId``.

There are three state commands allowing control over the comms.

- First to test your connection try ``TmState 5`` allowing for echoing of your bot chat text.
- Second stop the echo with ``TmState 4`` and enable sending text from Tasmota to telegram with command ``TmState 1``. Now enter on tasmota ``TmSend Howdy`` which should report ``Howdy`` in telegram.
- Now in addition to sending from Tasmota you may enable receiving commands from telegram with option ``TmState 3``. Enter any tasmota command in telegram and watch it's response in both tasmota AND in the chat window.

Default polling time is 10 seconds but can be changed with command ``PmPoll`` from 4 to 300 seconds.

Complete command syntax can be found in the header of file ``xdrv_40_telegram.ino``

s-hadinger, 2020-06-11:
Nice !!!

ademirci, 2020-06-11:
I really admire you guys. Thanks for this great tasmota software and thanks for this great effort to enable tasmota to send and receive telegram messages. I will test this as soon as possible and share my experience.

ademirci, 2020-06-11:
I just test this and working well. I can send commands to tasmota and receive messages from tasmota to my telegram account at my phone. Thanks again. I close this request.

closed 2020-06-11 by ademirci

JackNJack, 2020-07-07:
Hello! Tried Telegram my device can react to echo but when enable sending and when I send some text it always failed to send. 
This is my log:

> CMD: tmstate 5
01:04:31 SRC: WebConsole from 192.168.1.18
01:04:31 CMD: Group 0, Index 1, Command "TMSTATE", Data "5"
01:04:31 RSL: stat/tasmota_089E9D/RESULT = {"TmState":{"Send":"OFF","Receive":"OFF","Echo":"ON"}}
01:04:32 TGM: Started
01:04:33 TGM: getUpdates
01:04:36 TGM: sendMessage
01:04:42 WIF: Checking connection...
01:04:50 TGM: getUpdates
01:04:53 TGM: sendMessage
01:05:07 WIF: Checking connection...
01:05:08 TGM: getUpdates
01:05:17 CMD: tmstate 4
01:05:17 SRC: WebConsole from 192.168.1.18
01:05:17 CMD: Group 0, Index 1, Command "TMSTATE", Data "4"
01:05:17 RSL: stat/tasmota_089E9D/RESULT = {"TmState":{"Send":"OFF","Receive":"OFF","Echo":"OFF"}}
01:05:25 CMD: tmstate 1
01:05:25 SRC: WebConsole from 192.168.1.18
01:05:25 CMD: Group 0, Index 1, Command "TMSTATE", Data "1"
01:05:25 RSL: stat/tasmota_089E9D/RESULT = {"TmState":{"Send":"ON","Receive":"OFF","Echo":"OFF"}}
01:05:29 WIF: Checking connection...
01:05:33 CMD: tmsend hello
01:05:33 SRC: WebConsole from 192.168.1.18
01:05:33 CMD: Group 0, Index 1, Command "TMSEND", Data "hello"
01:05:33 TGM: sendMessage
01:05:35 RSL: stat/tasmota_089E9D/RESULT = {"TmSend":"Failed"}
01:05:52 WIF: Checking connection...
01:06:00 CMD: tmstate 5
01:06:00 SRC: WebConsole from 192.168.1.18
01:06:00 CMD: Group 0, Index 1, Command "TMSTATE", Data "5"
01:06:00 RSL: stat/tasmota_089E9D/RESULT = {"TmState":{"Send":"ON","Receive":"OFF","Echo":"ON"}}
01:06:05 TGM: getUpdates
01:06:08 CMD: tmsend hello!
01:06:08 SRC: WebConsole from 192.168.1.18
01:06:08 CMD: Group 0, Index 1, Command "TMSEND", Data "hello!"
01:06:08 TGM: sendMessage
01:06:11 RSL: stat/tasmota_089E9D/RESULT = {"TmSend":"Failed"}

This is my staus report:

> 01:08:26 CMD: Group 0, Index 1, Command "STATUS", Data "0"
01:08:26 RSL: stat/tasmota_089E9D/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_089E9D","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
01:08:26 RSL: stat/tasmota_089E9D/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"","RestartReason":"External System","Uptime":"0T00:04:33","StartupUTC":"2020-07-07T17:03:53","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2020-07-08T00:37:06","SaveCount":12,"SaveAddress":"F8000"}}
01:08:26 RSL: stat/tasmota_089E9D/STATUS2 = {"StatusFWR":{"Version":"8.3.1.6(tasmota)","BuildDateTime":"2020-07-08T00:35:17","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"398/699"}}
01:08:26 RSL: stat/tasmota_089E9D/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RadarCNSSO","RadarCNSSO"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000"]}}
01:08:26 RSL: stat/tasmota_089E9D/STATUS4 = {"StatusMEM":{"ProgramSize":646,"Free":356,"Heap":18,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","043683A1","020000CD","010013C0","C000F981","0000C024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,40","Sensors":"1,2,3,4,5,6,22"}}
01:08:26 RSL: stat/tasmota_089E9D/STATUS5 = {"StatusNET":{"Hostname":"tasmota_089E9D-7837","IPAddress":"192.168.1.28","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"C8:2B:96:08:9E:9D","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
01:08:26 RSL: stat/tasmota_089E9D/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_089E9D","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
01:08:26 RSL: stat/tasmota_089E9D/STATUS7 = {"StatusTIM":{"UTC":"2020-07-07T17:08:26","Local":"2020-07-08T01:08:26","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+08:00","Sunrise":"11:56","Sunset":"03:53"}}
01:08:27 RSL: stat/tasmota_089E9D/STATUS10 = {"StatusSNS":{"Time":"2020-07-08T01:08:26"}}
01:08:27 RSL: stat/tasmota_089E9D/STATUS11 = {"StatusSTS":{"Time":"2020-07-08T01:08:27","Uptime":"0T00:04:34","UptimeSec":274,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":2,"SSId":"********","BSSId":"0C:2A:86:DD:C5:AE","Channel":1,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:04"}}}

JackNJack, 2020-07-10:
Re: Bug on TelegramGetUpdates()
If you send a 'restart 1' command to the bot it will re-execute the restart after restarting. As if 'restart 1' is new unread command. Can anyone help?

JackNJack, 2020-07-22:
I experimented this almost a 3weeks now. I observed that `telegramClient->connect("api.telegram.org", 443)`  is triggering exception 29 whenever my connection to internet is slow. Also, it takes almost 2sec (sometimes 30sec even if I included 1sec timeout in my telegram request `String command = "bot" + _token + "/getUpdates?offset=" + offset + "&limit=1&timeout=1";`) to connect. Which is why some of the drivers like the webserver is also slow on its response. Can anyone show me a way to solve this? Or is it possible to multithread the process so that I will only process the request result if it is already available and entertain other drivers while waiting for the result.

s-hadinger, 2020-07-22:
30s is very long. Do you know if the time is spent at the TLS connection or later at HTTP level?

You could remove comments on line 105, to have a precise timing of TLS connection, and even move to `LOG_LEVEL_INFO`:
```
    AddLog_P2(LOG_LEVEL_DEBUG, PSTR("TGM: Connected in %d ms, max ThunkStack used %d"), millis() - tls_connect_time, telegramClient->getMaxThunkStackUse());

```

JackNJack, 2020-07-22:
>  AddLog_P2(LOG_LEVEL_DEBUG, PSTR("TGM: Connected in %d ms, max ThunkStack used %d"), millis() - tls_connect_time, telegramClient->getMaxThunkStackUse());

Thanks for the reply @s-hadinger, I actually uncommented that part and it shows ~2000ms, sometimes 30000ms.
`18:38:46 TGM: Connected in 21335 ms, max ThunkStack used 4516`
`TGM: Connected in 1941 ms, max ThunkStack used 4516` --> This is my usual connection about 1800 to 1900+

s-hadinger, 2020-07-22:
So it is definitely a problem during TLS connection. 21s is far too long.

You can try to temporarily double buffers, I suspect this may have some side effects (warning it eats up more RAM):
```
  static const uint32_t tls_rx_size = 4096;
  static const uint32_t tls_tx_size = 2048;
```

JackNJack, 2020-07-22:
> So it is definitely a problem during TLS connection. 21s is far too long.
> 
> You can try to temporarily double buffers, I suspect this may have some side effects (warning it eats up more RAM):
> 
> ```
>   static const uint32_t tls_rx_size = 4096;
>   static const uint32_t tls_tx_size = 2048;
> ```
Thanks @s-hadinger , i'll try it.
Based on Info: My RAM has 20KB free. When i dont use telegram it would go up to 28KB.

JackNJack, 2020-07-22:
@s-hadinger my RAM went down to 15KB free, but the same performance. Response in 2sec nominal, while sometimes it reached 5sec to 15sec
`TGM: Connected in 14571 ms, max ThunkStack used 4484`

s-hadinger, 2020-07-22:
Ok, it's not the problem then. Logs for bearssl are quite obscure, I don't have any easy solution.

Just to make sure, is the ESP running at 160MHz instead of standard 80 MHz?

arendst, 2020-07-23:
Just to note mine runs fine on 80MHz with quick responses around 2 Secs.

JackNJack, 2020-07-23:
Mine also runs at 80Mhz. When I reduce both TLS rx/tx size to 512 it performs much better with 1800ms to 2000ms response. But when internet is slow it goes up to 5sec. I reduce the size to 512 since in my application I only receive up to 10 chars of string with single message each GET.

s-hadinger, 2020-07-23:
Good to know that lowering the buffer size actually improves performance. I would still try 160 Mhz.

riker65, 2020-08-01:
Hi, great news with the telegram support
but where do I find docu how to configure

thanks for guiding

ascillato, 2020-08-01:
> where do I find docu how to configure

https://github.com/arendst/Tasmota/issues/8619#issuecomment-642764152

riker65, 2020-08-02:
> > where do I find docu how to configure
> 
> [#8619 (comment)](https://github.com/arendst/Tasmota/issues/8619#issuecomment-642764152)

Hi I thanks

I thought it is enabled in the latest prebuild release. 
So do I have to create an own build?

arendst, 2020-08-02:
As it's code footprint is large due to TLS support you will need to compile yourselves. You might even need to drop some features to get it even loaded.

ademirci, 2020-08-06:
Is it possible to send an image by using TmSend method? I have an old foscam lying around and I want to get a snapshot from it by esp and send it to my telegram. I just wanted to make sure if the function supports such image sending.

arendst, 2020-08-07:
@ademirci nope. Note enough resources (memory)

fragolinux, 2020-10-10:
hi, can someone explain better how to SEND commands to a tasmota device? I recompiled the bin file adding the 2 use_telegram* directives, all fine, i got all the needed commands... i followed instructions on https://github.com/arendst/Tasmota/issues/8619#issuecomment-642764152 link and succeeded in sending text via tmsend command to the bot, receiving it no problems... even thought the correct chatid is the one of the USER, not the one of the bot itself (which in the end is the part before ":" in the token string)... with MY chat id configured via tmchatid i can do tmsend, while putting in the bot one i get tmsend: failed results... so, given i can SEND text to the bot, and all the tmstate where set to on via "backlog tmstate 1; tmstate 3 ;tmstate 5", which is the correct syntax to send command from the bot TO the tasmota device? I tried "cmnd power 2", "power 2", "cmnd/albero/power 2", and even added a "/" in front of the same commands, nothing appens on tasmota device... thanks in advance :)

fragolinux, 2020-10-15:
hi, flashed a custom 9.0.0.2 (base: standard tasmota.bin config) with the 2 telegram support lines added, on a brand new wemos d1 mini, set as module 18 and with just led1i on d4 and relay1 on d2, nothing else except wifi config... reduced tmpoll to 4, added tmtoken and tmchatid, set tmstate to send and receive ON (not echo), and i can tmsend from device to bot with no problems (just a bit lag), but to receive commands and have them working it's all an other story... i've to intercept the correct instant on which it's going to run its periodical getupdates command to have them arrived on device and have status back as answer... not so reliable, and it's a clean setup... any hint? thanks

arendst, 2020-10-15:
Lowering the tmpoll has hardly any use as setting up the connection to telegram takes seconds. The same for sending back information.

To see what you receive enable more logging with ``seriallog 4``.

> i've to intercept the correct instant on which it's going to run its periodical getupdates command to have them arrived on device and have status back as answer... not so reliable

In my situation it always receives the command and sends back the result but as it is only polled once every ten seconds it has it's delays. Again this is all caused by setting up a TLS connection to Telegram. Sending one command per ten seconds works fine. More commands are just queued by telegram as Tasmota only handles one at a time for obvious reasons; there is more to do than servicing telegram.

fragolinux, 2020-10-15:
nope, it does not even recognize with sure result a message every 20 seconds... weblog 4 reports very low info, it seems i get a valid command action and response back only if i succeed to sent a command just before a getstatus appears in console... i sent about 20 "power 2" commands from tg bot chat, waiting even 20 or 30 seconds after a single one, or increasing "rate" to as fast i can type on phone, about 1 message every 2 seconds, it works only as i said... logs:

```
20:58:29 WIF: Checking connection...
20:58:38 CMD: tmstate
20:58:38 SRC: WebConsole from 192.168.1.57
20:58:38 CMD: Group 0, Index 1, Command "TMSTATE", Data ""
20:58:38 RSL: stat/tasmota_7E0E3E/RESULT = {"TmState":{"Send":"OFF","Receive":"OFF","Echo":"OFF"}}
20:58:46 CMD: tmstate 1
20:58:46 SRC: WebConsole from 192.168.1.57
20:58:46 CMD: Group 0, Index 1, Command "TMSTATE", Data "1"
20:58:46 RSL: stat/tasmota_7E0E3E/RESULT = {"TmState":{"Send":"ON","Receive":"OFF","Echo":"OFF"}}
20:58:49 WIF: Checking connection...
20:58:49 CMD: tmstate 3
20:58:49 SRC: WebConsole from 192.168.1.57
20:58:49 CMD: Group 0, Index 1, Command "TMSTATE", Data "3"
20:58:49 RSL: stat/tasmota_7E0E3E/RESULT = {"TmState":{"Send":"ON","Receive":"ON","Echo":"OFF"}}
20:58:50 TGM: Started
20:58:51 TGM: getUpdates
20:58:53 TGM: Response {"ok":true,"result":[]}
20:58:53 CMD: tmstate
20:58:53 SRC: WebConsole from 192.168.1.57
20:58:53 CMD: Group 0, Index 1, Command "TMSTATE", Data ""
20:58:53 RSL: stat/tasmota_7E0E3E/RESULT = {"TmState":{"Send":"ON","Receive":"ON","Echo":"OFF"}}
20:59:02 CMD: tmtoken
20:59:02 SRC: WebConsole from 192.168.1.57
20:59:02 CMD: Group 0, Index 1, Command "TMTOKEN", Data ""
20:59:02 RSL: stat/tasmota_7E0E3E/RESULT = {"TmToken":"105XXXX330:AAHLvwnvXXXXXXXXXXXXXXXXvdx-1kHESyo"}
20:59:05 TGM: getUpdates
20:59:06 TGM: Response {"ok":true,"result":[]}
20:59:06 CMD: tmchatid
20:59:06 SRC: WebConsole from 192.168.1.57
20:59:06 CMD: Group 0, Index 1, Command "TMCHATID", Data ""
20:59:06 RSL: stat/tasmota_7E0E3E/RESULT = {"TmChatId":"33XXXXX63"}
20:59:10 CMD: tmsend test
20:59:10 SRC: WebConsole from 192.168.1.57
20:59:10 CMD: Group 0, Index 1, Command "TMSEND", Data "test"
20:59:10 TGM: sendMessage
20:59:11 RSL: stat/tasmota_7E0E3E/RESULT = {"TmSend":"Done"}
20:59:14 WIF: Checking connection...
20:59:18 TGM: getUpdates
20:59:19 TGM: Response {"ok":true,"result":[]}
20:59:31 TGM: getUpdates
20:59:32 TGM: Response {"ok":true,"result":[]}
20:59:36 WIF: Checking connection...
20:59:44 TGM: getUpdates
20:59:45 TGM: Response {"ok":true,"result":[]}
20:59:57 WIF: Checking connection...
20:59:57 TGM: getUpdates
20:59:58 TGM: Response {"ok":true,"result":[]}
21:00:10 TGM: getUpdates
21:00:11 TGM: Response {"ok":true,"result":[]}
21:00:19 WIF: Checking connection...
21:00:20 CMD: weblog
21:00:20 SRC: WebConsole from 192.168.1.57
21:00:20 CMD: Group 0, Index 1, Command "WEBLOG", Data ""
21:00:20 RSL: stat/tasmota_7E0E3E/RESULT = {"WebLog":4}
21:00:23 TGM: getUpdates
21:00:24 TGM: Response {"ok":true,"result":[]}
21:00:36 TGM: getUpdates
21:00:37 TGM: Response {"ok":true,"result":[]}
21:00:41 WIF: Checking connection...
21:00:49 TGM: getUpdates
21:00:50 TGM: Response {"ok":true,"result":[]}
21:01:02 TGM: getUpdates
21:01:03 TGM: Response {"ok":true,"result":[]}
21:01:04 WIF: Checking connection...
21:01:15 TGM: getUpdates
21:01:16 TGM: Response {"ok":true,"result":[]}
21:01:25 WIF: Checking connection...
21:01:28 TGM: getUpdates
21:01:29 TGM: Response {"ok":true,"result":[{"update_id":xxxxxx,
"message":{"message_id":247,"from":{"id":33XXXXX63,"is_bot":false,"first_name":"xxxxxx","last_name":"xxxxxxx","username":"xxxxxxx","language_code":"it"},"chat":{"id":33XXXXX63,"first_name":"xxxxx","last_name":"xxxxxxx","username":"xxxxxxxx","type":"private"},"date":1602792069,"text":"Power 2"}}]}
21:01:29 TGM: Parsed update_id xxxxxxxxxxx, chat_id 33XXXXX63, text "Power 2"
21:01:29 SRC: Chat
21:01:29 CMD: Group 0, Index 1, Command "POWER", Data "2"
21:01:29 RSL: stat/tasmota_7E0E3E/RESULT = {"POWER":"ON"}
21:01:29 RSL: stat/tasmota_7E0E3E/POWER = ON
21:01:29 TGM: sendMessage
21:01:31 CFG: Saved to flash at F6, Count 30, Bytes 4096
21:01:41 TGM: getUpdates
21:01:44 TGM: Response 
21:01:50 WIF: Checking connection...
21:01:55 RSL: tele/tasmota_7E0E3E/STATE = {"Time":"2020-10-15T21:01:55","Uptime":"0T10:15:24","UptimeSec":36924,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"SharkIOT","BSSId":"20:28:18:A0:F0:42","Channel":11,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:06"}}
21:01:55 TGM: getUpdates
21:01:56 TGM: Response {"ok":true,"result":[]}
```

arendst, 2020-10-16:
I've done the same with the default tasmota.bin and just enabled define USE_TELEGRAM in my_user_config.h

- tmchatid is empty
- tmtoken is set correctly
- tmstate is set to 3
- entering power 2 on the phone performs just that (after at max 13 seconds as that's the default tasmota telegram poll timeout).

See output below:
```
11:11:43 CMD: tmchatid
11:11:43 SRC: Serial
11:11:43 CMD: Group 0, Index 1, Command "TMCHATID", Data ""
11:11:43 MQT: stat/wemos4/RESULT = {"TmChatId":""}
11:12:25 CMD: tmtoken
11:12:25 SRC: Serial
11:12:25 CMD: Group 0, Index 1, Command "TMTOKEN", Data ""
11:12:25 MQT: stat/wemos4/RESULT = {"TmToken":""}
11:12:30 CMD: tmtoken 1xxxxxxx08:AyyyyyyyyyyyyyyyzzzxxxxzzzzzzzzzzzM
11:12:30 SRC: Serial
11:12:30 CMD: Group 0, Index 1, Command "TMTOKEN", Data "1xxxxxxx08:AyyyyyyyyyyyyyyyzzzxxxxzzzzzzzzzzzM"
11:12:30 CFG: CR 398/699, Busy 0
11:12:30 MQT: stat/wemos4/RESULT = {"TmToken":"1xxxxxxx08:AyyyyyyyyyyyyyyyzzzxxxxzzzzzzzzzzzM"}
11:12:31 CFG: Saved to flash at F9, Count 79, Bytes 4096
11:12:40 TGM: Started
11:12:41 CMD: tmstate
11:12:41 SRC: Serial
11:12:41 CMD: Group 0, Index 1, Command "TMSTATE", Data ""
11:12:41 MQT: stat/wemos4/RESULT = {"TmState":{"Send":"OFF","Receive":"ON","Echo":"OFF"}}

on phone in chat: status 3

11:13:07 SRC: Chat
11:13:07 CMD: Group 0, Index 1, Command "STATUS", Data "3"
11:13:07 MQT: stat/wemos4/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["indebuurt_IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805FA000100060000005A00000000000000","00000000","00006000","00000000"]}}

on phone in chat: power 2

11:13:46 SRC: Chat
11:13:46 CMD: Group 0, Index 1, Command "POWER", Data "2"
11:13:46 MQT: stat/wemos4/RESULT = {"POWER":"ON"}
11:13:46 MQT: stat/wemos4/POWER = ON
11:13:47 CFG: Saved to flash at F8, Count 80, Bytes 4096

on phone in chat: power 2

11:14:12 SRC: Chat
11:14:12 CMD: Group 0, Index 1, Command "POWER", Data "2"
11:14:12 MQT: stat/wemos4/RESULT = {"POWER":"OFF"}
11:14:12 MQT: stat/wemos4/POWER = OFF
11:14:14 CFG: Saved to flash at F7, Count 81, Bytes 4096
11:14:34 CMD: tmchatid
11:14:34 SRC: Serial
11:14:34 CMD: Group 0, Index 1, Command "TMCHATID", Data ""
11:14:34 MQT: stat/wemos4/RESULT = {"TmChatId":""}
```

fragolinux, 2020-10-17:
nothing here... did as you say (removed tmchatid and left only tmtoken, and just tmstate 3), i waited even more than 30 seconds, i can get it work only if i send the command just before a "getupdates" appears in web console... i'll go for nodered at this point, i can get faster and sure results and i can block access to unknown src chatids... thanks anyway :)

JackNJack, 2020-10-17:
> nothing here... did as you say (removed tmchatid and left only tmtoken, and just tmstate 3), i waited even more than 30 seconds, i can get it work only if i send the command just before a [...]

Maybe studying how telegram works might help you better understand your problem. You may also add log to enhance your debugging.

[25 more replies]