arendst/Tasmota #8622: IPaddressx changing does not change first octet

fixx-er, 2020-06-05:
### PROBLEM DESCRIPTION
Using 8.3.1 or 8.3.1(lite) and in console to change IPaddress, using the supplied commands, the first octet does not change from 0.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): __Gosund wp3___
- [x ] Tasmota binary firmware version number used: _8.3.1____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
13:53:04 CMD: IPAddress1=192.168.0.62
13:53:04 RSL: stat/tasmota_BC2CEA/RESULT = {"IPAddress1":"0.168.0.62 (192.168.0.104)"}
13:53:26 CMD: IPAddress1  = 192.168.0.62
13:53:26 RSL: stat/tasmota_BC2CEA/RESULT = {"IPAddress1":"0.168.0.62 (192.168.0.104)"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-06-05:
The command is NOT `IPAddress1=192.168.0.62` correct is `IPAddress1 192.168.0.62`

fixx-er, 2020-06-05:
yeah thanks. just figured this out and was going to delete my issue.  but figure i will leave it for others.

ascillato2, 2020-06-05:
Closing this issue as it is multi **duplicated**. Sorry.

Please, do a fast search in old issues and in the docs before opening an issue as requested in the template. Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-06-05 by ascillato2