arendst/Tasmota #8626: OLED I2C 128x64 SSD1306 does not ork, I2C address mismatch

ShaggyDog18, 2020-06-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Connecting I2C: SSD1306 with no success. 
I also have I2C: BME280 connected and functioning correctly.
I observe the following I2C address mismatch:  result of Display command:
```
22:32:18 CMD: display
22:32:18 RSL: stat/tasmota_CAF0B7/RESULT = {"Display":{"Model":2,"Width":128,"Height":64,"Mode":2,"Dimmer":15,"Size":1,"Font":1,"Rotate":0,"Refresh":2,"Cols":[16,8],"Rows":2}}
```
At start up of WiFi module, the I2C address found - **0x3d**:
```
00:00:00 CFG: Loaded from flash at F4, Count 6858
00:00:00 I2C: SSD1306 found at 0x3d
00:00:00 Project tasmota Tasmota Version 8.3.1(tasmota)-2_7_1
00:00:00 UPL: Arduino OTA Enabled Port 8266
00:00:00 I2C: BME280 found at 0x76
```
However, i2cscan command returns **0x3C** address:
```
22:30:40 CMD: i2cscan
22:30:40 RSL: stat/tasmota_CAF0B7/RESULT = {"I2CScan":"Device(s) found at 0x3c 0x76"}
```
Could that be the reason for the issue?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU v1 based on ESP-12F
- [x] Tasmota binary firmware version number used: 8.3.1(tasmota)-2_7_1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: 2_7_1
- [x] Flashing tools used: VSCode
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
23:17:31 CMD: Backlog Template
23:17:31 RSL: stat/tasmota_CAF0B7/RESULT = {"NAME":"Sensors","GPIO":[255,255,255,255,6,5,255,255,255,56,255,255,255],"FLAG":15,"BASE":18}
```
```
23:18:50 CMD: Module
23:18:50 RSL: stat/tasmota_CAF0B7/RESULT = {"Module":{"0":"Sensors"}}
```
```
23:20:06 CMD: GPIO 255
23:20:06 RSL: stat/tasmota_CAF0B7/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
  Configuration output here:

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
23:21:49 RSL: stat/tasmota_CAF0B7/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":73,"Free":438,"Rules":"on tele-BME280#Temperature do DisplayText [s1f0x0y0]Temp: %value% C endon"}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
23:23:12 RSL: stat/tasmota_CAF0B7/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_CAF0B7","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:23:12 RSL: stat/tasmota_CAF0B7/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:01:26","StartupUTC":"2020-06-05T19:21:46","Sleep":250,"CfgHolder":4617,"BootCount":103,"BCResetTime":"2020-03-26T17:44:25","SaveCount":6863,"SaveAddress":"F7000"}}
23:23:12 RSL: stat/tasmota_CAF0B7/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-06-05T15:58:57","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"378/699"}}
23:23:12 RSL: stat/tasmota_CAF0B7/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["e3200-2",""],"TelePeriod":60,"Resolution":"55C180C0","SetOption":["01008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
23:23:12 RSL: stat/tasmota_CAF0B7/STATUS4 = {"StatusMEM":{"ProgramSize":628,"Free":372,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8FF867B7","043681A1","202006CD","000013C0","C000F981","00000024"],"Drivers":"1,2,3,4,6,7,8,9,10,12,13,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,8,9,18,26"}}
23:23:12 RSL: stat/tasmota_CAF0B7/STATUS5 = {"StatusNET":{"Hostname":"Tasmota-Sensor","IPAddress":"192.168.1.132","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"18:FE:34:CA:F0:B7","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:23:12 RSL: stat/tasmota_CAF0B7/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CAF0B7","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:23:12 RSL: stat/tasmota_CAF0B7/STATUS7 = {"StatusTIM":{"UTC":"2020-06-05T20:23:12","Local":"2020-06-05T23:23:12","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+03:00","Sunrise":"06:48","Sunset":"22:48"}}
23:23:12 RSL: stat/tasmota_CAF0B7/STATUS10 = {"StatusSNS":{"Time":"2020-06-05T23:23:12","BME280":{"Temperature":20.4,"Humidity":63.7,"DewPoint":13.3,"Pressure":744.1},"PressureUnit":"mmHg","TempUnit":"C"}}
23:23:12 RSL: stat/tasmota_CAF0B7/STATUS11 = {"StatusSTS":{"Time":"2020-06-05T23:23:12","Uptime":"0T01:01:26","UptimeSec":3686,"Heap":24,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":6,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"e3200-2","BSSId":"88:C3:97:07:9D:0C","Channel":3,"RSSI":100,"Signal":-30,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
23:23:58 CMD: weblog 4
23:23:58 RSL: stat/tasmota_CAF0B7/RESULT = {"WebLog":4}
23:23:59 CFG: Saved to flash at F6, Count 6864, Bytes 4096
23:24:00 DBG: [Humidit 63.9%   ]
23:24:02 DBG: [Pressur 744.2mmH]
23:24:04 DBG: [[s1f0x0y0]Temp: ]

```

### TO REPRODUCE
_Steps to reproduce the behavior:_  run the following commands:
DisplayModel 2
I2cDriver4
DisplayMode 2

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Screen is off, nothing is on the display

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ShaggyDog18, 2020-06-06:
Executed 
`DisplayAddress 0x3C`
Restarted, no address discrepancy any more. Still display does not work.
Re-checked the display with Arduino Nano and Adafruit_GFX and Adafruit_SSD1306 libraries - display works fine...

Quest24, 2020-06-09:
Have the oled i2c SSD 1306 working with  with 8.3.1 on a sonoff  mini. But, it did not work properly on a Sonoff Basic.  The problem with the basic is that the digits in the display do not overwrite, but the old information remains on the display resulting in a cloud of pixels.
You have to select the correct type of GPIO for the SCL and SDA. For the Sonoff Mini I use GPIO 1 for SCL and 3 for SDA which works fine.

ShaggyDog18, 2020-06-09:
> Have the oled i2c SSD 1306 working with with 8.3.1 on a sonoff mini. But, it did not work properly on a Sonoff Basic. The problem with the basic is that the digits in the display do not overwrite, [...]

Thank you for suggestion. The case is for NodeMCU based on ESP-12F module... I2C is configured correctly. There is BME280 sensor running on the same I2C bus without any issue...

ShaggyDog18, 2020-06-09:
What a surprise:  Display is on as soon as I switch Toggle ON on the main screen!
Is there anything about switching a display on and off by the Toggle button in Tasmota documentation?
So, resolved now :-)

closed 2020-06-09 by ShaggyDog18

ShaggyDog18, 2020-06-09:
@Quest24, Try to run the following command to clean display:
`displaytext [z]`

Quest24, 2020-06-10:
Yes, thanks. But the funny thing is: the Sonoff-Mini works fine, the Sonoff-Basic gives problems with the same version 6.3.1 and same configuration.

ShaggyDog18, 2020-06-10:
@Quest24  I would strongly suggest upgrading to the recent  version [Tasmota 8.3.1](https://github.com/arendst/Tasmota/releases/tag/v8.3.1) :-)

Quest24, 2020-06-10:
I run 8.3.1

jacoherotel, 2021-06-04:
> What a surprise: Display is on as soon as I switch Toggle ON on the main screen!
> Is there anything about switching a display on/off by the Toggle button in Tasmota documentation?
> So, resolved now :-)

Thank you so much ! I was struggling 3 hours, couldn't understand why it worked yesterday.....

genjitr, 2022-10-19:
I just had a similar problem. I got a reading at the display as I set the DIMMER TO 100. One other thing to consider is to connect the display to the correct SDA and SCL pins of the microcontroller (see pinout). If the wrong pins are connected, the device may be shown on i2cscan but the display won't work.