arendst/Tasmota #8630: Suggest to reopen ticket: Support for Odroid GO because ESP32 ist supported now

hagenbuch, 2020-06-06:
**Have you looked for this feature in other issues and in the docs?**  
Yes, I found this post:
https://github.com/arendst/Tasmota/issues/6775

**Is your feature request related to a problem? Please describe.**  
I would humbly suggest that this issue be reopened, now as ESP32 support is there.

**Describe the solution you'd like**  
Reopen the ticket :)

**Describe alternatives you've considered**  
I could start a new post but then the old one woudl carry contradictory information.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-06-06:
Looking at the circuit diagramm https://github.com/hardkernel/ODROID-GO/blob/master/Documents/ODROID-GO_REV0.1_20180518.pdf
There is nothing special. Tasmota will run on the device.
Only the Display ILI9341 is not supported.

hagenbuch, 2020-06-06:
Thanks. I'll try!

Jason2866, 2020-06-06:
@hagenbuch could not resist... https://www.pollin.de/p/odroid-go-mobile-spielekonsole-kit-810887

hagenbuch, 2020-06-06:
I know, that's exactly what I ordered and bullt already.

Jason2866, 2020-06-06:
No, Tasmota writes a new partition table. Everything will be overwritten

ascillato2, 2020-06-09:
Hi, Any news on this?

Jason2866, 2020-06-09:
Hi Adrian, my Odroid Go is ordered. Should arrive the next days.

arendst, 2020-06-09:
I'd like to order one too but alas not possible to Die Niederlande:

> Es befinden sich derzeit Artikel in Ihrem Warenkorb, welche nicht in Ihr angegebenes Lieferland versendet werden können. Bitte klicken Sie oben "Zurück zum Shop" und löschen Sie die rot markierten Artikel aus Ihrem Einkaufswagen um Ihre Bestellung fortzusetzen.

I probably need to urge someone to send it to me...

hagenbuch, 2020-06-09:
Well I could have sent you one.. :) Shall I still?

I guess templates will be going here: https://github.com/blakadder/templates/tree/master/_templates

Maybe the screen is new and / or the speakers. Also not sure how Tasmota deals with SD-Cards, would be awesome.

arendst, 2020-06-09:
There is one in the mail already.

Templates should go there but for now it's only on the ESP32 doc page.

gemu2015, 2020-06-09:
@hagenbuch 
scripting has full support for sd cards (also on esp32) 
you can display the card directory  and may upload and download files and manipulate them in a script.

Jason2866, 2020-06-19:
Theo has fixed the bug in the Displaydriver.
You can try now with latest development version and this (not complete!) Template.
Display is working and can be dimmed via slider.
`{"NAME":"Odroid","GPIO":[32,1,1,1,1,768,1,1,1,1,416,1,1,1,736,672,0,800,1,704,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}`

Jason2866, 2020-07-18:
No close

Jason2866, 2020-08-15:
in work...

closed 2020-12-30 by github-actions[bot]