arendst/Tasmota #8631: rules changing status by itself and when using rule<x>

riker65, 2020-06-06:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

rules changing status by itself and when using rule<x> 
sometimes 
how to check status of rule without changing it  .
Rules which where 1 are off. Tried to understand how this could happen.
this happens on various devices.
I thought it happens during restart but could not log it

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): esp8266, esp01
- [x] Tasmota binary firmware version number used: 8.3.1.2 and 8.3.1, 
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
10:14:25 CMD: Backlog Template; Module; GPIO 255
10:14:25 MQT: stat/TA_WP_ESP0143/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
10:14:26 MQT: stat/TA_WP_ESP0143/RESULT = {"Module":{"0":"Generic"}}
10:14:26 MQT: stat/TA_WP_ESP0143/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"4":"DS18x20"},"GPIO2":{"30":"Relay2i"},"GPIO3":{"21":"Relay1"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
10:14:58 CMD: backlog rule1 ;rule2;rule3;
10:14:58 RUL: Rule1 on System#Boot do 	Backlog var9 57;teleperiod 15;publish tele/WP/%topic%/A RB %timestamp%; rule1 1;rule2 1;rule3 1  endon   on Power1#State=0 do 	 backlog publish tele/WP/%topic%/P1_0 P1_0-Stop;WebSend [192.168.1.206] POWER7 0 endon on Power1#State=0 do 	 backlog     tele/WP/%topic%/S400 %timestamp%;publish cmnd/TA_WP_H_P1_P2_Lolin1/ruletimer2 2; 	        publish cmnd/TA_WP_H_P1_P2_Lolin1/power2 0;WebSend [192.168.1.205] POWER2 0;     publish cmnd/TA_WP_H_P1_P2_Lolin1/power3 0;	WebSend [192.168.1.205]
10:14:58 RUL: Rule1 +POWER3 0; publish tele/WP/%topic%/S401 %timestamp% ;RuleTimer1 2600; publish tele/WP/%topic%/SSRH40Timer t1-H40-start;WebSend [192.168.1.206] POWER7 1 endon   on Rules#Timer=1 do 	backlog publish tele/WP/%topic%/SSRH40Timer  t1-H40-Stop ;power 0 endon
10:14:58 MQT: stat/TA_WP_ESP0143/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":758,"Free":150,"Rules":"on System#Boot do \tBacklog var9 57;teleperiod 15;publish tele/WP/%topic%/A RB %timestamp%; rule1 1;rule2 1;rule3 1  endon   on Power1#State=0 do \t backlog publish tele/WP/%topic%/P1_0 P1_0-Stop;WebSend [192.168.1.206] POWER7 0 endon on Power1#State=0 do \t backlog     tele/WP/%topic%/S400 %timestamp%;publish cmnd/TA_WP_H_P1_P2_Lolin1/ruletimer2 2; \t        publish cmnd/TA_WP_H_P1_P2_Lolin1/power2 0;WebSend [192.168.1.205] POWER2 0;     publish cmnd/TA_WP_H_P1_P2_Lolin1/power3 0;\tWebSend [192.168.1.205] POWER..."}
10:14:59 RUL: Rule2 on event#SOLL_SSRHSichT do 	backlog MEM9 %value%;publish tele/WP/%topic%/SSRHSichT STN %value%C-%timestamp%; endon  on Tele-DS18B20#Temperature do 	backlog VAR1 %value%; publish cmnd/TA_WP_XXX_Lol3/event REC_VAR1_T=%value%; publish cmnd/TA_WP_lol11/event REC_VAR1_T=%value% endon   on Tele-DS18B20#Temperature > %MEM9% do 	backlog publish tele/WP/%topic%/WPT_Sich_Hot  I%VAR1%S_M%MEM9% ; power1 0;publish cmnd/TA_WP_ESP0143/Power1 0;publish cmnd/TA_WP_H_P1_P2_Lolin1/Power3 0;publish
10:14:59 RUL: Rule2 +cmnd/TA_WP_H_P1_P2_Lolin1/Power2 0; endon  on Time#Minute do 	backlog publish tele/WP/%topic%/THSichm HSich-S%VAR1%-Max%MEM9%C-%timestamp%; endon
10:14:59 MQT: stat/TA_WP_ESP0143/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":629,"Free":184,"Rules":"on event#SOLL_SSRHSichT do \tbacklog MEM9 %value%;publish tele/WP/%topic%/SSRHSichT STN %value%C-%timestamp%; endon  on Tele-DS18B20#Temperature do \tbacklog VAR1 %value%; publish cmnd/TA_WP_XXX_Lol3/event REC_VAR1_T=%value%; publish cmnd/TA_WP_lol11/event REC_VAR1_T=%value% endon   on Tele-DS18B20#Temperature > %MEM9% do \tbacklog publish tele/WP/%topic%/WPT_Sich_Hot  I%VAR1%S_M%MEM9% ; power1 0;publish cmnd/TA_WP_ESP0143/Power1 0;publish cmnd/TA_WP_H_P1_P2_Lolin1/Power3 0;publish cmnd/TA_WP_H_P1_P2_Lolin1/Po..."}
10:14:59 MQT: stat/TA_WP_ESP0143/RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
10:15:00 RUL: TIME#MINUTE performs "backlog publish tele/WP/TA_WP_ESP0143/THSichm HSich-S17.1-Max55C-2020-06-06T10:15:00;"

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
10:15:53 CMD: status 0
10:15:53 MQT: stat/TA_WP_ESP0143/STATUS = {"Status":{"Module":0,"DeviceName":"TA_WP_ESP0143","FriendlyName":["TA_WP_ESP0143","Tasmota2"],"Topic":"TA_WP_ESP0143","ButtonTopic":"0","Power":0,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:15:53 MQT: stat/TA_WP_ESP0143/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:04","StartupUTC":"2020-06-06T08:09:49","Sleep":50,"CfgHolder":4617,"BootCount":167,"BCResetTime":"2020-04-12T09:38:08","SaveCount":1267,"SaveAddress":"FA000"}}
10:15:53 MQT: stat/TA_WP_ESP0143/STATUS2 = {"StatusFWR":{"Version":"8.3.1.2(e4e9a39-tasmota)","BuildDateTime":"2020-05-31T17:01:01","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"445/699"}}
10:15:53 MQT: stat/TA_WP_ESP0143/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":2,"SysLog":1,"LogHost":"192.168.0.22","LogPort":514,"SSId":["TR7272","TR_ESP"],"TelePeriod":15,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","000000C8","00000000"]}}
10:15:53 MQT: stat/TA_WP_ESP0143/STATUS4 = {"StatusMEM":{"ProgramSize":588,"Free":412,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","043683A1","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
10:15:53 MQT: stat/TA_WP_ESP0143/STATUS5 = {"StatusNET":{"Hostname":"TA_WP_ESP0143-2479","IPAddress":"192.168.1.197","Gateway":"192.168.0.31","Subnetmask":"255.255.240.0","DNSServer":"192.168.0.31","Mac":"EC:FA:BC:9B:E9:AF","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
10:15:53 MQT: stat/TA_WP_ESP0143/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.9","MqttPort":1893,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9BE9AF","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
10:15:53 MQT: stat/TA_WP_ESP0143/STATUS7 = {"StatusTIM":{"UTC":"2020-06-06T08:15:53","Local":"2020-06-06T10:15:53","StartDST":"2020-03-08T02:00:00","EndDST":"2020-11-01T02:00:00","Timezone":"+02:00","Sunrise":"05:48","Sunset":"21:49"}}
10:15:53 MQT: stat/TA_WP_ESP0143/STATUS10 = {"StatusSNS":{"Time":"2020-06-06T10:15:53","DS18B20":{"Id":"021313A9DEAA","Temperature":17.0},"TempUnit":"C"}}
10:15:53 MQT: stat/TA_WP_ESP0143/STATUS11 = {"StatusSTS":{"Time":"2020-06-06T10:15:53","Uptime":"0T00:06:04","UptimeSec":364,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"TR7272","BSSId":"46:A4:3C:DD:17:E1","Channel":1,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- do not know
[... cut ...]
when doing rule1 I expected to get the rule status not to change the status

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

tried to use syslog , but do not get a syslog created. 
added rule1 1 to ruleset 

rule1 
on System#Boot do
	Backlog var9 57;teleperiod 15;publish tele/WP/%topic%/A RB %timestamp%; rule1 1;rule2 1;rule3 1 
endon 

**(Please, remember to close the issue when the problem has been addressed)**

bikemike, 2020-06-07:
Are you saying that you have a rule that is on, and it somehow turned off on its own? If so, I just had this happen to me about a week ago with one of my devices and it started behaving strange and doing things I really didn't want it to do (opening and closing my garage door).

Rule1, Rule2, and Rule3 mysteriously turned off. These rules overrode the switch behavior of some sensors I had connected. Once the rules turned off the switch behavior went back to the standard mode of controlling relay1 which I was specifically trying to avoid (https://github.com/arendst/Tasmota/issues/5732#issuecomment-488246893).

riker65, 2020-06-07:
yes, and hard to identify  the real status  of a rule. because when checking with rule1 , looks like it is toggling.

even entering a new rule1 .....  sometimes status changes.

how to have the rules persistent?

bikemike, 2020-06-07:
It doesn't toggle when I check: 

08:14:47 CMD: rule1
08:14:47 MQT: sonoff-5657/stat/RESULT = {"Rule1":"ON",...
08:14:49 CMD: rule1
08:14:49 MQT: sonoff-5657/stat/RESULT = {"Rule1":"ON",...
08:14:54 CMD: rule1
08:14:54 MQT: sonoff-5657/stat/RESULT = {"Rule1":"ON",...

I'm using Tasmota 6.5.0.9

ascillato2, 2020-06-09:
Using command rule1 won't change its state. It will just display the rule.

If restarting your device, your rules turned off, that is because a bootloop protection. Some issue on power not being stable or a bad config that hungs Tasmota when it boots.

See commands SetOption65 and SetOption36.

Disable them for testing and please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-06-09 by ascillato2

riker65, 2020-06-09:
Hi, ok thanks for confirming.
Question: will rule `on system#boot rule1 1 endon` , be executed? 

thanks T

ascillato, 2020-06-09:
No because it has a syntax error. It is missing a DO

ascillato, 2020-06-09:
Anyway, check the setoptions of previous comments. You have another issue going on. Rules are not disabling itselves just because. There is another issue there

bikemike, 2020-06-09:
Perhaps a warning should be added to the rules page that explains why rules may become disabled without user intervention. That is the first place I looked when I realized my rules had become disabled.

ascillato, 2020-06-09:
Please, help us and do that note. You can click on the pencil (Top-right of the page) and edit the page.