arendst/Tasmota #8635: 8.3.1 does not switch between AP1 and AP2

raspitoaster, 2020-06-06:
### PROBLEM DESCRIPTION
I want to migrate all my Tasmota devices from one SSID to an other.
Configuring a second AP and switching over with "AP 2" in the console does not work on every device. All devices running Tasmota v8.3.1 do not switch over from current AP to the other AP.
Independent if the current one is AP1 or AP2, the devices always connect to the current AP configuration and do not switch (console AP 1 or AP 2) to the other config.
The same devices downgraded to Tasmota v8.1.0 behave as expected.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU Amica, Sonoff Basic R3, Gosund SP111
- [x] Tasmota binary firmware version number used: 8.3.1, 8.1.0 - both pre-compiled and self-compiled
  - [x] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: tasmocompiler
- [x] Flashing tools used: webif
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. flash Gosund SP111 with Tasmota 8.3.1
2. configure AP1 and AP2 - different SSIDs, different KEYS, where AP1 is the active connection
3. type "AP 2" in console
4. log window shows "CMD: AP 2"
5.  00:00:00 WIF: Connecting to AP2 xxxxxxxxxxxxxxxxxxxxx Channel 1 BSSId aa:bb:cc:dd:ee:ff in mode 11N as gosund-pow-001-0555...
   00:00:09 WIF: Connect failed as AP cannot be reached
   00:00:09 WIF: Connecting to AP1 yyyyyyyyyyyyyyyyyy in mode 11N as gosund-pow-001-0555...
   00:00:14 WIF: Connected
6. device always comes back with AP 1 configuration, switching to AP 2 is not possible
7. flash with Tasmota 8.1.0
8. after restart, device uses AP 2
9. switching between AP 1 and AP 2 is possible

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
6. device comes back with AP 2 configuration

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-06-07:
To produce the behaviour you want you need to config that using command WIFICONFIG.

If you want further assistance, please complete the troubleshooting template or address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

raspitoaster, 2020-06-07:
Thanks for your answer, but i cannot change the behaviour with WIFICONFIG.
The two wireless networks configured with AP1 and AP2 are up and running both at the same time.
With 8.1.0 i can switch between them by sending "AP 1" and "AP 2" in the console (with WIFICONFIG 2 and WIFICONFIG 4).
With 8.3.1 they stay in the current wireless network and never change to the other wireless network; "Connect failed as AP cannot be reached" is logged.
I also tried to exchange AP1 and AP2 to test if one of them is preferred/ignored, but the behaviour is the same: the devices stick to the current wireless network.

Jason2866, 2020-06-08:
Take a look at `SetOption56`

ascillato, 2020-06-09:
I could reproduce this on latest version. AP command is not switching over other wifi networks.
Seems that this bug is new. I used AP recently in few versions ago.

Thanks for reporting

closed 2020-06-09 by ascillato2

raspitoaster, 2020-06-09:
I just tried out the dev branch and can confirm that the fix solves the problem.
Thanks!