arendst/Tasmota #8636: Rule trigger on webbutton press

Kugeleis, 2020-06-07:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

Use case: A tasmota rule controls the relais depending on a sensor (typical thermostat usage) but the user wants to be able to control the relais from the WebUI, too. 
Currently the webUI button toggles the relais directly but the rule will act immediately against it.

This use case is similar to the [Simple Thermostat Example](https://tasmota.github.io/docs/Rules/#simple-thermostat-example) but instead of the physical button the webUI-button should be used.

Solution:
Like "SwitchMode1 3" decouples the physical button a "WebUIMode1 3" should do this for the web button.

closed 2020-07-11 by stale[bot]

Oderik, 2022-06-26:
I'd like to have some Web UI input options to control rules, too!

I was thinking of Web UI buttons that could just be enabled (and labeled just like the buttons that come along with switches). Pressing them would raise an event that can be processed with rules.

Also, a dropdown with configurable options would be nice.

anishsane, 2023-06-30:
I'd like this feature too.

If I have this, I can use one tasmota as the master remote control and use Websend to send commands to other tasmota devices.

barbudor, 2023-06-30:
You already have 3 ways of doing such things:
- virtual relay + rules
- Scripting-Language
- Berry (on ESP32 only)

anishsane, 2023-06-30:
How to add virtual relays?
I have exhausted (almost) all the GPIO pins for switches/relays (8+8).

I need a 9th web button that would websend commands to other tasmota.

Right now I am planning to use berry to emulate a button and for refreshing the value, use web_sensor + some local variables in berry.

However, the positioning of the buttons gets a bit weird, because it shows the status above the existing sensors like (ESP32 temperature) or webcam FPS.

barbudor, 2023-06-30:
A virtual relay is a relay on an unused gpio
If you are out of gpio then you should consider the other options
If you still have 3 gpios you can configure a dummy HC595 shift register which can give you 8, 16, 24 virtual relays