arendst/Tasmota #8637: Using VL53L0X with MQTT and Domoticz

pakosel, 2020-06-07:
### PROBLEM DESCRIPTION
I am using a number of ESP8266 devices flashed with Tasmota. All of them are reporting the state using MQTT and the state is stored is Domoticz. Recently tried to add a new ESP8266 device with VL53L0X sensor. Since the support of this sensor is not **built-in** I've used GitPod (specifically a ***TasmoCompiler***) to turn a support of the sensor by defining USE_VL53L0X flag:
```
#define USE_VL53L0X
```

The sensor works fine but for unknown reason the values are not sent to MQTT broker. Neither the Domoticz input topic ("domoticz/in"). Anyone else having the same issue with this sensor?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [x] Tasmota binary firmware version number used: 8.3.1.2
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmote firmware update
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [x] Provide the output of this command: `Status 0`:
```
15:09:56 CMD: status 0
15:09:56 MQT: stat/wemos/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"wemos","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:09:56 MQT: stat/wemos/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:10:30","StartupUTC":"2020-06-07T13:59:26","Sleep":50,"CfgHolder":4617,"BootCount":80,"BCResetTime":"2020-06-06T18:39:17","SaveCount":165,"SaveAddress":"3F9000"}}
15:09:56 MQT: stat/wemos/STATUS2 = {"StatusFWR":{"Version":"8.3.1.2(tasmota)","BuildDateTime":"2020-06-07T13:57:46","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"335/699"}}
15:09:56 MQT: stat/wemos/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Monolithos",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
15:09:56 MQT: stat/wemos/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":2568,"Heap":27,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":2,"Features":["00000809","8F986587","04260001","000000C1","050000C0","C000F981","00000024"],"Drivers":"1,2,4,6,7,8,9,10,18,19,22,24,26,27,29,30,35,37","Sensors":"1,2,4,5,6,45"}}
15:09:56 MQT: stat/wemos/STATUS5 = {"StatusNET":{"Hostname":"wemos-7434","IPAddress":"192.168.50.254","Gateway":"192.168.50.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"BC:DD:C2:24:BD:0A","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:09:56 MQT: stat/wemos/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.8","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_24BD0A","MqttUser":"pi","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:09:56 MQT: stat/wemos/STATUS7 = {"StatusTIM":{"UTC":"2020-06-07T14:09:56","Local":"2020-06-07T15:09:56","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:47","Sunset":"20:50"}}
15:09:56 MQT: stat/wemos/STATUS10 = {"StatusSNS":{"Time":"2020-06-07T15:09:56","VL53L0X":{"Distance":122}}}
15:09:56 MQT: stat/wemos/STATUS11 = {"StatusSTS":{"Time":"2020-06-07T15:09:56","Uptime":"0T00:10:30","UptimeSec":630,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{ ... }}}
```

### TO REPRODUCE
1. Compile latest source with uncommented `#define USE_VL53L0X`
2. Connect to VL53L0X sensor
3. Verify if the data is reported to MQTT broker with Domoticz IDX

### EXPECTED BEHAVIOUR
Report data to MQTT broker with Domoticz IDX

### SCREENSHOTS
Console Output:
```
00:00:00 CFG: Loaded from flash at 3FA, Count 156
00:00:00 Project tasmota Tasmota Version 8.3.1(tasmota)-2_6_1
00:00:00 I2C: VL53L0X found at 0x29
00:00:00 WIF: Connecting to AP1 Monolithos Channel 11 BSSId A8:5E:45:0E:A3:98 in mode 11N as wemos-7434...
00:00:02 WIF: Connected
00:00:02 HTP: Web server active on wemos-7434 with IP address 192.168.50.254
15:25:41 MQT: Attempting connection...
15:25:41 MQT: Connected
15:25:41 MQT: tele/wemos/LWT = Online (retained)
15:25:41 MQT: cmnd/wemos/POWER = 
15:25:41 MQT: tele/wemos/INFO1 = {"Module":"Generic","Version":"8.3.1(tasmota)","FallbackTopic":"cmnd/DVES_24BD0A_fb/","GroupTopic":"cmnd/tasmotas/"}
15:25:41 MQT: tele/wemos/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos-7434","IPAddress":"192.168.50.254"}
15:25:41 MQT: tele/wemos/INFO3 = {"RestartReason":"Software/System restart"}
15:25:45 MQT: tele/wemos/STATE = {"Time":"2020-06-07T15:25:45","Uptime":"0T00:00:10","UptimeSec":10,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{ ... }}
15:25:45 MQT: tele/wemos/SENSOR = {"Time":"2020-06-07T15:25:45","VL53L0X":{"Distance":86}}
```

pakosel, 2020-06-07:
It turns out that current implementation of VL53L0X sensor does not notify Domoticz at all. 
I've implement it based on BH1750 implementation and added as a pull request #8640. In my implementation the Domoticz IDX should be bound using ***Illuminance*** field which corresponds to the official documentation for this sensor in the [Tasmota docs](https://tasmota.github.io/docs/VL53L0x/): *After a reboot the driver will detect VL530L0x automatically and **display Illuminance**.*

pakosel, 2020-06-08:
@arendst - thanks! I will make further tests in the coming days. It might be useful to replace a constant (fixed) threshold with a percentage value but it requires some more testing to be sure

closed 2020-06-08 by pakosel

roger-, 2020-12-22:
@pakosel did you also need to use `#define USE_I2C` in Tasmocompiler? Built with just `#define USE_VL53L0X` but I2C seems to be missing (9.2.0).

pakosel, 2020-12-23:
@roger-  I don't see any reason why using I2C should be mandatory here. I haven't been using it either when working with the sensor so if it was undefined by default I was working in such a way. Are you trying to use more than only this sensor at a time?