arendst/Tasmota #8639: RF Birdge doesn't sniff command and reflash RF module

mastrikGit, 2020-06-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Sonoff RF Brindge with Portisch RF firmware doesn't output sniffed command to console. RF module is also not reflashing

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff RF Bridge R2
- [x ] Tasmota binary firmware version number used: 8.3.1 standart
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: Tasmotizer.py
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here: {"GPIO0":{"17":"Button1"},"GPIO1":{"148":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"149":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
17:16:18 RSL: stat/tasmota_FA83B7/STATUS = {"Status":{"Module":25,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_FA83B7","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:16:18 RSL: stat/tasmota_FA83B7/STATUS1 = {"StatusPRM":{"Baudrate":19200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T00:07:34","StartupUTC":"2020-06-07T16:08:44","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2020-06-07T17:01:04","SaveCount":27,"SaveAddress":"FA000"}}
17:16:18 RSL: stat/tasmota_FA83B7/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"342/699"}}
17:16:18 RSL: stat/tasmota_FA83B7/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jack_the_Bar_2G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
17:16:18 RSL: stat/tasmota_FA83B7/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
17:16:18 RSL: stat/tasmota_FA83B7/STATUS5 = {"StatusNET":{"Hostname":"tasmota_FA83B7-0951","IPAddress":"192.168.1.22","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:FA:83:B7","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:16:18 RSL: stat/tasmota_FA83B7/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FA83B7","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:16:18 RSL: stat/tasmota_FA83B7/STATUS7 = {"StatusTIM":{"UTC":"2020-06-07T16:16:18","Local":"2020-06-07T17:16:18","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:47","Sunset":"20:50"}}
17:16:18 RSL: stat/tasmota_FA83B7/STATUS10 = {"StatusSNS":{"Time":"2020-06-07T17:16:18"}}
17:16:18 RSL: stat/tasmota_FA83B7/STATUS11 = {"StatusSTS":{"Time":"2020-06-07T17:16:18","Uptime":"0T00:07:34","UptimeSec":454,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Jack_the_Bar_2G","BSSId":"D0:17:C2:B1:95:D8","Channel":3,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```

```

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
17:16:50 CMD: rfraw 177
17:16:50 SRC: WebConsole from 192.168.1.7
17:16:50 CMD: Group 0, Index 1, Command "RFRAW", Data "177"
17:16:50 RSL: stat/tasmota_FA83B7/RESULT = {"RfRaw":"ON"}
17:16:50 RSL: tele/tasmota_FA83B7/RESULT = {"Time":"2020-06-07T17:16:50","RfRaw":{"Data":"AAA055"}}

```
RF led is blinking when buttons are pressed on remote. But no output in console.

### TO REPRODUCE
_Steps to reproduce the behavior:_
flash 8.3.1 with tasmotizer.py
setup as (25) bridge
flash Portish

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Get sniffed data and have possibility to reflash rf module.

send data command outputs to console and rf led is flashing while sending.
```
17:36:36 CMD: Group 0, Index 1, Command "BACKLOG", Data "RfRaw AAB0210314016703F924180101011001100110010101100110011001010110010101100255; RfRaw 0"
17:36:36 SRC: Backlog
17:36:36 CMD: Group 0, Index 1, Command "RFRAW", Data "AAB0210314016703F924180101011001100110010101100110011001010110010101100255"
17:36:36 RSL: stat/tasmota_FA83B7/RESULT = {"RfRaw":"ON"}
17:36:37 SRC: Backlog
17:36:37 CMD: Group 0, Index 1, Command "RFRAW", Data "0"
17:36:37 RSL: stat/tasmota_FA83B7/RESULT = {"RfRaw":"OFF"}
```
### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
```
17:17:58 CMD: Group 0, Index 1, Command "RFRAW", Data "255"
17:17:58 RSL: stat/tasmota_FA83B7/RESULT = {"RfRaw":"ON"}
17:17:58 RSL: tele/tasmota_FA83B7/RESULT = {"Time":"2020-06-07T17:17:58","RfRaw":{"Data":"AA0255"}}
```
FW is latest from Tasmota GIT and any attempts to reflash fw results with 
"Upload Failed

Failed to erase RF chip"
```
Weblog for this:
17:19:26 CFG: Saved to flash at FB, Count 30, Bytes 4096
17:19:26 UPL: File RF-Bridge-EFM8BB1.hex ...
17:19:26 HTP: Upload done
17:19:39 WIF: Checking connection...
17:19:39 HTP: Main Menu
17:19:40 HTP: Console
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-06-08:
Cant reproduce. It seems your hardware is defect.

ascillato2, 2020-06-09:
Sorry, I can't reproduce it either.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. 

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-06-09 by ascillato2