arendst/Tasmota #8641: Generate random number feature

calcipher49, 2020-06-07:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
Goal is to wait random number of seconds between 2 commands like DELAY command

**Describe the solution you'd like**  
New RDELAY command, identical to DELAY, but with random delay between 0.2s and number specified in 0.1 seconds step
Alternate more general way can be new command to assign random number into variable

**Describe alternatives you've considered**  
Only existing solution found is Window option for timers which is too restrictive (only for timers and use minutes increment), 
 
**Additional context**  

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-06-07:
Why do you need that?
Which is the use case for that?

meingraham, 2020-06-07:
I'm sure the OP has something specific, but these occur to me off the top of my head... because I've needed these. I do the first one in openHAB but could do it all in Tasmota. The second one is a nut I have not cracked yet.

1. A timer that turns a light on and off randomly for when one is away from home so it's not a fixed cycle that gives away that it's an automation. A set of rules with random time can be used to cycle the light to be on for varying (random) lengths, to be off for a random.e amount of time before turning it back on again. Add to this to not start before sunset, etc.

This one is more of a stretch... but to get the brainstorming going...
2. A cycle for a diffuser do something similar to spread out the amount of time it's on over a set time. Say you want to diffuse 10 minutes and hour over the course of 12 hours. You'd need the random cycle but also a way to accumulate total time and total on time. Basically spread out 120 minutes of running but keep track of how much of the 12 hours remains and how many more minutes of on time remains so you can catch up toward the end of the cycle.

calcipher49, 2020-06-07:
Hi
My first usage is to avoid trigger different identical devices at exactly same time.
Need it also to simulate random duration for device activation

Also random number generator can be useful for many other cases... for example random color light for led strip.

hagenbuch, 2020-06-08:
> Why do you need that?
> Which is the use case for that?

I would use it in a "vacancy switch" in order to make potential burglars believe I am there and have no regular pattern ob being at home etc.

arendst, 2020-06-09:
See my answer here https://github.com/arendst/Tasmota/issues/8643#issuecomment-641113116

closed 2020-06-09 by ascillato2