arendst/Tasmota #8643: SONOFF 4CH R2 Timer

jimebner, 2020-06-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I am using the timer function in a SONOFF 4CH R2 device flashed with Tasmota 8.3.1. All is good except that I cannot get the timer function on the device to do what I expect.

The timer schedule works fine when I manually set it up using the module UI. However, I would like to enable/disable the timer using MQTT commands from the Node Red dashboard. I can send commands and they get accepted OK. So I thought that using the "ARM" function would enable/disable the pre-set schedule for that particular timer. The arm button is set/cleared ok in the module but the timer does not fire when the arm is set when the current time is within the timer window. The timer does fire when the arm is set outside the window and the current time then moves into the window.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ Y] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ Y] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ Y] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [Y ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ Y] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ Y] Device used (e.g., Sonoff Basic): 4CH R2
- [ Y] Tasmota binary firmware version number used: __8.3.1___
  - [ Y] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ Y] Flashing tools used: __Tasmotizer___
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The power switch does not adopt the ON state, i.e. the timer does not trigger when the current time is within the window of the timer. Same applies to all 12 timer slots.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-06-08:
As designed.

The timer ONLY triggers at the requested time as that's the time the timer is scheduled to act. It checks armed timers at the time scheduled.

jimebner, 2020-06-08:
Thanks Theo. 
Do you think you might alter the design to trigger when armed if the current time is within the scheduled time?

Jim Ebner

arendst, 2020-06-08:
A timer is a one-shot action within 24 hours.

A single timer cannot know (easily) that a second timer is related to it providing the window you talk about. 

So I see no easy fix.

ascillato, 2020-06-08:
Tasmota Timers are **_One-Shot timers_**. When the Pre-Set time is reached, an action is executed.

The timer type you want is an **_INTERVAL TIMER_**. When the time is between 2 pre-set Times, an action is executed.

The use of both types are for different automations types.

In the case of the second, you can achieve that type by using rules.

For example, If you want to control your Outside Lights by SunSet and SunRise, and make them controlled by an interval timer, you can use Tasmota normal ONE-SHOT timers for turning them ON and OFF, and also add rules at boot time for the case, your device is powered up AFTER any of the ONE-SHOT timers. 

What I use for my Outside Lights:

TIMERS SET to (turn on light on sunset) (turn off light on sunrise)
```lua
poweronstate 0
rule1 1
rule1 on time#initialized do backlog event checksunrise=%time%; event checksunset=%time% endon on event#checksunset>%sunset% do power1 1 endon on event#checksunrise<%sunrise% do power1 1 endon
```

So, using Timers+Rules you achieve Interval Timers in Tasmota.

ascillato, 2020-06-08:
At this moment, Timers are like the following screenshot:

[image]

At this moment, Interval Timers can be achieved in TASMOTA using rules, but would be useful to add something like the following?

[image]

So, when marking the INTERVAL checkbox, it could show 2 times for setting the start and the end of an interval where that action should be triggered.

[image]

So, if Tasmota boots up and syncs the time, and the time is between that interval, the PRE-SET action is executed.
This could be used for lights (as the previous comment example) or for triggering rules or states of a Thermostat.

In the use-case of the outside lights, we were using 2 interval-timers. One for Turning the lights ON and the other for turning them OFF.

What do you think?

ascillato, 2020-06-08:
Besides, this can have a RANDOMIZE interval option, similar to the request on https://github.com/arendst/Tasmota/issues/8641

jimebner, 2020-06-09:
Thank you for your feedback. I can see now that the current design is one shot timers.

This interval timer is exactly the function I was thinking of. 

So I need to look at implementing this in a rule - I am not yet familiar with how to create nor implement these, so need to experiment. I was using Node Red to send MQTT commends to the timer to arm/disarm.

Instead of sunrise and sunset, presumably we could use any 2 fixed times; begin time set in timer 1 for on, then end time in timer 2 for off ?

arendst, 2020-06-09:
> Instead of sunrise and sunset, presumably we could use any 2 fixed times; begin time set in timer 1 for on, then end time in timer 2 for off ?

Correct. That's interval as planned using less code as possible. The new interval suggestion is way to code hungry (it needs time and random window testing of start and end time) to be integrated so we keep it as it is.

Regarding randimization that has always been there too. By setting a minute value in the +/- field  you enable randomization within -minutes to +minutes of trigger time. So the current implementation allows a random window of 30 minutes.

closed 2020-06-09 by ascillato2

jimebner, 2020-06-12:
Hello 

I have been trying to make this work using your suggestion to use rules.

Unfortunately I cannot do an operation on the start time which is stored as hh:mm, unlike %time% and %sunrise% which are stored as “minutes elapsed since midnight.”

Do you have any suggestions as to how to store the start times of the sonoff timer in tasmota?

Many thanks.

ascillato, 2020-06-12:
> I have been trying to make this work using your suggestion to use rules.

For using an interval timer, you need to use timers AND rules like in the comment:

https://github.com/arendst/Tasmota/issues/8643#issuecomment-640850996

jimebner, 2020-06-12:
My email was not clear apologies.

I have been using TIMERS and RULES.

I want to compare %TIME% with the value of Time stored in e.g. Timer1.

%TIME% is e.g. 920 minutes after midnight, but Time is e.g. 18.30.

The question is: How to compare these different quantities to decide on the action?

ascillato, 2020-06-12:
> I want to compare %TIME% with the value of Time stored in e.g. Timer1.

Not possible. Not implemented.

dangtrungluong, 2025-07-13:
how to add and build INTERVAL?

[image]

sfromis, 2025-07-13:
Using two timers, one for "on" and the other for "off", you'll get an interval.

dangtrungluong, 2025-07-14:
timers1 18:00 => on
timers2 06:00 => off
This only true in case tasmota is running before 18h and before 06h. 
I want tasmota to start after the set timers run properly