arendst/Tasmota #8650: Tasmota IR, errors in fujistu timings

JoachimF, 2020-06-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Using the tasmota-ir firmware to control my fujistu HVAC, I found that it can't decode fuji remote.
I tried on two ESP and multiples remote. Then used debug mode to read the timing and found that timing was not correct.
I changed the timing, and now tasmota is able de decode the frame

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_
Is it normal or only from my french remotes ?
- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
Tasmota_ir 8.3.1
ESP8266 generic module or Sonoff T1 1CH
Fujitsu remote AR-REB1E

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just try to receive Fujitsu Model 1 or 3 HVAC frame

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
frame decoded

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Correction in "ir_Fujitsu.cpp" 
const uint16_t kFujitsuAcBitMark = 394; //was 448
const uint16_t kFujitsuAcZeroSpace = 432; // was 390

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-06-09:
Hi,

Thanks for reporting.

This is an issue of the IR library. Can you post this at https://github.com/crankyoldgit/IRremoteESP8266

When this change is merged into the library, we can update the library.

Thanks.

closed 2020-06-09 by ascillato2

JoachimF, 2020-06-09:
I'll do that,

Thanks