arendst/Tasmota #8651: Devicename and friendly name and other text info overwritten

Quest24, 2020-06-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Devicename and friendly name and other text info overwritten after a few days of running.
Names of the Device name is overwritten by the preset "Tasmota". The device name itself can be blank or also Tasmota.   In the main menu, the word <Toggle1> is also changed by  "Tasmotax"

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff Mini
- [ ] Tasmota binary firmware version number used: 2.6.3
  - [ ] Pre-compiled
  - [ ] Self-compiled Yes
    - [ ] IDE / Compiler used: Arduino 1.8.12
- [ ] Flashing tools used: Arduino 1.8.12
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

15:09:24 MQT: stat/Sonoff-Badkamer/RESULT = {"NAME":"","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
15:09:25 MQT: stat/Sonoff-Badkamer/RESULT = {"Module":{"18":"Generic"}}
15:09:25 MQT: stat/Sonoff-Badkamer/RESULT = {"GPIO0":{"2":"AM2301"},"GPIO1":{"5":"I2C SCL"},"GPIO2":{"0":"None"},"GPIO3":{"6":"I2C SDA"},"GPIO4":{"10":"Switch2"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"22":"Relay2"},"GPIO16":{"9":"Switch1"}}

Configuration output here
Program Version	8.3.1(tasmota)
Build Date & Time	2020-05-22T21:34:03
Core/SDK Version	2_7_1/2.2.2-dev(38a443e)
Uptime	0T00:30:58
Flash write Count	1383 at 0xF4000
Boot Count	6
Restart Reason	Software/System restart
Friendly Name 1	Sonoff-Badkamer
Friendly Name 2	Sonoff-Badkamer
Friendly Name 3	Sonoff-Badkamer
 
AP1 SSId (RSSI)	xxxxxxxxxxxx (62%, -69 dBm)
Hostname	Sonoff-Badkamer
IP Address	192.168.178.141
Gateway	192.168.178.1
Subnet Mask	255.255.255.0
DNS Server	84.116.46.20
MAC Address	60:01:94:FB:3B:43
 
MQTT Host	192.168.178.63
MQTT Port	1883
MQTT User	mqtt
MQTT Client	DVES_FB3B43
MQTT Topic	Sonoff-Badkamer
MQTT Group Topic 1	cmnd/tasmotas/
MQTT Full Topic	cmnd/Sonoff-Badkamer/
MQTT Fallback Topic	cmnd/DVES_FB3B43_fb/
 
Emulation	Disabled
mDNS Discovery	Disabled
 
ESP Chip Id	16464707
Flash Chip Id	0x144051
Flash Size	1024kB
Program Flash Size	1024kB
Program Size	552kB
Free Program Space	384kB
Free Memory	24k  

- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:

  Rules output here:
Rule1
on switch2#state=1 do
	add1 1
endon  
on switch2#state=1 do
	power2 1
endon  
on switch2#state=0 do
	power2 0 
endon  
on var1#state!=0 do
	backlog delay 6;var1 0
endon  
on var1#state==2 do
	power1 1
endon

rule2 
ON mqtt#connected DO Subscribe test2, cmd/Sonoff-Badkamer/MEM2 ENDON     ON Event#test2 DO backlog MEM2=%value%; displaytext [a2p6x2y24f2]%value% Lt ENDON   On tele-AM2301#Temperature do
	 displaytext [a2p6x4y48f1]%value% C
endon   
on tele-AM2301#Humidity do
	DisplayText [a2p6x60y48f1]%value% %%
endon

rule3
ON mqtt#connected DO backlog displayfont 2; displaymode 0; displaytext [a2zr126:64] ;Subscribe test, cmd/Sonoff-Badkamer/MEM1 ENDON    ON Event#test DO backlog MEM1=%value%; displaytext [a2p6x2y2f2]%value% M  ENDON

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
15:15:43 CMD: status 0
15:15:43 MQT: stat/Sonoff-Badkamer/STATUS = {"Status":{"Module":18,"DeviceName":"Sonoff-Badkamer","FriendlyName":["Sonoff-Badkamer","Sonoff-Badkamer","Sonoff-Badkamer"],"Topic":"Sonoff-Badkamer","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,2,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:15:43 MQT: stat/Sonoff-Badkamer/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:16:05","StartupUTC":"2020-06-09T13:59:38","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2020-06-04T08:27:47","SaveCount":1383,"SaveAddress":"F4000"}}
15:15:43 MQT: stat/Sonoff-Badkamer/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-22T21:34:03","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"525/699"}}
15:15:43 MQT: stat/Sonoff-Badkamer/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Broersen2.4",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
15:15:43 MQT: stat/Sonoff-Badkamer/STATUS4 = {"StatusMEM":{"ProgramSize":552,"Free":384,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","09702596","00008120","22040484","000000C0","00000800","00000000"],"Drivers":"1,2,3,4,7,9,10,12,13,16,29","Sensors":"2,3,6,9,15,22,26"}}
15:15:43 MQT: stat/Sonoff-Badkamer/STATUS5 = {"StatusNET":{"Hostname":"Sonoff-Badkamer","IPAddress":"192.168.178.141","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"84.116.46.20","Mac":"60:01:94:FB:3B:43","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:15:43 MQT: stat/Sonoff-Badkamer/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.63","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FB3B43","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:15:43 MQT: stat/Sonoff-Badkamer/STATUS7 = {"StatusTIM":{"UTC":"2020-06-09T14:15:43","Local":"2020-06-09T15:15:43","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00"}}
15:15:43 MQT: stat/Sonoff-Badkamer/STATUS10 = {"StatusSNS":{"Time":"2020-06-09T15:15:43","Switch1":"ON","Switch2":"OFF","AM2301":{"Temperature":20.1,"Humidity":45.0,"DewPoint":7.8},"TempUnit":"C"}}
15:15:43 MQT: stat/Sonoff-Badkamer/STATUS11 = {"StatusSTS":{"Time":"2020-06-09T15:15:43","Uptime":"0T00:16:05","UptimeSec":965,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"Broersen2.4","BSSId":"AC:22:05:27:AC:8E","Channel":11,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:04"}}}
15:15

- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

not available

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

It happens after some time. Don't have any clues

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-06-09:
Hi,

Seems that you had an event of inestable power, and this event have triggered the bootloop protection or the [fast power cycle detection](https://tasmota.github.io/docs/Device-Recovery/#fast-power-cycle-device-recovery).

See commands `SetOption65` and `SetOption36` in the Tasmota Docs ( https://tasmota.github.io/docs/Commands/ )

You can disable those if you have power issues.

Issue duplicated of https://github.com/arendst/Tasmota/issues/8414, https://github.com/arendst/Tasmota/issues/8328, https://github.com/arendst/Tasmota/issues/7735, https://github.com/arendst/Tasmota/issues/7384, https://github.com/arendst/Tasmota/issues/8631, https://github.com/arendst/Tasmota/issues/8303, https://github.com/arendst/Tasmota/issues/7740, https://github.com/arendst/Tasmota/issues/6966,

closed 2020-06-09 by ascillato2

Quest24, 2020-06-10:
To be clear. I did not lose my configuration. All configuration information is still there. The only thing is the Device name is gone and the text in the buttons is overwritten. The device is still working properly.

arendst, 2020-06-10:
For debugging purposes you might want to try and change your rule use of MEM to a VAR. Every MEM writes to flash while a VAR only writes to RAM.

MEM is written in the same area as strings which you observe as changed. There might be an issue with simultanous writing in this area. By changing MEM to VAR we can rule out that MEM is causing issues.

Pls report back if you re-encounter this issue.

arendst, 2020-06-10:
As an alternative you might want to try the latest dev version which attempts to finish string writes before it starts to read. In that case you must keep your MEM as is.

Again pls report back if you re-encounter this issue.