arendst/Tasmota #8654: Wrong brightness in HA with Auto discovery of independant RGB and WW + relay channel

septillion-git, 2020-06-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have a generic ESP8266 flashed with Tasmota which is connected to 4 MOSFETs (as PWM channel) and 1 (solid state) relay. PWM channels are set up independent with SO37 = 128 and autodiscovery in Home Assistant.

Brightness of the RGB isn't display correct in HA (100% all the time). Brightness of White displayed as brightness of RGB. Both correct in WebUI and console.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Generic ESP8266 (WeMos D1 Mini)
- [x] Tasmota binary firmware version number used: 8.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: iTead OTA and Tasmota OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:40:50 MQT: Hall Light short/stat/RESULT = {"NAME":"Hall Light","GPIO":[255,255,255,255,37,38,255,255,40,21,39,255,255],"FLAG":15,"BASE":18}
21:40:50 MQT: Hall Light short/stat/RESULT = {"Module":{"0":"Hall Light"}}
21:40:51 MQT: Hall Light short/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"37":"PWM1"},"GPIO5":{"38":"PWM2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"40":"PWM4"},"GPIO13":{"21":"Relay1"},"GPIO14":{"39":"PWM3"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:43:17 MQT: Hall Light short/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Hall Light short","FriendlyName":["Spot","RGB","WW"],"Topic":"%hostname%","ButtonTopic":"0","Power":6,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:43:17 MQT: Hall Light short/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"1T01:15:32","StartupUTC":"2020-06-08T19:27:45","Sleep":50,"CfgHolder":4617,"BootCount":504,"BCResetTime":"2020-06-04T16:16:05","SaveCount":1074,"SaveAddress":"FA000"}}
21:43:17 MQT: Hall Light short/stat/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-05-18T15:39:35","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"395/699"}}
21:43:17 MQT: Hall Light short/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["UniFy NoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["400A8009","2805C8000180060000005A00000000000000","00000208","00000000"]}}
21:43:17 MQT: Hall Light short/stat/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":3,"Features":["00000809","8FDAE797","043683A0","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
21:43:17 MQT: Hall Light short/stat/STATUS5 = {"StatusNET":{"Hostname":"Hall Light short","IPAddress":"192.168.47.77","Gateway":"192.168.47.1","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"60:01:94:70:3B:93","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:43:17 MQT: Hall Light short/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.47.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_703B93","MqttUser":"mqtt","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:43:17 MQT: Hall Light short/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-06-09T20:43:17","Local":"2020-06-09T21:43:17","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:47","Sunset":"20:51"}}
21:43:17 MQT: Hall Light short/stat/STATUS10 = {"StatusSNS":{"Time":"2020-06-09T21:43:17"}}
21:43:17 MQT: Hall Light short/stat/STATUS11 = {"StatusSTS":{"Time":"2020-06-09T21:43:17","Uptime":"1T01:15:32","UptimeSec":90932,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":2,"POWER1":"OFF","POWER2":"ON","Dimmer2":23,"POWER3":"ON","Dimmer3":46,"Color":"59,12,0,117","HSBColor":"12,100,23","White":46,"Channel":[23,5,0,46],"Scheme":0,"Fade":"ON","Speed":2,"LedTable":"ON","Wifi":{"AP":1,"SSId":"UniFy NoT","BSSId":"B6:FB:E4:F7:E3:60","Channel":6,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
- Set up a Tasmota with 1 relay and 4 PWM
- Set independant led channels (SetOption37 = 128)
- Relay discovered as relay or light (SetOption30) doesn't matter
- Autodiscover for HA on (SetOption19 = 1)
- Change / view brightness of the RGB and the WW channels in HA

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect the values in HA to match the values in WebUI and Console.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Problem is in the STATE-json does not match the discovery-JSON. It also does not match the documentation.

tele/STATE response
```
{"Time":"2020-06-09T21:58:06","Uptime":"1T01:30:21","UptimeSec":91821,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":2,"POWER1":"OFF","POWER2":"ON","Dimmer2":23,"POWER3":"ON","Dimmer3":23,"Color":"59,12,0,59","HSBColor":"12,100,23","White":23,"Channel":[23,5,0,23],"Scheme":0,"Fade":"ON","Speed":2,"LedTable":"ON","Wifi":{"AP":1,"SSId":"UniFy NoT","BSSId":"B6:FB:E4:F7:E3:60","Channel":6,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:04"}}
```
Discovery JSON for RGB (703B93_LI_2 as 703B93_LI_1 is the relay)
```
{"name":"Hall Light short RGB","stat_t":"Hall Light short/tele/STATE","avty_t":"Hall Light short/tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","cmd_t":"Hall Light short/cmnd/POWER2","val_tpl":"{{value_json.POWER2}}","pl_off":"OFF","pl_on":"ON","uniq_id":"703B93_LI_2","dev":{"ids":["703B93"]},"bri_cmd_t":"Hall Light short/cmnd/Dimmer1","bri_stat_t":"Hall Light short/tele/STATE","bri_scl":100,"on_cmd_type":"brightness","bri_val_tpl":"{{value_json.Dimmer1}}","rgb_cmd_t":"Hall Light short/cmnd/Color2","rgb_stat_t":"Hall Light short/tele/STATE","rgb_val_tpl":"{{value_json.Color.split(',')[0:3]|join(',')}}","fx_cmd_t":"Hall Light short/cmnd/Scheme","fx_stat_t":"Hall Light short/tele/STATE","fx_val_tpl":"{{value_json.Scheme}}","fx_list":["0","1","2","3","4"]}
```

The status uses dimmer2 dimmer3 to report the brightness of RGB and WW respectively. But the discovery expects them to be dimmer1 and dimmer2. Last is also in agreement with the documentation of dimmer<x>. Although you could argue it would make sense to have power<x> and dimmer<x> match.

Trouble now is, dimmer1 does not exist so HA assumes 100%. Tasmota uses dimmer2 to report the brightness of RGB but HA uses it for WW brightness.

**Bonus** if SetOption37 > 128, can the whole white channel be skipped in the discovery? Now it's (or should be if it worked correctly) just a mirror of the brightness. It's just redundant.

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-06-09:
Thanks for reporting. I'll look on it ASAP.

septillion-git, 2020-06-09:
Thank you very much! 😄

closed 2020-06-11 by arendst

septillion-git, 2020-09-01:
@effelle Is this already pushed to 8.4? Because I don't see it in the change log although it's marked fixed and pulled.

And is it a known problem you can't only change RGB with SO37>=128 via commands or MQTT? Color will always set W-channel to 0. If it's not yet on the radar I'll make an issue but I don't want to spam you with duplicate issues although I can't find it.

PR #8667 (2020-06-11): Fix Dimmer tele inconsistency when SO37 128
## Description:

Fix Dimmer number inconsistency when reporting teleperiod and `SO37 128`.

**Related issue (if applicable):** fixes #8654 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core ESP8266 V.2.7.1
  - [x] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license- [... cut]