arendst/Tasmota #8655: Option to disable state reporting for discovered devices (Home Assistant)

macrojames, 2020-06-10:
**Have you looked for this feature in other issues and in the docs?** 
Sure did, even in the code

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
The problem is sensors showing state=unavailable in home-assistant if I use deepsleep.
I understand the lwt should be offline to show the actual state. Home Assistant with the auto discovery of mqtt devices does not support overriding this behaviour besides the config topic.
(https://www.home-assistant.io/integrations/binary_sensor.mqtt/#configuration)

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Let me set an option to remove the availability topic (avty_t) from HASS_DISCOVER_BASE.
That is the recommended (and tested) way to keep the state after entering deep sleep and ignoring the state. 
I can set that configuration manually and it works until the next time the device set the config topic.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Manually configuring the mqtt devices, not recommened on Home Assistants side, GUI and integration workflows are preferred. It also doesn't scale and interferes with the charming auto discovery for device groups.

**Additional context**  
_Add any other context or screenshots about the feature request here._
[image]
You can see the time when i enabled DeepSleepTime 60

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-06-11:
@grizewald

The PR 7231 was done because, as it was before, if you restart your device, OR if you change the topic (that will make Tasmota to restart) it will remain connected to the mqtt, while the reality was incorrect. So you could end up with a lot of old devices being online.

About DeepSleep, it needs some fixes as @effelle have proposed. If you have a relay in a deepsleep device, It should be shown as offline while it is not available. That is why it is here the request to make a DeepSleep sensor to not be shown as OFFLINE or ONLINE. That is easy to implement on the Automatic Discovery.

macrojames, 2020-06-12:
Maybe I failed to explain my proposed solution correctly.
I wouldn't change the LWT topic or it's behaviour. It's just the field avty_t in the hass discovery config I am concerned about. If you omit that field, Home Assistant will keep the analog/temp/humidity/... sensor value with its last state. The online state of the device can still be tracked on the lwt topic and sensor.
But I like the conditions for read-only sensors @effelle proposed.

PR #8724 (2020-06-18): Discovery Update
## Description:
Changelog:
- Add a new definition, `DEEPSLEEP_LWT_HA_DISCOVERY`, to enable LWT topic and its payloads for read-only sensors  and binary_sensors on HAss Discovery. When commented out (default) all read-only sensors and binary_sensors won't be shown as `OFFLINE` on Home Assistant when the device is DeepSleeping
**NOTE**: This is only for read-only sensors and binary_sensors, relays/lights and status sensor will be shown as `OFFLINE`);
- Add support for Tuya and Sonoff MCU relays/lights;
- Some more sensors added.

**Related issue (if applicable):** fixes #8655

# [... cut]
closed 2020-06-18 by arendst