arendst/Tasmota #8662: OpenTherm status	FAULT (0x0)

yann1420, 2020-06-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I compiled tasmota 8.3.1 code and configured according instructions https://tasmota.github.io/docs/OpenTherm/ for a Ihor Melnyk adaptor (http://ihormelnyk.com/opentherm_adapter) adapter. Hooked up an Intergas Prestige boiler (https://www.intergas-verwarming.nl/en/trade/products/prestige/prestige-cw6/) on pins 11&12 of the X4 connector .

**when connecting I get these type of errors
_11:25:41 [OTH]: perform handshake
11:25:42 [OTH]: getSlaveConfiguration failed. Status=TIMEOUT_**

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [* ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [* ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [* ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [* ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [* ] Device used (e.g., Sonoff Basic): _wemos mini____
- [* ] Tasmota binary firmware version number used: _8.3.1. ____
  - [ ] Pre-compiled
  - [* ] Self-compiled
    - [* ] IDE / Compiler used: _gitpod____
- [ *] Flashing tools used: __wemos___
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
11:30:03 [OTH]: perform handshake
11:30:04 [OTH]: getSlaveConfiguration failed. Status=TIMEOUT
11:30:12 CMD: Backlog Template; Module; GPIO 255`
11:30:12 SRC: WebConsole from 192.168.178.70
11:30:12 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255`"
11:30:12 SRC: Backlog
11:30:12 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
11:30:12 MQT: stat/tasmota-3805/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:30:12 SRC: Backlog
11:30:12 CMD: Group 0, Index 1, Command "MODULE", Data ""
11:30:12 MQT: stat/tasmota-3805/RESULT = {"Module":{"18":"Generic"}}
11:30:13 SRC: Backlog
11:30:13 CMD: Group 0, Index 1, Command "GPIO", Data "255`"

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
11:31:59 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
11:31:59 SRC: Backlog
11:31:59 CMD: Group 0, Index 1, Command "RULE", Data ""
11:31:59 MQT: stat/tasmota-3805/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
11:31:59 SRC: Backlog
11:31:59 CMD: Group 0, Index 2, Command "RULE", Data ""
11:31:59 MQT: stat/tasmota-3805/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
11:32:00 SRC: Backlog
11:32:00 CMD: Group 0, Index 3, Command "RULE", Data ""
11:32:00 MQT: stat/tasmota-3805/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
11:32:00 [OTH]: perform handshake
11:32:01 [OTH]: getSlaveConfiguration failed. Status=TIMEOUT

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
11:33:01 CMD: status 0
11:33:01 SRC: WebConsole from 192.168.178.70
11:33:01 CMD: Group 0, Index 1, Command "STATUS", Data "0"
11:33:01 MQT: stat/tasmota-3805/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota-3805","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:33:01 MQT: stat/tasmota-3805/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Exception","Uptime":"1T15:07:56","StartupUTC":"2020-06-09T19:25:05","Sleep":50,"CfgHolder":4617,"BootCount":36,"BCResetTime":"2020-06-05T14:44:07","SaveCount":57,"SaveAddress":"F9000"}}
11:33:01 MQT: stat/tasmota-3805/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-06-05T07:34:26","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"364/699"}}
11:33:01 MQT: stat/tasmota-3805/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"MqttLog":2,"SysLog":0,"LogHost":"syslog","LogPort":514,"SSId":["titania","openwrt"],"TelePeriod":20,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
11:33:01 MQT: stat/tasmota-3805/STATUS4 = {"StatusMEM":{"ProgramSize":602,"Free":400,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashMode":3,"Features":["00000809","8FDAE797","043683A1","000000CD","010013C0","C000F981","00000224"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,69"}}
11:33:01 MQT: stat/tasmota-3805/STATUS5 = {"StatusNET":{"Hostname":"tasmota-3805-3805","IPAddress":"192.168.178.81","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.60","Mac":"80:7D:3A:78:EE:DD","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:33:01 MQT: stat/tasmota-3805/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_78EEDD","MqttUser":"schepers","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
[... 31 lines cut ...]
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
**The adapter communicates with the boiler given the following response:**
11:39:41 CMD: backlog ot_flags ;ot_tboiler ;ot_twater ;ot_save_setpoints ;ot_ch 
11:39:41 SRC: WebConsole from 192.168.178.70
11:39:41 CMD: Group 0, Index 1, Command "BACKLOG", Data "ot_flags ;ot_tboiler ;ot_twater ;ot_save_setpoints ;ot_ch "
11:39:41 SRC: Backlog
11:39:41 CMD: Group 0, Index 1, Command "OT_FLAGS", Data ""
11:39:42 SRC: Backlog
11:39:42 CMD: Group 0, Index 1, Command "OT_TBOILER", Data ""
11:39:42 MQT: stat/tasmota-3805/RESULT = {"ot_tboiler":60.0}
11:39:42 SRC: Backlog
11:39:42 CMD: Group 0, Index 1, Command "OT_TWATER", Data ""
11:39:42 MQT: stat/tasmota-3805/RESULT = {"ot_twater":45.0}
11:39:42 SRC: Backlog
11:39:42 CMD: Group 0, Index 1, Command "OT_SAVE_SETPOINTS", Data ""
11:39:42 MQT: stat/tasmota-3805/RESULT = {"ot_save_setpoints":"Done"}
11:39:42 SRC: Backlog
11:39:42 CMD: Group 0, Index 1, Command "OT_CH", Data ""
11:39:42 MQT: stat/tasmota-3805/RESULT = {"ot_ch":0}
11:39:50 [OTH]: perform handshake

**(Please, remember to close the issue when the problem has been addressed)**

yury-sannikov, 2020-06-28:
hi @yann1420 can you please provide more information on your hardware and how it's connected? Which GPIO pins do you use to connect the adapter to the board? Which voltage do you have on the OpenTherm line when you connect the adapter to the boiler?
Also, you may use https://github.com/ihormelnyk/opentherm_library/blob/master/examples/OpenThermMaster_Demo/OpenThermMaster_Demo.ino to make sure your master (your adapter) is able to connect to the slave (boiler) and check that your hardware part is 100% operational.

yury-sannikov, 2020-06-28:
I'm using Wemos D1 R2 and this is my config:
[image]

yury-sannikov, 2020-06-28:
Can you also copy the output of `GPIO 255`?
I don't see anything like `"GPIO0":{"203":"OpenTherm TX"}` in your configuration output above.
```
MQT: stat/boiler/RESULT = {"GPIO0":{"203":"OpenTherm TX"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"202":"OpenTherm RX"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}```

yann1420, 2020-06-29:
Hello Yuriy, Per your request :

12:25:05 CMD: GPIO 255
12:25:05 SRC: WebConsole from 192.168.178.70
12:25:05 CMD: Group 0, Index 1, Command "GPIO", Data "255"
12:25:05 MQT: stat/tasmota-3805/RESULT = {"GPIO0":{"205":"OpenTherm TX"},"GPIO1":{"0":"None"},"GPIO2":{"4":"DS18x20"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"204":"OpenTherm RX"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
12:25:05 [OTH]: perform handshake
12:25:07 [OTH]: Processing response. Status=TIMEOUT, Response=0x0
12:25:07 [OTH]: getSlaveConfiguration failed. Status=TIMEOUT
[image]

I see I have a different "generic" module then you have (generic-18 vs generic-0). Can that make the difference?

yury-sannikov, 2020-06-29:
The configuration looks good. I would suggest playing with the author's https://github.com/ihormelnyk/opentherm_library/blob/master/examples/OpenThermMaster_Demo/OpenThermMaster_Demo.ino to make sure the hardware and wiring part is ok.

yury-sannikov, 2020-06-29:
Also, if you connect an adapter to the boiler the OT line voltage should read something between 15-18 volts (24..48 volts from the boiler if it's open-ended). If you put 3.3v between IN pin (near R4) and GND the voltage on the OT line should drop to something between 5 and 7 volts. That way you can check the adapter -> boiler hardware part.

yury-sannikov, 2020-07-21:
@yann1420 any luck with troubleshooting it?

yann1420, 2020-08-19:
Hello Yury, I have been out of the (home) office the last couple of weeks and will now be on holiday for 3 weeks.
My goal is to respond the second week of September.

closed 2020-09-20 by stale[bot]

mr-varga, 2020-11-03:
Have you solve the problem? I have the same...

santik, 2021-05-19:
I have the same

yury-sannikov, 2021-05-21:
hi @santik, I know you're working with Igor on this issue. Could you please update the thread when you figure this out?

yann1420, 2021-11-03:
Hello Yury,
I assume you have better understanding of OT or at least of the code that has been used here and therefore I have questions that you can answer:
- From the OT v2.2 specification (found on the web) I read there are plenty more functions.  For instance, exported as **OT18** is the "**CH water pressure**". And **OT19** is the  **DHW flow rate** but appears not to be exported. Is it because of the code or the implementation of my CH-boiler that I see only a subset?
- There is a field called **RAW** , what does this represent?
- What is the difference between the command **ot_ch** and **ot_flags ch**

Thanks in advance, Yann

malinduta, 2022-01-22:
Hello @yury-sannikov,
I am also interested on the informations from the post above.
Can you provide more information?

krzakx, 2022-10-03:
Hello.
I've tried pretty sure every combination and still got TIMEOUT (like open ends?)
Having ~17V on Boiler Terminals while everything wired up.

My Boiler has OpenTherm version 4 (double confirmed with support). Could this be an issue?

yann1420, 2022-10-04:
OpenTherm v4 ?  AFAIK, V3 is the latest.
The voltage is DC or AC ?

krzakx, 2022-10-04:
Yes it is v4.
In general is blurry, but you can find device on market that does support OP**+** (yes, plus) v4.
Additionaly please take look at this URL https://www.opentherm.eu/document/opentherm-protocol-specifications-v4-0/ it's empty if opened but the addres speaks for itself.

The voltage on Boiler terminals is DC. Wire connections really looks ok. From soft configuration I've checked every combinantion. I am looking for seting-up another new ESP8266+OT Board set....

krzakx, 2022-10-07:
Side question.
Do you have room regulator/room thermostat communicating in OT? if yes, than is Wemos between somehow as gateway?

In my case I do not posses any room regulator either vendor specific either potential free relay.

yann1420, 2022-10-08:
I am using [Home Assistant ](https://www.home-assistant.io/) (HA) that amongst others uses [Tasmota](https://tasmota.github.io/docs/) as a protocol to communicate with IoT devices. One of them is a Wemos D1 Mini (ESP8622) that is hooked up to a [Ihor Melnyk OpenTherm adapter](http://ihormelnyk.com/).
The temperature reading comes from a little [Xiaomi BLE](https://nl.aliexpress.com/item/4000528071010.html?spm=a2g0o.productlist.0.0.20251d3fgJyDgp&algo_pvid=9f128ca8-1356-4ccc-9c7f-7a1ff004fd62&algo_exp_id=9f128ca8-1356-4ccc-9c7f-7a1ff004fd62-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000028197988292%22%7D&pdp_npi=2%40dis%21EUR%219.63%216.74%21%21%21%21%21%402100bdcf16652565570623618efb25%2112000028197988292%21sea&curPageLogUid=dzgSy6sNIL6Q) thermometer of which the signal is going to HA. HA has on its turn various types software thermostats that controls the Wemos.
Is this what you are asking?

yann1420, 2022-10-08:
> Yes it is v4. In general is blurry, but you can find device on market that does support OP**+** (yes, plus) v4. Additionaly please take look at this URL [...]

In the meanwhile I updated my small amount of OT knowledge. Indeed there is a V4 that has some extra ID's (commands) and more clarity on the protocol.

krzakx, 2022-10-10:
> Is this what you are asking?
Yes, this is what I was asking for as I was wondering if that I am lacking any OT Vendor specific thermostat is an my issues but NO that is not the case.

The news is that with ESPHome OT specific config burnt to my Wemos it work out.... Unfortunately I have no clue why I failed with Tasmota.

PS. Can you please share source for OT v4. knowledge?

yann1420, 2022-10-10:
I understand that you killed your Wemos using ESPhome and that you want to use Tasmota instead. Correct?

My OT knowledge comes from an old pal who worked as a programmer for boiler producer and in that time he had access to the info. And AFAIK there is no Opentherm information public available (apart from V2.2 )

krzakx, 2022-10-11:
Oh.... no no, by "burnt" I meant flashed ESPHome specific configured binary file into Wemos ;).
Regrads

yann1420, 2022-10-11:
OK, that is another explanation of the meaning of **burnt**.  :) 
Would you mind sharing your YAML, I would like to compare but do not want to reflash the MQTT Wemos at the beginning of the winter season.

krzakx, 2022-10-12:
Config that I've use comes from this project:
https://github.com/rsciriano/ESPHome-OpenTherm/blob/main/opentherm.yaml

yann1420, 2022-10-12:
> Config that I've use comes from this project: [ESPHome-OpentTherm](https://github.com/rsciriano/ESPHome-OpenTherm/blob/main/opentherm.yaml) Note: this is YAML confi file for ESPHome core component **not** for HA configuration.yaml

thnx

serhat-ozkara, 2022-10-31:
I also have the same issue with my ESP32 running 12.2.0