arendst/Tasmota #8663: Support for 2 channel Tuya MCU dimmer

blackscreener, 2020-06-11:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
Default implementation of tuya mcu api in xdrv_16_tuyamcu.ino support only dimmer with one channel. Also we need two brightness sliders in web interface, 

**Describe the solution you'd like**  
Support for two brightness sliders in web interface when fnId: 22 (2nd dimmer) is define with non zero dpId. I tried to add TUYA_MCU_FUNC_DIMMER2 to xdrv_16_tuyamcu.ino but workaround with two sliders from PWM dimmer sends dim command to 2 channels. Also input from button send on/off state to 2 channels. Likely bad executecommand in:
`if (fnId == TUYA_MCU_FUNC_DIMMER2) { //2 channel
          AddLog_P2(LOG_LEVEL_DEBUG, PSTR("TYA: 2CH RX Dim State=%d"), packetValue);
          Tuya.new_dim2 = changeUIntScale(packetValue, 0, Settings.dimmer_hw_max, 0, 100);
          if (Tuya.ignore_dimmer_cmd_timeout2 < millis()) {
            if ((power || Settings.flag3.tuya_apply_o20) &&  // SetOption54 - Apply SetOption20 settings to Tuya device
                (Tuya.new_dim2 > 0) && (abs(Tuya.new_dim2 - Settings.light_dimmer) > 1)) {
              Tuya.ignore_dim2 = true;
              snprintf_P(scmnd, sizeof(scmnd), PSTR(D_CMND_DIMMER "3 %d"), Tuya.new_dim2 );
              ExecuteCommand(scmnd, SRC_SWITCH);
            }
          }`

**Describe alternatives you've considered**  

**Additional context**  
DpID's for MS-105B: 2 Gang Dimmer Switch:
dpID 1 - on, off 1 ch
dpID 2 - dimmer 1 ch
dpID 7 - on, off 2 ch
dpID 8 - dimmer 2 ch

Logs from @LukevdPalen:
ACTION | Frame Header | Version | Command Word | Data Length | dpID | Data Type | Function Length | Function Command | Verification Method
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
1 ON | 55aa | 00 | 06 | 0005 | 01 | 01 | 0001 | 01 | 0e
1 OFF | 55aa | 00 | 06 | 0005 | 01 | 01 | 0001 | 00 | 0d
1 dim10 | 55aa | 00 | 06 | 0008 | 02 | 02 | 0004 | 00000065 | 7a
1 dim20 | 55aa | 00 | 06 | 0008 | 02 | 02 | 0004 | 000000c3 | d8
1 dim30 | 55aa | 00 | 06 | 0008 | 02 | 02 | 0004 | 00000127 | 3d
1 dim40 | 55aa | 00 | 06 | 0008 | 02 | 02 | 0004 | 0000018b | a1
1 dim50 | 55aa | 00 | 06 | 0008 | 02 | 02 | 0004 | 000001ef | 05
1 dim60 | 55aa | 00 | 06 | 0008 | 02 | 02 | 0004 | 0000025b | 72
1 dim70 | 55aa | 00 | 06 | 0008 | 02 | 02 | 0004 | 000002be | d5
1 dim80 | 55aa | 00 | 06 | 0008 | 02 | 02 | 0004 | 0000031d | 35
1 dim90 | 55aa | 00 | 06 | 0008 | 02 | 02 | 0004 | 00000381 | 99
1 dim100 | 55aa | 00 | 06 | 0008 | 02 | 02 | 0004 | 000003e8 | 00
2 ON | 55aa | 00 | 06 | 0005 | 07 | 01 | 0001 | 01 | 14
2 OFF | 55aa | 00 | 06 | 0005 | 07 | 01 | 0001 | 00 | 13
2 dim10 | 55aa | 00 | 06 | 0008 | 08 | 02 | 0004 | 00000061 | 7c
2 dim20 | 55aa | 00 | 06 | 0008 | 08 | 02 | 0004 | 000000c3 | de
2 dim30 | 55aa | 00 | 06 | 0008 | 08 | 02 | 0004 | 00000129 | 45
2 dim40 | 55aa | 00 | 06 | 0008 | 08 | 02 | 0004 | 0000018d | a9
2 dim50 | 55aa | 00 | 06 | 0008 | 08 | 02 | 0004 | 000001ef | 0b
2 dim60 | 55aa | 00 | 06 | 0008 | 08 | 02 | 0004 | 00000254 | 71
2 dim70 | 55aa | 00 | 06 | 0008 | 08 | 02 | 0004 | 000002b9 | d6
2 dim80 | 55aa | 00 | 06 | 0008 | 08 | 02 | 0004 | 0000031e | 3c
2 dim90 | 55aa | 00 | 06 | 0008 | 08 | 02 | 0004 | 00000380 | 9e
2 dim100 | 55aa | 00 | 06 | 0008 | 08 | 02 | 0004 | 000003e8 | 06

[xdrv_16_tuyamcu.zip](https://github.com/arendst/Tasmota/files/4764542/xdrv_16_tuyamcu.zip)

diasarmando, 2020-07-06:
There is any version with this request included?

effelle, 2020-08-29:
Hold on.

closed 2020-09-17 by arendst