arendst/Tasmota #8664: Home Assistant MQTT Discovery (SetOption19)

stingrayuk, 2020-06-11:
### PROBLEM DESCRIPTION
Using Tasmota 8.3.1 and the SetOption19 the 8.3.1 does not create the MQTT device correctly in home assistant.

Using 8.2 Tasmota created a named MQTT device based on the friendly name with the correct entities based on the friendly name defined. For example if Friendly Name is "Joe Bedside Light" Home Assistant creates
light.joe_bedside_light

Using 8.3.1 Tasmota (with both Friendly Name and Device Name set) created a named MQTT device based on device name with the switch entity based on a combination of the device name followed by the friendly name. For example if friendly name is "fn Joe Bedside Light" and device name is "dn Joe Bedside Light" Home Assistant creates
light.dn_joe_bedside_light_fn_joe_bedside_light

From the documentation I understood that Device Name only should be used by HA auto discovery and that its default value is friendly name

From https://tasmota.github.io/docs/Commands/#devicename
DeviceName | Device name displayed in the webUI and used for HA autodiscovery.<value> = set device name (default = FriendlyName1 value)
-- | --

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic R3, Gosund UP111
- [x] Tasmota binary firmware version number used: 8.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Web OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
Sonoff Basic R3
15:47:37 CMD: Backlog Template; Module; GPIO 255
15:47:37 MQT: joe/bedside/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
15:47:38 MQT: joe/bedside/stat/RESULT = {"Module":{"1":"Sonoff Basic"}}
15:47:38 MQT: joe/bedside/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

Gosund UP111
15:48:45 CMD: Backlog Template; Module; GPIO 255
15:48:45 MQT: conservatory/flyzapper/stat/RESULT = {"NAME":"Gosund UP111","GPIO":[0,52,0,17,134,132,0,0,131,157,21,0,0],"FLAG":0,"BASE":18}
15:48:45 MQT: conservatory/flyzapper/stat/RESULT = {"Module":{"0":"Gosund UP111"}}
15:48:46 MQT: conservatory/flyzapper/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"52":"Led1"},"GPIO2":{"0":"None"},"GPIO3":{"17":"Button1"},"GPIO4":{"134":"BL0937 CF"},"GPIO5":{"132":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"131":"HLWBL SELi"},"GPIO13":{"157":"LedLink"},"GPIO14":{"21":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here: N/A

```
- [x] Provide the output of this command: `Status 0`:
```
Sonoff Basic R3
15:50:03 CMD: status0
15:50:03 MQT: joe/bedside/stat/STATUS = {"Status":{"Module":1,"DeviceName":"dn Joe Bedside Light","FriendlyName":["fn Joe Bedside Light"],"Topic":"joe/bedside","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

Gosund UP111
15:50:46 CMD: status0
15:50:46 MQT: conservatory/flyzapper/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Conservatory Fly Zapper","FriendlyName":["Conservatory Fly Zapper"],"Topic":"conservatory/flyzapper","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
Console Output when SetOption19 1 is issued
Sonoff Basic R3
15:55:16 CMD: setoption19 1
15:55:16 SRC: WebConsole from 192.168.55.192
15:55:16 CMD: Group 0, Index 19, Command "SETOPTION", Data "1"
15:55:16 MQT: joe/bedside/stat/RESULT = {"SetOption19":"ON"}
15:55:17 MQT: homeassistant/device_automation/703C50_SW_1_TOGGLE/config =  (retained)
[... 61 lines cut ...]
15:53:21 MQT: homeassistant/switch/250542_RL_6/config =  (retained)
15:53:21 MQT: homeassistant/light/250542_LI_7/config =  (retained)
15:53:21 MQT: homeassistant/switch/250542_RL_7/config =  (retained)
15:53:21 MQT: homeassistant/light/250542_LI_8/config =  (retained)
15:53:21 MQT: homeassistant/switch/250542_RL_8/config =  (retained)
15:53:21 MQT: homeassistant/sensor/250542_status/config = {"name":"Conservatory Fly Zapper status","stat_t":"conservatory/flyzapper/tele/HASS_STATE","avty_t":"conservatory/flyzapper/tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attr_t":"conservatory/flyzapper/tele/HASS_STATE","unit_of_meas":"%","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"250542_status","dev":{"ids":["250542"],"name":"Conservatory Fly Zapper","mdl":"Gosund UP111","sw":"8.3.1(tasmota)","mf":"Tasmota"}} (retained)

```

### TO REPRODUCE
SetOption19 0
Set Friendly Name
Set Device Name
SetOption19 1

### EXPECTED BEHAVIOUR
Home Assistant should create a MQTT device based on the device name

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2020-06-11:
You did not understand correctly. 

Please read the explanation https://tasmota.github.io/docs/Home-Assistant/#automatic-discovery and note there has been a breaking change in 8.3

effelle, 2020-06-11:
As designed. Device name defaults to friendly name IF you don't set it. Meaning HA will have always two strings to deal with.
So if you call your relay `fn Joe Bedside Light` and use as device name `dn Joe Bedside Light` Home Assistant will use the combination Device Name + Friendly Name + Entity type. 
More information on how a device name is built [here](https://tasmota.github.io/docs/Home-Assistant/#finalising-setup).

closed 2020-06-11 by effelle