arendst/Tasmota #8668: Problems with pull request #7231

grizewald, 2020-06-11:
### PROBLEM DESCRIPTION
Pull request [#7231](https://github.com/arendst/Tasmota/pull/7231) which is currently in the development branch and release 8.3.1, breaks deep sleep sensors for Home Assistant by marking the sensor as "Unavailable" as soon as it sleeps. The way the changes in #7231 have been implemented means they cannot be turned off. (Which is far from the pull request's description of being optional).

A Will message is for use on abnormal disconnects only! The disconnect command is allowed to specify a reason for the disconnect and if sent with 4 as the status code, will cause the server to send the Will. As the PubSubClient library does not support parameters to the disconnect method, you could always just close the WiFi connection before restarting when the MQTT settings have changed rather than cleanly disconnecting. This would have the same effect as disconnecting with status code 4.
Unconditionally changing the way the MQTT protocol works at the level of the PubSubClient library is a bad way to solve the original problem.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1
- [x] Tasmota binary firmware version number used: 3.8.1 + 3.8.1.2
  - [x] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: Arduino IDE 1.8.12
- [x] Flashing tools used: Arduino / esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

N/A
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

N/A
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

N/A

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Connect a sensor device using deep sleep. Wait until it sleeps.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Sensor remains "Online" as far as Home Assistant is concerned.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-06-11:
Duplicated Request

See the open issue where this is being discussed and worked. https://github.com/arendst/Tasmota/issues/8655

effelle, 2020-06-11:
Let's discuss on the other topic then.

closed 2020-06-11 by effelle