arendst/Tasmota #8670: Rotary encoder support for H801 LED WiFi controller

ximonline, 2020-06-12:
**Have you looked for this feature in other issues and in the docs?**  
Yes, there no feature "rotary encoder" exept xiaomi lamp.
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No problem.
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Wifi LED RGBWW module H801 supported by Tasmota (device #20). 5 pwm channels has level from 0 to 100 (I used all 5 channels to white stripes). It will be better adjust dimmer level with rotary encoder like KY-040. H801 has three free pins (gpio0, gpio2, gpio3).
Scenario:
- on "rotary encoder click" switch off LED/switch on LED (with level before off).
- on "rotate right" level up "dimmer level",
- on "rotate left" level down "dimmer level".

P.S. I found in Tasmota code module "ROTARY_V1", but only for Xiaomy lamp. I think, it will be suitable for H802 too.
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-06-12:
Duplicated of https://github.com/arendst/Tasmota/issues/8263

closed 2020-06-12 by ascillato2

ximonline, 2020-06-13:
> Duplicated of #8263

My feature request is not duplicate. In #8263 requested some function improvement for xiaomi mi desk lamp. But feature from this lamp is not available for other devices. I said about it in my post. Please,  open my issue again.

arendst, 2020-06-30:
The latest rotary encoder implementation supports:

- Dimming
- When pressed and configured as button1 controls colortemp on ColdWarm lights
- When pressed and configured as button1 controls color on RGB lights