arendst/Tasmota #8673: PowerOnState ON not working on Sonoff D1 Dimmer

digitlength, 2020-06-13:
### PROBLEM DESCRIPTION
I have a Sonoff D1 Dimmer flashed with Tasmota wired in to some ceiling downlights. Power to the Sonoff is via a light switch.

Flipping the light switch off removes power from the Sonoff. Flipping the light switch on restores power to the Sonoff.

I have set `PowerOnState ON` via the console, and confirmed that the change has stuck. However, when removing power, then restoring power, the lights remain off.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff D1 Dimmer
- [ ] Tasmota binary firmware version number used: Tasmota 8.3.1.2
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: espEasy
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

08:08:01 MQT: stat/swUpstairsHallwayLight/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
08:08:01 MQT: stat/swUpstairsHallwayLight/RESULT = {"Module":{"74":"Sonoff D1"}}
08:08:01 MQT: stat/swUpstairsHallwayLight/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"148":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"149":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

  No rules used
```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

08:09:48 MQT: stat/swUpstairsHallwayLight/STATUS = {"Status":{"Module":74,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"swUpstairsHallwayLight","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:09:48 MQT: stat/swUpstairsHallwayLight/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T00:14:26","StartupUTC":"2020-06-13T06:55:22","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2020-06-07T10:02:52","SaveCount":176,"SaveAddress":"F4000"}}
08:09:48 MQT: stat/swUpstairsHallwayLight/STATUS2 = {"StatusFWR":{"Version":"8.3.1.2(54ab264-tasmota)","BuildDateTime":"2020-06-06T21:02:22","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"364/699"}}
08:09:48 MQT: stat/swUpstairsHallwayLight/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ACPRO1",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
08:09:48 MQT: stat/swUpstairsHallwayLight/STATUS4 = {"StatusMEM":{"ProgramSize":588,"Free":412,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","043683A1","000000CD","010013C0","C000F981","00000024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
08:09:48 MQT: stat/swUpstairsHallwayLight/STATUS5 = {"StatusNET":{"Hostname":"swUpstairsHallwayLight-2866","IPAddress":"192.168.1.123","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"C8:2B:96:5A:2B:32","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:09:48 MQT: stat/swUpstairsHallwayLight/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.92","MqttPort":1883,"MqttClientMask":"swUpstairsHallwayLight","MqttClient":"swUpstairsHallwayLight","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:09:48 MQT: stat/swUpstairsHallwayLight/STATUS7 = {"StatusTIM":{"UTC":"2020-06-13T07:09:48","Local":"2020-06-13T08:09:48","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:54"}}
08:09:48 MQT: stat/swUpstairsHallwayLight/STATUS10 = {"StatusSNS":{"Time":"2020-06-13T08:09:48"}}
08:09:48 MQT: stat/swUpstairsHallwayLight/STATUS11 = {"StatusSTS":{"Time":"2020-06-13T08:09:48","Uptime":"0T00:14:26","UptimeSec":866,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ACPRO1","BSSId":"E0:63:DA:65:EA:4A","Channel":6,"RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```

### TO REPRODUCE
Set `PowerOnState` to `ON`
Remove power to the device.
Restore power to the device

### EXPECTED BEHAVIOUR
After restoring power, I would expect the lights to turn on.

ascillato, 2020-06-13:
Hi, please, set weblog to 4 and provide the console output when you restart your device.

digitlength, 2020-06-13:
Setting weblog:

```
13:48:02 CMD: weblog 4
13:48:02 MQT: stat/swUpstairsHallwayLight/RESULT = {"WebLog":4}
13:48:02 CFG: Saved to flash at F4, Count 184, Bytes 4096
13:48:08 WIF: Checking connection...
13:48:28 WIF: Checking connection...
13:48:48 WIF: Checking connection...
13:49:08 WIF: Checking connection...
```

Re-powering:

```
00:00:00 CFG: Loaded from flash at F4, Count 184
00:00:00 QPC: Flag 7E
00:00:00 CFG: CR 364/699
00:00:00 SER: Set to 8N1 9600 bit/s
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.3.1.2(54ab264-tasmota)-2_7_1
00:00:00 CFG: Saved to flash at FB, Count 185, Bytes 4096
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 ACPRO1 Channel 6 BSSId E0:63:DA:65:EA:4A in mode 11N as swUpstairsHallwayLight-2866...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on swUpstairsHallwayLight-2866 with IP address 192.168.1.123
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/swUpstairsHallwayLight/LWT = Online (retained)
00:00:05 MQT: cmnd/swUpstairsHallwayLight/POWER = 
00:00:05 MQT: Subscribe to cmnd/swUpstairsHallwayLight/#
00:00:05 MQT: Subscribe to cmnd/tasmotas/#
00:00:05 MQT: Subscribe to cmnd/swUpstairsHallwayLight_fb/#
00:00:05 MQT: tele/swUpstairsHallwayLight/INFO1 = {"Module":"Sonoff D1","Version":"8.3.1.2(54ab264-tasmota)","FallbackTopic":"cmnd/swUpstairsHallwayLight_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:05 MQT: tele/swUpstairsHallwayLight/INFO2 = {"WebServerMode":"Admin","Hostname":"swUpstairsHallwayLight-2866","IPAddress":"192.168.1.123"}
00:00:05 MQT: tele/swUpstairsHallwayLight/INFO3 = {"RestartReason":"Power On"}
00:00:05 MQT: stat/swUpstairsHallwayLight/RESULT = {"POWER":"ON"}
00:00:05 MQT: stat/swUpstairsHallwayLight/POWER = ON
00:00:06 QPC: Reset
00:00:08 APP: Boot Count 15
00:00:08 CFG: Saved to flash at FA, Count 186, Bytes 4096
00:00:09 MQT: tele/swUpstairsHallwayLight/STATE = {"Time":"1970-01-01T00:00:09","Uptime":"0T00:00:11","UptimeSec":11,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":75,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ACPRO1","BSSId":"E0:63:DA:65:EA:4A","Channel":6,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:05"}}
00:00:23 WIF: Checking connection...

```

ascillato, 2020-06-13:
Ok, from your logs it can be seen that Tasmota is sending the command `power on` to the dimming MCU, but seems that the mcu haven't finished booting at that moment.

Until this is resolved, you can use the following rule as a workaround:

```lua
Rule1 1 
Rule1 on mqtt#connected do power1 1 endon
```

Please, report back if the rule makes your device to start with the power on.

digitlength, 2020-06-13:
OK, supplying that rule:

```
18:36:35 CMD: rule1
18:36:35 SRC: WebConsole from 192.168.1.73
18:36:35 CMD: Group 0, Index 1, Command "RULE", Data ""
18:36:35 MQT: stat/swUpstairsHallwayLight/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":35,"Free":476,"Rules":"on mqtt#connected do power1 1 endon"}
```

Then switching off and back on again:

```
00:00:00 CFG: Loaded from flash at F7, Count 205
00:00:00 QPC: Flag 7E
00:00:00 CFG: CR 364/699
00:00:00 SER: Set to 8N1 9600 bit/s
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.3.1.2(54ab264-tasmota)-2_7_1
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 ACPRO1 Channel 6 BSSId E0:63:DA:65:EA:4A in mode 11N as swUpstairsHallwayLight-2866...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on swUpstairsHallwayLight-2866 with IP address 192.168.1.123
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/swUpstairsHallwayLight/LWT = Online (retained)
00:00:05 MQT: cmnd/swUpstairsHallwayLight/POWER = 
00:00:05 MQT: Subscribe to cmnd/swUpstairsHallwayLight/#
00:00:05 MQT: Subscribe to cmnd/tasmotas/#
00:00:05 MQT: Subscribe to cmnd/swUpstairsHallwayLight_fb/#
00:00:05 MQT: tele/swUpstairsHallwayLight/INFO1 = {"Module":"Sonoff D1","Version":"8.3.1.2(54ab264-tasmota)","FallbackTopic":"cmnd/swUpstairsHallwayLight_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:05 MQT: tele/swUpstairsHallwayLight/INFO2 = {"WebServerMode":"Admin","Hostname":"swUpstairsHallwayLight-2866","IPAddress":"192.168.1.123"}
00:00:05 MQT: tele/swUpstairsHallwayLight/INFO3 = {"RestartReason":"Power On"}
00:00:05 MQT: stat/swUpstairsHallwayLight/RESULT = {"POWER":"ON"}
00:00:05 MQT: stat/swUpstairsHallwayLight/POWER = ON
00:00:05 RUL: MQTT#CONNECTED performs "power1 1"
00:00:05 SRC: Rule
00:00:05 CMD: Group 0, Index 1, Command "POWER", Data "1"
00:00:05 MQT: stat/swUpstairsHallwayLight/RESULT = {"POWER":"ON"}
00:00:05 MQT: stat/swUpstairsHallwayLight/POWER = ON
00:00:06 QPC: Reset
00:00:08 APP: Boot Count 18
00:00:08 CFG: Saved to flash at F6, Count 206, Bytes 4096
00:00:09 MQT: tele/swUpstairsHallwayLight/STATE = {"Time":"1970-01-01T00:00:09","Uptime":"0T00:00:11","UptimeSec":11,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":59,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ACPRO1","BSSId":"E0:63:DA:65:EA:4A","Channel":6,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:05"}}
00:00:23 WIF: Checking connection...
```

Lights still do not switch on, unfortunately.

ascillato, 2020-06-13:
Weird. 5 seconds and did not respond.

Let's try just disabling poweronstate by doing: `poweronstate 0` and let's try again with that rule.

Thanks for all the testings.

digitlength, 2020-06-14:
> Let's try just disabling poweronstate by doing: `poweronstate 0` and let's try again with that rule.

```
00:01:00 CMD: Group 0, Index 1, Command "POWERONSTATE", Data "OFF"
00:01:00 MQT: stat/swUpstairsHallwayLight/RESULT = {"PowerOnState":0}
00:01:00 CFG: Saved to flash at F6, Count 214, Bytes 4096
```

Power off and back on again:

```
00:00:00 CFG: Loaded from flash at FB, Count 217
00:00:00 QPC: Flag 7E
00:00:00 CFG: CR 364/699
00:00:00 SER: Set to 8N1 9600 bit/s
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.3.1.2(54ab264-tasmota)-2_7_1
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 ACPRO1 Channel 6 BSSId E0:63:DA:65:EA:4A in mode 11N as swUpstairsHallwayLight-2866...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on swUpstairsHallwayLight-2866 with IP address 192.168.1.123
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/swUpstairsHallwayLight/LWT = Online (retained)
00:00:05 MQT: cmnd/swUpstairsHallwayLight/POWER = 
00:00:05 MQT: Subscribe to cmnd/swUpstairsHallwayLight/#
00:00:05 MQT: Subscribe to cmnd/tasmotas/#
00:00:05 MQT: Subscribe to cmnd/swUpstairsHallwayLight_fb/#
00:00:05 MQT: tele/swUpstairsHallwayLight/INFO1 = {"Module":"Sonoff D1","Version":"8.3.1.2(54ab264-tasmota)","FallbackTopic":"cmnd/swUpstairsHallwayLight_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:05 MQT: tele/swUpstairsHallwayLight/INFO2 = {"WebServerMode":"Admin","Hostname":"swUpstairsHallwayLight-2866","IPAddress":"192.168.1.123"}
00:00:05 MQT: tele/swUpstairsHallwayLight/INFO3 = {"RestartReason":"Power On"}
00:00:05 MQT: stat/swUpstairsHallwayLight/RESULT = {"POWER":"OFF"}
00:00:05 MQT: stat/swUpstairsHallwayLight/POWER = OFF
00:00:05 RUL: MQTT#CONNECTED performs "power1 1"
00:00:05 SRC: Rule
00:00:05 CMD: Group 0, Index 1, Command "POWER", Data "1"
00:00:05 MQT: stat/swUpstairsHallwayLight/RESULT = {"POWER":"ON"}
00:00:05 MQT: stat/swUpstairsHallwayLight/POWER = ON
00:00:05 CFG: Saved to flash at FA, Count 218, Bytes 4096
00:00:06 QPC: Reset
00:00:08 APP: Boot Count 20
00:00:08 CFG: Saved to flash at F9, Count 219, Bytes 4096
00:00:09 MQT: tele/swUpstairsHallwayLight/STATE = {"Time":"1970-01-01T00:00:09","Uptime":"0T00:00:11","UptimeSec":11,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":47,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ACPRO1","BSSId":"E0:63:DA:65:EA:4A","Channel":6,"RSSI":100,"Signal":-34,"LinkCount":1,"Downtime":"0T00:00:05"}}
00:00:23 WIF: Checking connection...
```

Again, unfortunately, the lights aren't physically on (the relay hasn't flipped), even though Tasmota thinks they are!

arendst, 2020-06-14:
I've done some testing and I do not see your behaviour; it just works.

What troubles me though is that I do not see any serial comms dump in your logging. When no dump there is no comms between Tasmota and the Sonoff D1 controller so that won't work.

I'm also wondering if you really removed power from the Sonoff D1 os just performed a restart command. In the latter case the PowerOnState code is NOT executed (as it is not a power ON anyway) so then you won't see what you expect.

Let me show you my logging when ``PowerOnState ON`` is set:

```
11:55:15 CMD: weblog 3
11:55:15 SRC: WebConsole from 192.168.2.1
11:55:15 CMD: Group 0, Index 1, Command "WEBLOG", Data "3"
11:55:15 MQT: stat/d1/RESULT = {"WebLog":3}
11:55:19 CMD: power
11:55:19 SRC: WebConsole from 192.168.2.1
11:55:19 CMD: Group 0, Index 1, Command "POWER", Data ""
11:55:19 MQT: stat/d1/RESULT = {"POWER":"ON"}
11:55:19 MQT: stat/d1/POWER = ON
11:55:22 CMD: dimmer
11:55:22 SRC: WebConsole from 192.168.2.1
11:55:22 CMD: Group 0, Index 1, Command "DIMMER", Data ""
11:55:22 MQT: stat/d1/RESULT = {"Dimmer":40}
11:55:30 CMD: poweronstate ON
11:55:30 SRC: WebConsole from 192.168.2.1
11:55:30 CMD: Group 0, Index 1, Command "POWERONSTATE", Data "ON"
11:55:30 MQT: stat/d1/RESULT = {"PowerOnState":1}
11:55:40 CMD: power off
11:55:40 SRC: WebConsole from 192.168.2.1
11:55:40 CMD: Group 0, Index 1, Command "POWER", Data "off"
11:55:40 MQT: stat/d1/RESULT = {"POWER":"OFF"}
11:55:40 MQT: stat/d1/POWER = OFF
11:55:40 DMP: AA 55 01 04 00 00 05
11:55:40 CFG: Saved to flash at FA, Count 92, Bytes 4096
```
The light is off. Now I remove power from the device and watch the log:
```
00:00:00 CFG: Loaded from flash at FA, Count 92
00:00:00 QPC: Flag 7E
00:00:00 SER: Set to 8N1 9600 bit/s
00:00:00 SRC: Restart
00:00:00 Project tasmota D1 Version 8.3.1.3(theo)-STAGE
00:00:00 CFG: Saved to flash at F9, Count 93, Bytes 4096
00:00:00 DMP: AA 55 01 04 00 00 05
00:00:00 DMP: AA 55 01 04 00 00 05
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP2, SSId indebuurt_IoT, Channel 1, BSSId 74:83:C2:2A:F1:AC, RSSI -87, Encryption 1
00:00:04 WIF: Network 1, AP-, SSId indebuurt2, Channel 5, BSSId A0:AB:1B:7D:42:AC, RSSI -26, Encryption 1
00:00:04 WIF: Network 2, AP-, SSId FRITZ!Box 7581 BY, Channel 10, BSSId 82:C7:A6:70:09:21, RSSI -95, Encryption 1
00:00:04 WIF: Network 3, AP2, SSId indebuurt_IoT, Channel 11, BSSId 18:E8:29:CA:17:C1, RSSI -21, Encryption 1
00:00:04 WIF: Connecting to AP2 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11N as d1...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on d1 with IP address 192.168.2.212
11:57:35 NTP: UTC 2020-06-14T09:57:34, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
11:57:35 QPC: Reset
11:57:36 MQT: Attempting connection...
11:57:36 MQT: Connected
11:57:36 MQT: tele/d1/LWT = Online (retained)
11:57:36 MQT: cmnd/d1/POWER = 
11:57:36 MQT: Subscribe to cmnd/d1/#
11:57:36 MQT: Subscribe to cmnd/tasmotas/#
11:57:36 MQT: Subscribe to cmnd/DVES_D9E7CB_fb/#
11:57:36 MQT: tele/d1/INFO1 = {"Module":"Sonoff D1","Version":"8.3.1.3(theo)","FallbackTopic":"cmnd/DVES_D9E7CB_fb/","GroupTopic":"cmnd/tasmotas/"}
11:57:36 MQT: tele/d1/INFO2 = {"WebServerMode":"Admin","Hostname":"d1","IPAddress":"192.168.2.212"}
11:57:36 MQT: tele/d1/INFO3 = {"RestartReason":"Power On"}
11:57:36 MQT: stat/d1/RESULT = {"POWER":"ON"}
11:57:36 MQT: stat/d1/POWER = ON
11:57:37 APP: Boot Count 32
11:57:37 CFG: Saved to flash at F8, Count 94, Bytes 4096
11:57:40 MQT: tele/d1/STATE = {"Time":"2020-06-14T11:57:40","Uptime":"0T00:00:13","UptimeSec":13,"Heap":29,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":85,"MqttCount":1,"POWER":"ON","Dimmer":40,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-27,"LinkCount":1,"Downtime":"0T00:00:07"}}
```
The light is back on WITH THE SAME DIMMER INTENSITY AS IT HAD BEFORE THE POWER OFF COMMAND WAS PERFOROMED on the D1.

If the light was turned off with a dimmer 0 value it won't turn on based on ``PowerOnState`` as poweronstate ONLY turns on the power; it doesn't control the dimmer value. That maybe the issue too in your case.

arendst, 2020-06-14:
If your issue is the dimmer 0 value you might want to try this rule to be executed once MQTT is connected:

```
Rule1 1 
Rule1 on mqtt#connected do dimmer 40 endon
```

This will turn on the light with a default dimmer state of 40 overruling the previous dimmer value but it takes some seconds before MQTT is connected.

arendst, 2020-06-14:
With the latest dev version a new rule trigger is made available allowing quick execution of a rule after restart. In that case try the below rule:

```
Rule1 1 
Rule1 on system#init do dimmer 40 endon
```

digitlength, 2020-06-14:
Hi Theo!

To rule a few of these things out:

Dimmer is at 100% before I turn the device off
```
11:23:16 CMD: dimmer
11:23:16 MQT: stat/swUpstairsHallwayLight/RESULT = {"Dimmer":100}
```

As mentioned in the first post the device is connect via a light switch. When the light switch is off, there is no power. I've tested this with a multimeter just to be sure. I've even switched the circuit off at the breaker/fuse for extra fun - I get the same result every time: no light on on power up.

Changing the rule:
```
11:22:51 CMD: Rule1 on mqtt#connected do dimmer 40 endon
11:22:51 RUL: Stored uncompressed, would compress from 36 to 24 (-33%)
11:22:51 MQT: stat/swUpstairsHallwayLight/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":36,"Free":475,"Rules":"on mqtt#connected do dimmer 40 endon"}
```
IT WORKED! Lights are on. Log below:

```
00:00:00 CFG: Loaded from flash at FA, Count 226
00:00:00 QPC: Flag 7E
00:00:00 CFG: CR 364/699
00:00:00 SER: Set to 8N1 9600 bit/s
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.3.1.2(54ab264-tasmota)-2_7_1
00:00:00 CFG: Saved to flash at F9, Count 227, Bytes 4096
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 ACPRO1 Channel 6 BSSId E0:63:DA:65:EA:4A in mode 11N as swUpstairsHallwayLight-2866...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on swUpstairsHallwayLight-2866 with IP address 192.168.1.123
11:31:15 NTP: UTC 2020-06-14T10:31:14, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
11:31:16 MQT: Attempting connection...
11:31:16 MQT: Connected
11:31:16 MQT: tele/swUpstairsHallwayLight/LWT = Online (retained)
11:31:16 MQT: cmnd/swUpstairsHallwayLight/POWER = 
11:31:16 MQT: Subscribe to cmnd/swUpstairsHallwayLight/#
11:31:16 MQT: Subscribe to cmnd/tasmotas/#
11:31:16 MQT: Subscribe to cmnd/swUpstairsHallwayLight_fb/#
11:31:16 MQT: tele/swUpstairsHallwayLight/INFO1 = {"Module":"Sonoff D1","Version":"8.3.1.2(54ab264-tasmota)","FallbackTopic":"cmnd/swUpstairsHallwayLight_fb/","GroupTopic":"cmnd/tasmotas/"}
11:31:16 MQT: tele/swUpstairsHallwayLight/INFO2 = {"WebServerMode":"Admin","Hostname":"swUpstairsHallwayLight-2866","IPAddress":"192.168.1.123"}
11:31:16 MQT: tele/swUpstairsHallwayLight/INFO3 = {"RestartReason":"Power On"}
11:31:16 MQT: stat/swUpstairsHallwayLight/RESULT = {"POWER":"OFF"}
11:31:16 MQT: stat/swUpstairsHallwayLight/POWER = OFF
11:31:16 RUL: MQTT#CONNECTED performs "dimmer 40"
11:31:16 SRC: Rule
11:31:16 CMD: Group 0, Index 1, Command "DIMMER", Data "40"
11:31:16 SRC: Light
11:31:16 MQT: stat/swUpstairsHallwayLight/RESULT = {"POWER":"ON","Dimmer":40}
11:31:16 DMP: AA 55 01 04 00 00 05
11:31:16 CFG: Saved to flash at F8, Count 228, Bytes 4096
11:31:17 QPC: Reset
11:31:19 APP: Boot Count 22
11:31:19 CFG: Saved to flash at F7, Count 229, Bytes 4096
11:31:20 MQT: tele/swUpstairsHallwayLight/STATE = {"Time":"2020-06-14T11:31:20","Uptime":"0T00:00:11","UptimeSec":11,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":52,"MqttCount":1,"POWER":"ON","Dimmer":40,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ACPRO1","BSSId":"E0:63:DA:65:EA:4A","Channel":6,"RSSI":100,"Signal":-35,"LinkCount":1,"Downtime":"0T00:00:05"}}
```

OK, so now to test I'm going to change the rule back to the original power rule, switch off the light which is now set to 40%, then see what happens:

```
11:31:46 CMD: Rule1 on mqtt#connected do power1 1 endon
11:31:46 SRC: WebConsole from 192.168.1.73
11:31:46 CMD: Group 0, Index 1, Command "RULE", Data "on mqtt#connected do power1 1 endon"
[... 96 lines cut ...]
11:36:16 MQT: cmnd/swUpstairsHallwayLight/POWER = 
11:36:16 MQT: Subscribe to cmnd/swUpstairsHallwayLight/#
11:36:16 MQT: Subscribe to cmnd/tasmotas/#
11:36:16 MQT: Subscribe to cmnd/swUpstairsHallwayLight_fb/#
11:36:16 MQT: tele/swUpstairsHallwayLight/INFO1 = {"Module":"Sonoff D1","Version":"8.3.1.2(54ab264-tasmota)","FallbackTopic":"cmnd/swUpstairsHallwayLight_fb/","GroupTopic":"cmnd/tasmotas/"}
11:36:16 MQT: tele/swUpstairsHallwayLight/INFO2 = {"WebServerMode":"Admin","Hostname":"swUpstairsHallwayLight-2866","IPAddress":"192.168.1.123"}
11:36:16 MQT: tele/swUpstairsHallwayLight/INFO3 = {"RestartReason":"Power On"}
11:36:16 MQT: stat/swUpstairsHallwayLight/RESULT = {"POWER":"ON"}
11:36:16 MQT: stat/swUpstairsHallwayLight/POWER = ON
11:36:17 QPC: Reset
11:36:19 APP: Boot Count 24
11:36:19 CFG: Saved to flash at F7, Count 237, Bytes 4096
11:36:20 MQT: tele/swUpstairsHallwayLight/STATE = {"Time":"2020-06-14T11:36:20","Uptime":"0T00:00:11","UptimeSec":11,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":76,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ACPRO1","BSSId":"E0:63:DA:65:EA:4A","Channel":6,"RSSI":100,"Signal":-35,"LinkCount":1,"Downtime":"0T00:00:05"}}
```
Does *not* work. Just double checking the dimmer value after this switch-on:

```
11:37:18 CMD: dimmer
11:37:18 SRC: WebConsole from 192.168.1.73
11:37:18 CMD: Group 0, Index 1, Command "DIMMER", Data ""
11:37:18 MQT: stat/swUpstairsHallwayLight/RESULT = {"Dimmer":100}
```
So, looks like it will work if we use the dimmer command via mqtt, but there is quite a lag between switching on and the lights actually turning on, as you said there would be. Is there anything else I can check to try and get `PowerOnState` working?

EDIT: Just seen your last post. Will try that.

digitlength, 2020-06-14:
Latest dev branch compiled and uploaded (first time I tried Gitpod - worked great!)

`Status 0`

```
11:55:02 MQT: stat/swUpstairsHallwayLight/STATUS = {"Status":{"Module":74,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"swUpstairsHallwayLight","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:55:02 MQT: stat/swUpstairsHallwayLight/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:06","StartupUTC":"2020-06-14T10:53:56","Sleep":50,"CfgHolder":4617,"BootCount":25,"BCResetTime":"2020-06-07T10:02:52","SaveCount":244,"SaveAddress":"F9000"}}
11:55:02 MQT: stat/swUpstairsHallwayLight/STATUS2 = {"StatusFWR":{"Version":"8.3.1.3(tasmota)","BuildDateTime":"2020-06-14T10:49:16","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"366/699"}}
11:55:02 MQT: stat/swUpstairsHallwayLight/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ACPRO1",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
11:55:02 MQT: stat/swUpstairsHallwayLight/STATUS4 = {"StatusMEM":{"ProgramSize":589,"Free":412,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","043683A1","000000CD","010013C0","C000F981","00004024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
11:55:02 MQT: stat/swUpstairsHallwayLight/STATUS5 = {"StatusNET":{"Hostname":"swUpstairsHallwayLight-2866","IPAddress":"192.168.1.123","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"C8:2B:96:5A:2B:32","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:55:02 MQT: stat/swUpstairsHallwayLight/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.92","MqttPort":1883,"MqttClientMask":"swUpstairsHallwayLight","MqttClient":"swUpstairsHallwayLight","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:55:02 MQT: stat/swUpstairsHallwayLight/STATUS7 = {"StatusTIM":{"UTC":"2020-06-14T10:55:02","Local":"2020-06-14T11:55:02","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:54"}}
11:55:02 MQT: stat/swUpstairsHallwayLight/STATUS10 = {"StatusSNS":{"Time":"2020-06-14T11:55:02"}}
11:55:02 MQT: stat/swUpstairsHallwayLight/STATUS11 = {"StatusSTS":{"Time":"2020-06-14T11:55:02","Uptime":"0T00:01:06","UptimeSec":66,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ACPRO1","BSSId":"E0:63:DA:65:EA:4A","Channel":6,"RSSI":100,"Signal":-34,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```

Use the new rule:
```
11:57:11 CMD: Rule1 on system#init do dimmer 40 endon
11:57:11 RUL: Stored uncompressed, would compress from 33 to 22 (-33%)
11:57:11 MQT: stat/swUpstairsHallwayLight/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":33,"Free":478,"Rules":"on system#init do dimmer 40 endon"}
```
Switching on and off at the main circuit breaker/fuse:

```
00:00:00 CFG: Loaded from flash at F8, Count 253
00:00:00 QPC: Flag 7E
00:00:00 CFG: CR 366/699, Busy 0
00:00:00 SER: Set to 8N1 9600 bit/s
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.3.1.3(tasmota)-2_7_1
00:00:00 RUL: SYSTEM#INIT performs "dimmer 40"
00:00:00 SRC: Rule
00:00:00 CMD: Group 0, Index 1, Command "DIMMER", Data "40"
00:00:00 RSL: stat/swUpstairsHallwayLight/RESULT = {"POWER":"ON","Dimmer":40}
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 ACPRO1 Channel 6 BSSId E0:63:DA:65:EA:4A in mode 11N as swUpstairsHallwayLight-2866...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on swUpstairsHallwayLight-2866 with IP address 192.168.1.123
12:00:09 NTP: UTC 2020-06-14T11:00:08, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
12:00:09 MQT: Attempting connection...
12:00:09 MQT: Connected
12:00:09 MQT: tele/swUpstairsHallwayLight/LWT = Online (retained)
12:00:09 MQT: cmnd/swUpstairsHallwayLight/POWER = 
12:00:09 MQT: Subscribe to cmnd/swUpstairsHallwayLight/#
12:00:09 MQT: Subscribe to cmnd/tasmotas/#
12:00:09 MQT: Subscribe to cmnd/swUpstairsHallwayLight_fb/#
12:00:09 MQT: tele/swUpstairsHallwayLight/INFO1 = {"Module":"Sonoff D1","Version":"8.3.1.3(tasmota)","FallbackTopic":"cmnd/swUpstairsHallwayLight_fb/","GroupTopic":"cmnd/tasmotas/"}
12:00:09 MQT: tele/swUpstairsHallwayLight/INFO2 = {"WebServerMode":"Admin","Hostname":"swUpstairsHallwayLight-2866","IPAddress":"192.168.1.123"}
12:00:09 MQT: tele/swUpstairsHallwayLight/INFO3 = {"RestartReason":"Power On"}
12:00:09 MQT: stat/swUpstairsHallwayLight/RESULT = {"POWER":"ON"}
12:00:09 MQT: stat/swUpstairsHallwayLight/POWER = ON
12:00:10 QPC: Reset
12:00:12 APP: Boot Count 28
12:00:12 CFG: Saved to flash at F7, Count 254, Bytes 4096
12:00:13 MQT: tele/swUpstairsHallwayLight/STATE = {"Time":"2020-06-14T12:00:13","Uptime":"0T00:00:11","UptimeSec":11,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":78,"MqttCount":1,"POWER":"ON","Dimmer":40,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ACPRO1","BSSId":"E0:63:DA:65:EA:4A","Channel":6,"RSSI":100,"Signal":-35,"LinkCount":1,"Downtime":"0T00:00:05"}}
12:00:27 WIF: Checking connection...
```

Does *not* work - lights stay off.

Have I just got some sort of strange setup?!

arendst, 2020-06-14:
I have no explanaination why your D1 fails.

As you can see in my log there are hex dumps in the beginning of a power on cycle:
```
00:00:00 CFG: Loaded from flash at FA, Count 92
00:00:00 QPC: Flag 7E
00:00:00 SER: Set to 8N1 9600 bit/s
00:00:00 SRC: Restart
00:00:00 Project tasmota D1 Version 8.3.1.3(theo)-STAGE
00:00:00 CFG: Saved to flash at F9, Count 93, Bytes 4096
00:00:00 DMP: AA 55 01 04 00 00 05
00:00:00 DMP: AA 55 01 04 00 00 05
```
They are the result of ``PowerOnState On`` and represent the data sent to the D1 light controller. I don't see this in your logging so there might be a comms problem between Tasmota and your D1 light controller.

When you perform power on/off and/or dimmer commands do you always see these DMP messages (when weblog is at least 3) like I do:
```
15:12:38 CMD: power 2
15:12:38 SRC: WebConsole from 192.168.2.1
15:12:38 CMD: Group 0, Index 1, Command "POWER", Data "2"
15:12:38 MQT: stat/d1/RESULT = {"POWER":"ON"}
15:12:38 MQT: stat/d1/POWER = ON
15:12:38 DMP: AA 55 01 04 00 00 05
15:12:39 CFG: Saved to flash at F8, Count 135, Bytes 4096
15:12:45 CMD: power 2
15:12:45 SRC: WebConsole from 192.168.2.1
15:12:45 CMD: Group 0, Index 1, Command "POWER", Data "2"
15:12:45 MQT: stat/d1/RESULT = {"POWER":"OFF"}
15:12:45 MQT: stat/d1/POWER = OFF
15:12:45 DMP: AA 55 01 04 00 00 05
15:12:46 CFG: Saved to flash at F7, Count 136, Bytes 4096
15:13:04 CMD: dimmer 40
15:13:04 SRC: WebConsole from 192.168.2.1
15:13:04 CMD: Group 0, Index 1, Command "DIMMER", Data "40"
15:13:04 SRC: Light
15:13:04 MQT: stat/d1/RESULT = {"POWER":"ON","Dimmer":40}
15:13:04 DMP: AA 55 01 04 00 00 05
15:13:04 CFG: Saved to flash at F6, Count 137, Bytes 4096
```

arendst, 2020-06-14:
BTW how quickly do you toggle the physical power switch? If it's too quick the device either resets parts because of multiple power switches or it doesn't notice it's a power on sequence.

Wait at least ten seconds between physical power toggles.

And to stop the device from resetting after too many quick power cycles set ``SetOption65 1`` and ``SetOption36 0``

digitlength, 2020-06-14:
Nope, I don't seem to have those dumps:

```
14:18:20 CMD: weblog 3
14:18:20 RSL: RESULT = {"WebLog":3}
14:18:20 CFG: Saved to flash at F9, Count 332, Bytes 4096
14:18:33 CMD: Power ON
14:18:33 SRC: WebConsole from 192.168.1.99
14:18:33 CMD: Group 0, Index 1, Command "POWER", Data "ON"
14:18:33 RSL: RESULT = {"POWER":"ON"}
14:18:33 RSL: POWER = ON
14:18:41 CMD: Dimmer 50
14:18:41 SRC: WebConsole from 192.168.1.99
14:18:41 CMD: Group 0, Index 1, Command "DIMMER", Data "50"
14:18:41 RSL: RESULT = {"POWER":"ON","Dimmer":50}
14:18:41 DMP: AA 55 01 04 00 00 05
14:18:42 CFG: Saved to flash at F8, Count 333, Bytes 4096
14:18:51 CMD: Power OFF
14:18:51 SRC: WebConsole from 192.168.1.99
14:18:51 CMD: Group 0, Index 1, Command "POWER", Data "OFF"
14:18:51 RSL: RESULT = {"POWER":"OFF"}
14:18:51 RSL: POWER = OFF
14:18:51 DMP: AA 55 01 04 00 00 05
14:18:52 CFG: Saved to flash at F7, Count 334, Bytes 4096
```

arendst, 2020-06-14:
You have them at 14:18:41 and 14:18:51

digitlength, 2020-06-14:
> You have them at 14:18:41 and 14:18:51

Sorry, yes, but not for the first `Power ON`

arendst, 2020-06-14:
Ah consistently. Well the first time it was probably ON already so it doesn't send again

digitlength, 2020-06-14:
> Ah consistently. Well the first time it was probably ON already so it doesn't send again

Possibly. I'm having that ghost switching issue mentioned elsewhere - I've yet to remove the resistors - so they might have been on: I didn't check before running the commands.

Regarding switching on/off - I was usually waiting about 5 seconds. I'll check what happens after 10 seconds in a bit.

digitlength, 2020-06-14:
And just to double check: should I have changed `SetOptions` on the Sonoff D1? I haven't tried changing anything from default at the moment...

arendst, 2020-06-14:
I did not change them as I normally do not remove power from my devices. In your case I would change them as some day you do a quick power cycle and loose configuration.

digitlength, 2020-06-14:
With:

 - Initial setup:
   - `Rule1 0`
   - `PowerOnState ON`
   - `Power OFF`
   - `Dimmer 50`
   - `SetOption65 1`
   - `SetOption36 0`

Then switching off, going to the toilet, and coming back (a very comfortable number of minutes...), I still get the same - lights do not switch on, and no hex dump on startup:

```
00:00:00 CFG: Loaded from flash at FB, Count 338
00:00:00 CFG: CR 398/699, Busy 0
00:00:00 SER: Set to 8N1 9600 bit/s
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.3.1.3(tasmota)-2_7_1
00:00:00 CFG: Saved to flash at FA, Count 339, Bytes 4096
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 ACPRO1 Channel 6 BSSId E0:63:DA:65:EA:4A in mode 11N as swUpstairsHallwayLight-2866...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on swUpstairsHallwayLight-2866 with IP address 192.168.1.123
14:53:25 NTP: UTC 2020-06-14T13:53:24, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
14:53:25 RSL: INFO1 = {"Module":"Sonoff D1","Version":"8.3.1.3(tasmota)","FallbackTopic":"cmnd/swUpstairsHallwayLight_fb/","GroupTopic":"cmnd/tasmotas/"}
14:53:25 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"swUpstairsHallwayLight-2866","IPAddress":"192.168.1.123"}
14:53:25 RSL: INFO3 = {"RestartReason":"Power On"}
14:53:25 RSL: RESULT = {"POWER":"ON"}
14:53:25 RSL: POWER = ON
14:53:29 APP: Boot Count 43
14:53:29 RSL: STATE = {"Time":"2020-06-14T14:53:29","Uptime":"0T00:00:10","UptimeSec":10,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":50,"MqttCount":0,"POWER":"ON","Dimmer":50,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"ACPRO1","BSSId":"E0:63:DA:65:EA:4A","Channel":6,"RSSI":100,"Signal":-34,"LinkCount":1,"Downtime":"0T00:00:05"}}
14:53:29 CFG: Saved to flash at F9, Count 340, Bytes 4096
```
-----------------------------
Not sure whether relevant, but going right back to the beginning this I how I flashed the device:
```
sudo esptool.py -p /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 /home/pc-user/Downloads/tasmota.bin
```

using this to interface between my Ubuntu 18.04 PC and the device:

https://www.aliexpress.com/item/32761423124.html?spm=a2g0s.9042311.0.0.27424c4dSaIhm2

arendst, 2020-06-14:
Flashing info is fine and OK.

I'm out of ideas other than that I noticed your flashchip id is different from mine. It should still work but it might be a sign that the D1 controller firmware has changed too and fails to re-act to tasmota on startup.

Could you provide the DMP values when you press the Power Off button on your remote please.

Mine reports
```
17:24:31 DMP: AA 55 01 04 00 0A 00 32 FF FF FF FF FF FF FF FF 39
17:24:31 SRC: Switch
17:24:31 MQT: stat/d1/RESULT = {"POWER":"OFF"}
17:24:31 MQT: stat/d1/POWER = OFF
17:24:31 CFG: Saved to flash at F9, Count 169, Bytes 4096
```

digitlength, 2020-06-14:
> Could you provide the DMP values when you press the Power Off button on your remote please.

As in the RF remote? I haven't even opened the box for that yet...

arendst, 2020-06-14:
Yes, the RF remote.

You'll have a challenge pairing the thing but it must be possible as I did it too ;-)

arendst, 2020-06-14:
Well it doesn't really matter. From what your controller returned when you did the power and dimmer commands it seems to be the same as I have so that's fine too.

Sorry, I'm out of ideas.

digitlength, 2020-06-14:
> You'll have a challenge pairing the thing but it must be possible as I did it too ;-)

As an aside, have you had this ghost switching/breathing issue mentioned in https://github.com/arendst/Tasmota/issues/7598 ?

arendst, 2020-06-14:
To be honest I don't know. I wrote the driver. Played with it a few times and dropped the device in the my device bin on to the next challenge. I never had it on for over a few days and never experienced ghost switching.

closed 2020-06-19 by ascillato2

digitlength, 2020-06-19:
To be clear, whilst the label `workaround` has been added, it doesn't actually work for me as detailed above, so the issue still persists.

aguedob, 2020-06-23:
The proposed workaround didn't work fine for me. I think I found a workaround to get the lights on at 100% and every time I use the power switch. I checked that doing a power off/on cycle in the web interface works for me, so I included the same actions in a rule like the following one

on Dimmer#Boot do backlog power off; power on; dimmer 100 endon

The dimmer action is not required.

digitlength, 2020-06-24:
> on Dimmer#Boot do backlog power off; power on; dimmer 100 endon
> 
> The dimmer action is not required.

Well I'm glad I'm not the only one with this problem - I was starting to feel left out!!

Your rule works for me, but with the obvious drawback that it takes some seconds for the rule to fire after power on!

javier6d2, 2020-07-20:
Also works to me with rule1

on Dimmer#Boot do backlog dimmer 80; dimmer 100 endon

Kind regards

Tino4, 2021-01-24:
i have the same problem :PROBLEM DESCRIPTION

I have a Sonoff D1 Dimmer flashed with Tasmota wired in to some ceiling downlights. Power to the Sonoff is via a light switch.

Flipping the light switch off removes power from the Sonoff. Flipping the light switch on restores power to the Sonoff.

I have set PowerOnState ON via the console, and confirmed that the change has stuck.

I've flashed the firmware already 2 times (hardware method)(bootmode :short the pads marked S1 on the pcb for flashing),firmware flash no solution.

Tino4, 2021-01-24:
i guess that there is a bug in the hardware ,i have also noticed that when i switch from on to off ,the lights fade out instead of go out immediatly.FYI.Best Regards , Tino

javier6d2, 2021-01-24:
> the lights fade out instead of go out immediatly

Because the dimmer has a soft start circuit that, gradually increases the startup current from zero to the final value and allow the output voltage to rise at a slower rate, resulting to a lower peak current required for the startup. will extend the life of your downlights

The issue was solved with aguedob solution.

I hope it helps you

digitlength, 2021-01-24:
> The issue was solved with aguedob solution.

As the OP, I guess we should be clear that the issue itself isn't solved. There is a workaround, but as previously mentioned there is a noticeable delay between physically switching and the lights turning on using any workaround based on a rule.

> the lights fade out instead of go out
Probably a capacitor slowly discharging after the device has had power removed, unless it's a feature of the lights themselves.

javier6d2, 2021-03-02:
> there is a noticeable delay between physically switching and the lights turning on using any workaround based on a rule

Hi, today I have found another way to avoid delay between physically switching and the lights turning on. To use it you must compile your build. Add the following to user_config_override.h:
```

#ifndef USE_SCRIPT
#define USE_SCRIPT  // adds about 17k flash size, variable ram size
#endif
#ifdef USE_RULES
#undef USE_RULES
#endif  
```
Use this script:
```
>D
power=0
dimmer=0
>B
delay(10)
=>dimmer 90
=>power 1
=>dimmer 100
```

jenshenk, 2021-06-13:
@javier6d2 This looks very promising! I will try as soon as I have a bit of time for tinkering again. Thanks for this addition! 

I wanted to share my experience and also have a question: So I installed my first D1 today and was originally planning on switching it on and off with a dumb switch, like you. The dimming levels I wanted to control with this switch (they are cheap and look really nice!) https://www.aliexpress.com/item/1005001921814148.html 
The issue that the lamps stay dark after off/on with dumb switch kinda destroyed my original plan so the workaround with the rule works okayish... to get rid of the delay I use the switch from the link now for dimming and on/off switching 

One thing that is actually handy but I don't really get why it happens is this: so I issued the command "poweronstate 0"  and applied the rule from above (on Dimmer#Boot do backlog dimmer 80; dimmer 100 endon), yet the rule seems to only does its thing when I cut power while the lamps were turned on. This is very handy, since otherwise a poweroutage would lead to the lamps always turning on afterwards... but shouldn't the rule be applied no matter which state the dimmer was before cutting power?

jenshenk, 2021-06-14:
Additional experience:
As a last easy thing to try out I now upgraded to the latest development release (on ota.tasmota ... it says 9.4.0.4 on the device it says 9.4.0.5). I did not really study the change logs but I'm happy to report that with the workaround the lights come on at least 1 second faster now compared to the stable 9.4.0!

m-hume, 2022-02-24:
Just ran into exactly the same problem
The script suggested by @javier6d2 worked but always set the brightness to 100 after the power came back
I managed to get the script to return to the previous brightness after a power cycle
```
>D
;define variables
slider=-1

>B
;trigger the first "Tasmota MQTT RESULT" with the current dimmer value
=>dimmer
delay(350)
print --slider is %slider%
if slider=0
then =>dimmer 1
else =>dimmer %(slider-1)%
endif
=>power 1
=>dimmer %slider%

>E
;set slider var to current val if first "Tasmota MQTT RESULT"
if slider<0
then slider=Dimmer
print --set slider to %slider%
endif
```
Would of been a lot easier if you could query the last stored dimmer value directly.
If anyone works out how to do that then please let me know!)
cheers