arendst/Tasmota #8675: node_exporter for ESP 8266 & 32, metrics for prometheus

beeepus, 2020-06-13:
**Have you looked for this feature in other issues and in the docs?**  
Yes, google around and didn't find anything about this. 

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
The problem is related to motoring the devices so we can act if it stop working or acting faulty.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
A common library that can export HW and SW metrics, like the CPU load, up time, firmware version, network errors and so on.  The metrics are published via http or https as clear text. Prometheus service then connects to devices and read metrics and have a good infrastructure and features to handling instances, alerts, history and have good integrations with other systems, for an example with grafana where you can visual see trends and charts from various devices and metrics.
More info about prometheus please follow this link https://prometheus.io/

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
There are other alternatives to prometheus. But i did not dig deep in to compare them. 

**Additional context**  
_Add any other context or screenshots about the feature request here._
I choice to ask for this here at Tasmota. I think this is a good community =O)  

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-06-13:
You might have searched the issues first to find that there is already some Prometheus code available.

Search the code base for define USE_PROMETHEUS

closed 2020-06-13 by ascillato2

beeepus, 2020-06-13:
But I Did, have you? "PROMETHEUS" 0 matches on open or closed. Just this one.
But that's' fantastic. I'm happy.

::beep

meingraham, 2020-06-13:
As Theo (the owner of the Tasmota project) suggested, search for `USE_PROMETHEUS` in the GitHub search box and it will return all instances of that term in the Tasmota repository... including the entry that points to the code Prometheus driver module. `USE_PROMETHEUS` is a compilation directive which you must "enable" and then compile Tasmota in order to have this capability in Tasmota.

https://github.com/arendst/Tasmota/search?q=USE_PROMETHEUS&unscoped_q=USE_PROMETHEUS

@beeepus  
> But I Did, have you? "PROMETHEUS" 0 matches on open or closed. Just this one.

Care to amend your snarky response?

beeepus, 2020-06-13:
@meingraham you are late to the match! I thing Theo understands me =O) he already helped me, that's way I'm happy. A big thank you Theo. 
Peace

meingraham, 2020-06-13:
I wasn't that late as our posts were within minutes of each other. I knew you'd been "made happy" when I posted.

Theo helped regardless of your snark.