arendst/Tasmota #8677: MODBUS with script is not returning any values

tiagofreire-pt, 2020-06-13:
### PROBLEM DESCRIPTION
I'm using a script to get values from a smart energy meter, using a RS485-to-TTL and a Wemos D1 mini.

I see the TX and RX leds on the RS485 converter blinking within the same rhythm defined in the script (3500 ms). So, I'm assuming the smart meter is communicating with success with the Wemos d1 mini.

Connections, on the TTL side (converter -> wemos):

RX -> D4 (GPIO2)
TX -> D1 (GPIO5)

I **successfully** get values from this same smart energy meter with the MODBUS integration on Home Assistant:

```yaml

modbus:
  type: serial
  method: rtu
  port: /dev/ttyUSB0
  baudrate: 9600
  stopbits: 1
  bytesize: 8
  parity: N
  timeout: 10

sensor:
  - platform: modbus
    scan_interval: 2
    registers:
      - name: 'Instantaneous Frequency'
        hub: energy_box_1
        slave: 1
        register: 127
        register_type: input
        unit_of_measurement: 'Hz'
        count: 1
        scale: 0.1
        offset: 0
        precision: 2
        # long unsigned
        data_type: uint
       
      - name: 'Switch Control State'
        hub: energy_box_1
        slave: 1
        register: 132
        register_type: input
        count: 1
        # conversão de 1/256
        scale: 0.00390625
        offset: 0
        ## 0 para disconectado - 1 para conectado - 2 para preparado a reconectar
        # precision: 0
        data_type: uint
 
      - name: 'Instantaneous Voltage L1'
        hub: energy_box_1
        slave: 1
        register: 108
        register_type: input
        unit_of_measurement: 'V'
        count: 1
        scale: 0.1
        offset: 0
        precision: 1
        # long unsigned -- https://github.com/edphackathon/EDPIoTHackathon2017/blob/master/Documentation/hardware-intermediate-eb-comm.md#modbus-master-slave-for-arduino
        data_type: uint
 
      - name: 'Instantaneous Current L1'
        hub: energy_box_1
        slave: 1
[... 175 lines cut ...]
### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

**General data types send by this smart energy meter:**

[image]

in: https://github.com/edphackathon/EDPIoTHackathon2017/blob/master/Documentation/hardware-intermediate-eb-comm.md

**MODBUS:**

[image]

in: https://github.com/edphackathon/EDPIoTHackathon2017/blob/master/Documentation/hardware-intermediate-eb.md#eb-introduction

**Register addresses and data types:**

[image]

In: https://www.edpdistribuicao.pt/sites/edd/files/normative_docs/DEF-C44-509.pdf

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

tiagofreire-pt, 2020-06-13:
I detected that is not available an option for transmission with 8N2, just for 8N1:

```
- m = MODBus binary coding with serial mode 8N1
- M = MODBus binary coding with serial mode 8E1
```

In: https://tasmota.github.io/docs/Smart-Meter-Interface/#meter-declaration

This smart meter uses 8N2:

[image]

In: https://www.edpdistribuicao.pt/sites/edd/files/normative_docs/DEF-C44-509.pdf

Is this relevant?

gemu2015, 2020-06-13:
did you check in debug mode what comes back from the meter ?
go to console and type : sensor53 d1
you should see the hex values coming from the meter.
(to revert to normal mode you must type sensor53 d0)

according 8N2 your home assistant setup also uses 8N1
don't think that is the reason

tiagofreire-pt, 2020-06-13:
> did you check in debug mode what comes back from the meter ?
> go to console and type : sensor53 d1
> you should see the hex values coming from the meter.
> (to revert to normal mode you must type sensor53 d0)
> 
> according 8N2 your home assistant setup also uses 8N1
> don't think that is the reason

This is the output:

```
16:37:14 MQT: stat/tasmota_4C713F/RESULT = {"Time":"2020-06-13T16:37:14","SML":{"CMD":"dump: 1"}}
16:37:14 : 01 04 08 00 00 01 b3 00 00 00 00 
16:37:14 : 20 06 
16:37:17 : 01 84 04 42 c3 
16:37:18 WIF: Checking connection...
16:37:20 : 01 04 0a 01 f4 01 02 09 0b 0e 
16:37:20 : 0a 0d 0c d7 57 
16:37:23 : 01 04 04 09 16 00 15 
16:37:23 : d8 13 
16:37:26 : 01 84 04 42 c3 
16:37:29 : 01 04 08 00 00 01 a1 00 00 00 
16:37:29 : 00 98 05 
16:37:32 : 01 84 04 42 c3 
16:37:35 : 01 04 0a 01 f4 01 02 09 0b 0e 
16:37:35 : 0a 0d 0c d7 57 
16:37:38 : 01 04 04 09 19 00 14 29 d0 
16:37:38 WIF: Checking connection...
16:37:41 : 01 84 04 42 c3 
16:37:44 : 01 04 08 00 00 01 84 00 00 00 00 d5 
16:37:44 : c2 
```

gemu2015, 2020-06-13:
looks a bit strange
try requesting one register only
(from the docs: the number of requested registers is fixed to 2, however with the char 'r' before the hex string the complete request string may be specified
...r010400000001,r010400020003,...)

try with one register only and one value

+1,2,m,0,9600,MODBUS,5,30,r0104006C0001

tiagofreire-pt, 2020-06-13:
With this script:

```
>D
 
>B
=>sensor53 r
 
>M 1
 
+1,14,m,1,9600,MODBUS,5,100,0104006C
 
1,010402UUuu@i0:0.1,Voltage,V,Voltage_P1,18
#
```

The output, for > 30 seconds (polling each 10 sec):

```
17:03:14 CMD: sensor53 d1
17:03:14 SRC: WebConsole from 192.168.10.33
17:03:14 CMD: Group 0, Index 53, Command "SENSOR", Data "d1"
17:03:14 MQT: stat/tasmota_4C713F/RESULT = {"Time":"2020-06-13T17:03:14","SML":{"CMD":"dump: 1"}}
17:03:19 : 01 04 04 09 18 00 17 38 11 
17:03:29 : 01 04 04 09 1a 00 18 
17:03:29 : d9 d5 
17:03:30 WIF: Checking connection...
17:03:39 : 01 04 04 09 1b 00 18 88 15 
17:03:49 : 01 04 04 09 1a 00 17 99 d1 
17:03:50 WIF: Checking connection...
17:03:55 MQT: tele/tasmota_4C713F/STATE = {"Time":"2020-06-13T17:03:55","Uptime":"0T00:10:10","UptimeSec":610,"Vcc":2.993,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":99,"MqttCount":1,"Wifi":{"AP":1,"SSId":"PSA_WIFI","BSSId":"2C:FD:A1:3E:31:58","Channel":11,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:03:55 MQT: tele/tasmota_4C713F/SENSOR = {"Time":"2020-06-13T17:03:55","MODBUS":{"Voltage_P1":0.00}}
```

gemu2015, 2020-06-13:
ok looks better,
the decoder however is wrong needs 
010404 .... 
may be a long or float value

tiagofreire-pt, 2020-06-13:
This parameter (Voltage_P1) is a "Long unsigned". 

New script:

```
>D
 
>B
=>sensor53 r
 
>M 1
 
+1,14,m,1,9600,MODBUS,5,100,0104006C
 
1,010404UUuuUUuu@i0:0.1,Voltage,V,Voltage_P1,18
#
```

The output:

```
17:12:27 CMD: sensor53 d1
17:12:27 SRC: WebConsole from 192.168.10.33
17:12:27 CMD: Group 0, Index 53, Command "SENSOR", Data "d1"
17:12:27 MQT: stat/tasmota_4C713F/RESULT = {"Time":"2020-06-13T17:12:27","SML":{"CMD":"dump: 1"}}
17:12:29 : 01 04 04 09 24 00 17 f8 1d 
17:12:30 WIF: Checking connection...
17:12:39 : 01 04 04 09 22 00 
17:12:39 : 17 18 1c 
17:12:49 : 01 04 04 09 22 00 17 
17:12:49 : 18 1c 
17:12:50 WIF: Checking connection...
17:12:59 : 01 04 04 09 24 
17:12:59 : 00 17 f8 1d 
17:13:09 : 01 04 04 09 22 00 17 
17:13:09 : 18 1c 
17:13:10 WIF: Checking connection...
```

The voltage value continues to be reported as 0.00 Volt. :(

tiagofreire-pt, 2020-06-13:
The manual states the "long unsigned" are 2 bytes long. The UUuu (2 bytes?) previously used fails, and also UUuuUUuu (4 bytes?)

tiagofreire-pt, 2020-06-13:
@arendst , could you also give a hand on this, please? :)

Thanks for both of you.

gemu2015, 2020-06-13:
the meter returns 4 bytes payload (+2 bytes crc)
e.g. 01 04 04 09 22 00 17 18 1c
09 22 00 17 being the payload 
according to your table above register 6c = unsigned long
however 0922 = 2338 looks like voltage times 10 =????
if the decoder fails probably the crc is wrong
if the crc would be ok it would decode something 
the crc on transmit is obviously ok, otherwise the meter would not respond

gemu2015, 2020-06-13:
did you switch back to normal mode ?
in debug mode the decoder is not active

tiagofreire-pt, 2020-06-13:
> the meter returns 4 bytes payload (+2 bytes crc)
> e.g. 01 04 04 09 22 00 17 18 1c
> 09 22 00 17 being the payload
>
> however 0922 = 2338 looks like voltage times 10 =????

Yes, the scaler is 10^-1, or, 0.1, leading to a voltage of 233.8 Volt.

> did you switch back to normal mode ?
> in debug mode the decoder is not active

No, I'm running this with debug (sensor53 d1) all the time, since you suggest it. lol

Disabling it, with this one:

```
>D
 
>B
=>sensor53 r
 
>M 1
 
+1,14,m,1,9600,MODBUS,5,100,0104006C
 
1,010404UUuuuuuu@i0:0.1,Voltage,V,Voltage_P1,2
#
```

I now get a value, but widely missing the target:

```
17:39:42 MQT: tele/tasmota_4C713F/SENSOR = {"Time":"2020-06-13T17:39:42","MODBUS":{"Voltage_P1":200.00}}
```

gemu2015, 2020-06-13:
your decoder is still wrong
UUuuuuuu is not defined.
use either uuUUuuUU or UUuuUUuu

tiagofreire-pt, 2020-06-13:
@gemu2015 , finally!

```
>D
 
>B
=>sensor53 r
 
>M 1
 
+1,14,m,1,9600,MODBUS,5,100,0104006C
 
1,010404UUuuxxxx@i0:10,Voltage,V,Voltage_P1,2
#
```

[image]

YAY! :D :D :D

gemu2015, 2020-06-13:
ok, now proceed to the other values
good look

ascillato2, 2020-06-13:
Great to know that now it is working for you. 

Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-06-13 by ascillato2

tiagofreire-pt, 2020-06-13:
@gemu2015 , 

I'm getting errors, reading the L1 current values:

```
>D 

>B
=>sensor53 r

>M 1

+1,14,m,1,9600,MODBUS,5,100,0104006D
 
1,010404UUuuxxxx@i0:10,Current,V,Current_P1,2
#
```

Output:

```
18:29:55 : 01 84 04 42 c3 
```

[image]

I'm sure the register is correct (0x006C or 109).

gemu2015, 2020-06-13:
go to raw mode for all your queries, may be your meter doesn't like 2 registers at once
+1,14,m,1,9600,MODBUS,5,100,r0104006D0001

tiagofreire-pt, 2020-06-13:
> +1,14,m,1,9600,MODBUS,5,100,r0104006D0001

That instruction does not output anything on debug mode:

```
>D 

>B
=>sensor53 r

>M 1

+1,14,m,1,9600,MODBUS,5,100,r0104006D0001
 
1,010404UUuuxxxx@i0:10,Current,V,Current_P1,2
#
```

tiagofreire-pt, 2020-06-13:
This behavior occurs also with the 0x007B register (power factor).

gemu2015, 2020-06-13:
there are only 2 modes without prefix 2 registers are fetched.
with prefix r the number of register to fetch must be appended.
since it worked to fetch 6c 
try the following, fetching 2 registers at once (both with index 0)

+1,14,m,1,9600,MODBUS,5,100,0104006C
 
1,010404UUuuxxxx@i0:10,Voltage,V,Voltage_P1,2
1,010404xxxxUUuu@i0:10,Current,V,Current_P1,2
#

tiagofreire-pt, 2020-06-13:
> +1,14,m,1,9600,MODBUS,5,100,0104006C
> 
> 1,010404UUuuxxxx@i0:10,Voltage,V,Voltage_P1,2
> 1,010404xxxxUUuu@i0:10,Current,V,Current_P1,2

Does not work properly, as both voltage and current get the same value (the one from voltage):

```
18:56:59 MQT: tele/tasmota_4C713F/SENSOR = {"Time":"2020-06-13T18:56:59","EB":{"Voltage_P1":232.20,"Current_P1":232.20,"Power_P1":367.00,"Frequency_P1":50.00}}
```

gemu2015, 2020-06-13:
that should not be the case. from the values of your dump above the "long" words are different.
post your complete script

tiagofreire-pt, 2020-06-13:
> that should not be the case. from the values of your dump above the "long" words are different.
> post your complete script

The one: 

```
>D
 
>B
=>sensor53 r
 
>M 1
 
+1,14,m,1,9600,EB,5,50,0104006C
 
1,010404UUuuxxxx@i0:10,Voltage,V,Voltage_P1,2
1,010404UUuuxxxx@i0:10,Current,A,Current_P1,2
#
```

gemu2015, 2020-06-13:
second line is wrong, look at my post

tiagofreire-pt, 2020-06-13:
> second line is wrong, look at my post

I'm just blind after all these hours looking at this! :D

It worked.

```
19:12:09 MQT: tele/tasmota_4C713F/SENSOR = {"Time":"2020-06-13T19:12:09","EB":{"Voltage_P1":231.10,"Current_P1":2.60}}
```

What I'm not be able to address is the power factor, reading its own register (0x007B), the smart meter returns an error.

gemu2015, 2020-06-13:
i have no idea why your meter returns an error here.
since others work, these are probably not implemented at your model

tiagofreire-pt, 2020-06-13:
> i have no idea why your meter returns an error here.
> since others work, these are probably not implemented at your model

Oddly, it works on HA, using its decimal number 123.

kugelkopf123, 2020-06-13:
Perhaps it would be great to add this meter to the docs. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

tiagofreire-pt, 2020-06-13:
> Perhaps it would be great to add this meter to the docs.
> 
> Sent with [GitHawk](http://githawk.com)

I will, soon. As this is a huge step into it and widely expected in the community. :)

gemu2015, 2020-06-13:
ok, i think i found the reason,
7B is there, but 7c is only there on 3 phase meters. do you have a single phase meter ?
this would explain the error when grabbing 2 registers at once

tiagofreire-pt, 2020-06-13:
Mine is single phase.

Even using this:

```
>D
 
>B
=>sensor53 r
 
>M 1
 
+1,14,m,1,9600,EB,5,50,0104007B
 
1,010404UUuuxxxx@i0:1000,PF1,pu,PFactor_P1,3
#
```

I get error messages.

gemu2015, 2020-06-13:
yes but try to fetch 7a, but read 2. part of payload
+1,14,m,1,9600,EB,5,50,0104007A
1,010404xxxxUUuu@i0:1000,PF1,pu,PFactor_P1,3

gemu2015, 2020-06-13:
ok, but 7a is "double long" above decoder won't work
but the error should be gone

gemu2015, 2020-06-13:
still wondering why raw mode does not work (single register read)
+1,14,m,1,9600,EB,5,50,r0104007B0001
1,010404UUuu@i0:1000,PF1,pu,PFactor_P1,3

gemu2015, 2020-06-13:
if reading 2 registers at 7a works, this decoder might work
+1,14,m,1,9600,EB,5,50,0104007A
1,010406xxxxxxxxUUuu@i0:1000,PF1,pu,PFactor_P1,3

tiagofreire-pt, 2020-06-13:
@gemu2015 , you deserve a really huge bronze statue!!! 

```
>D 

>B
=>sensor53 r

>M 1

+1,14,m,1,9600,EB,5,10,0104006C,01040079,0104007F,0104007A
 
1,010404UUuuxxxx@i0:10,Voltage,V,Voltage_P1,2
1,010404xxxxUUuu@i0:10,Current,V,Current_P1,2
1,010408UUuuUUuuxxxxxxxxxxxx@i1:1,Power,W,Power_P1,2
1,01040aUUuuxxxx@i2:10,Frequency,Hz,Frequency_P1,2
1,010406xxxxxxxxUUuu@i3:1000,Power Factor,pu,PFactor_P1,3
#
```

Many, many thanks!