arendst/Tasmota #8684: Home Assistant invalid RGB received

duhow, 2020-06-14:
### PROBLEM DESCRIPTION
Exception found in Home Assistant 0.111 when switching on a Tasmota bulb light.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Teckin S50
- [x] Tasmota binary firmware version number used: 8.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tuya-Convert
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
11:11:58 CMD: Backlog Template; Module; GPIO 255
11:11:58 MQT: stat/tasmota/RESULT = {"NAME":"Teckin SB50v3","GPIO":[0,0,0,0,40,0,0,0,38,39,37,0,0],"FLAG":0,"BASE":18}
11:11:58 MQT: stat/tasmota/RESULT = {"Module":{"0":"Teckin SB50v3"}}
11:11:58 MQT: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"40":"PWM4"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"38":"PWM2"},"GPIO13":{"39":"PWM3"},"GPIO14":{"37":"PWM1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE

Exception in rgb_received when handling msg on 'tele/tasmota/STATE':
```json
'{"Time":"2020-06-14T10:38:33","Uptime":"0T00:00:10","UptimeSec":10,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":155,"MqttCount":1,"POWER":"OFF","Dimmer":100,"Color":"000000FF","HSBColor":"0,0,0","White":100,"Channel":[0,0,0,100],"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"XXXXXXXXX","BSSId":"XX:XX:XX:XX:XX:XX","Channel":8,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:05"}}'
```
```
Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 35, in wrapper msg_callback(msg) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_basic.py", line 364, in rgb_received rgb = [int(val) for val in payload.split(",")] File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_basic.py", line 364, in <listcomp> rgb = [int(val) for val in payload.split(",")] ValueError: invalid literal for int() with base 10: '000000FF'
```

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-06-14:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

effelle, 2020-06-14:
As designed, Discovery needs the values as decimals for colors. 
Check the status of `SetOption17` (Discovery set it to 1 automatically) if is set to 0 then you have your guilty, since Tasmota will transmit color value in hexadecimal instead of decimal values:
```
11:25:03 MQT: stat/test1/RESULT = 
{"POWER":"ON","Dimmer":57,"Color":"145,62,124,0","HSBColor":"315,57,57",
"White":0,"Channel":[57,24,49,0]}
```

closed 2020-06-14 by ascillato2