arendst/Tasmota #8685: HUE Simulation missing ", so ignorde by Alexa

TheRocktagon, 2020-06-14:
Usinge Scripting and Hue (and Zigbee usw)
I found an Error in xdrv_10

Line: 4821 *response+=String(EncodeLightId(hue_devs+devices_present+1))+"\":";
must be (adding "\""+):
                *response+="\""+String(EncodeLightId(hue_devs+devices_present+1,0))+"\":";

The " is missing in the String so it only reports:

http://192.168.178.58/apicode/lights
{235104865":{"state":........

and Alexa ignore it.
Withe the "\""+ it reports: 

{"235104865":{"state":........

and Alexa found the device.

closed 2020-06-14 by close-issue-app[bot]

s-hadinger, 2020-06-15:
@gemu2015 Can you please check?

I'm not sure that the patch proposed here works in all cases.

gemu2015, 2020-06-15:
@TheRocktagon 
can you check this fix ?

if (response) {
        if (devices_present) {
          *response+=",\"";
        }
        else {
          if (hue_devs>0) *response+=",\"";
          else *response+="\"";
        }
        *response+=String(EncodeLightId(hue_devs+devices_present+1))+"\":";
        Script_HueStatus(response,hue_devs);

gemu2015, 2020-06-15:
ok reactivated my Alexa setup and checked, my fix works in all conditions doing pr soon.

TheRocktagon, 2020-06-15:
Checked it without my modyfication and it works good. -- Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

TheRocktagon, 2020-06-16:
Hello, sry for the short disturbance again. Just a little hint, init the MCP Ic (Sensor29) in >B section by using scripting it doesen't work, just errors by booting.I only changing in tasmota. Ino:put "XsnsCall(FUNC_INIT)" before "XdrvCall(FUNC_INIT)" so the sensors would be init bevore the script starts and all works fine.Picture included.GreetsTorsten-- Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

gemu2015, 2020-06-16:
ok, we have to ask @arends if something speaks against it.
otherwise good hint.

gemu2015, 2020-06-16:
i will solve this by introducing a new section after sensor init, so to not break anything
executed after sensor init
>BS
will make a pr soon

  XdrvCall(FUNC_INIT);
  XsnsCall(FUNC_INIT);
#ifdef USE_SCRIPT
  Run_Scripter(">BS",3,0);
#endif

arendst, 2020-06-16:
Yes. The driver needs to be init BEFORE the sensors. 

@gemu your solution looks OK.

PR #8700 (2020-06-15): scripter fix hue, add ticker, add tesla json read
## Description:
fixes HUE >H issue, when no other device configured
add ticker support (up to 4 Tickers with callbacks)
add Tesla Powerwall Controller access via JSON (currently ESP32 only, uses HTTPS,TLS)

**Related issue (if applicable):** fixes #8685

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core ESP8266 V.2.7.1
  - [x] The code change is tested and works on core ESP32 V.1. [... cut]