arendst/Tasmota #8691: Add End Stops for A4988 Stepper Control

holtm, 2020-06-14:
**Have you looked for this feature in other issues and in the docs?**  
I have searched the docs and looked through the code and can not find where end stops are implemented for the A4988 Stepper build option.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I would like to use Tasmota to open/close a window using the A4988 Stepper option, but can't find where I can use end stops. End stops are switches (physical or optical) that indicate when the physical location of the movement is at the end. In my case they indicate when the the window is fully open or closed. This feature maybe useful for other uses when an indication of physical movement or location is needed.
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I would like to add of have added options to assign end stop pins and only have the motor advance until the end stop triggers or when the commanded steps are complete if the end stop pins are assigned. I have tested this and would like to formally have the code included in the dev branch if it would of use to others. No additional Commands would be added. Only pin assignment options and stopping movement based on the status of the assigned pins (if assigned).
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I looked at using rules or disabling the stepper when the end point was reached. Neither of these options appear to work with the current code with out complication or I couldn't figure out how to implement.
**Additional context**  
_Add any other context or screenshots about the feature request here._
This would augment the enhancement implemented by Pull Request #6370

**(Please, remember to close the issue when the problem has been addressed)**

TrDA-hab, 2020-06-16:
@holtm
1. You must use the control method for the A4988 from this article:
https://tasmota.github.io/docs/Blinds-and-Shutters/#using-stepper-motors
2. You must create a rule for the Switch as described in the example from this article:
https://tasmota.github.io/docs/Buttons-and-Switches/
3. You must use the ShutterStop<x> command as an argument to your rule.
https://tasmota.github.io/docs/Commands/#shutters

Good luck!

ascillato2, 2020-06-16:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-06-16 by ascillato2

holtm, 2020-06-17:
Thanks for the info and for your work on this system. It's very configurable.  I had previously tried to use the Blinds and Shutters way of driving the stepper and it was much more complicated. Plus, I was not able to get it to be reliable with my configuration. It appeared the watchdog timer would reboot the Wemos D1 mini and the counter would reset to 0 and never reach the target steps. So I added the end stops to the A4988 code instead. I'll try again sometime and see if I can't debug why the watchdog caused the reboot -- but that should be in a separate post.