arendst/Tasmota #8692: Display comands are not working

gamer123, 2020-06-14:
### PROBLEM DESCRIPTION
The display comands they are listet in the comand list do not work.
But I think there is another problem.

I tried the pre-compiled version tasmota-display 8.3.1.
I tried the pre-compiled version tasmota-display 8.1.0.
I tried the pre-compiled version sonoff-display 6.4.0.

And i made some own binaries and i used 8.3.1

#define USE_I2C // I2C using library wire (+10k code, 0k2 mem, 124 iram)
#ifdef USE_I2C

#define USE_DISPLAY // Add I2C Display Support (+2k code)
#define USE_DISPLAY_MODES1TO5 // Enable display mode 1 to 5 in addition to mode 0
#define USE_DISPLAY_SSD1306 // [DisplayModel 2] [I2cDriver4] Enable SSD1306 Oled 128x64 display (I2C addresses 0x3C and 0x3D) (+16k code)

#endif // USE_I2C
.

The problem is that the commands in the console are acknowledged as unknown.
I try to set up an OLED display with 64x48 pixel.

Display
DisplayModel 2
DisplayWidth 64
DisplayHeight 4

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Wemos d1 mini Pro 
- [x ] Tasmota binary firmware version number used: tasmota display
  - [x ] Pre-compiled
  - [ x] Self-compiled
    - [x ] IDE / Compiler used: 2.5.2
- [X ] Flashing tools used: Tasmota PyFlasher 
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [x ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

00:00:00 CFG: Loaded from flash at F5, Count 7
00:00:00 Project tasmota Tasmota Version 8.3.1(display)-2_7_1
00:00:00 WIF: Connecting to AP1 Pencil_Gast Channel 11 BSSId 38:10:D5:DB:42:A5 in mode 11N as test...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on test with IP address 192.168.0.77
18:10:16 RSL: tele/tasmota_F4E97B/INFO1 = {"Module":"Sonoff Basic","Version":"8.3.1(display)","FallbackTopic":"cmnd/DVES_F4E97B_fb/","GroupTopic":"cmnd/tasmotas/"}
18:10:16 RSL: tele/tasmota_F4E97B/INFO2 = {"WebServerMode":"Admin","Hostname":"test","IPAddress":"192.168.0.77"}
18:10:16 RSL: tele/tasmota_F4E97B/INFO3 = {"RestartReason":"Software/System restart"}
18:10:16 RSL: stat/tasmota_F4E97B/RESULT = {"POWER":"OFF"}
18:10:16 RSL: stat/tasmota_F4E97B/POWER = OFF
18:10:20 RSL: tele/tasmota_F4E97B/STATE = {"Time":"2020-06-14T18:10:20","Uptime":"0T00:00:09","UptimeSec":9,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Pencil_Gast","BSSId":"38:10:D5:DB:42:A5","Channel":11,"RSSI":36,"Signal":-82,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:10:41 CMD: display
18:10:41 RSL: stat/tasmota_F4E97B/RESULT = {"Command":"Unknown"}
18:11:10 CMD: displaymodel 2
18:11:10 RSL: stat/tasmota_F4E97B/RESULT = {"Command":"Unknown"}
18:11:24 CMD: DisplayModel 2
18:11:24 RSL: stat/tasmota_F4E97B/RESULT = {"Command":"Unknown"}
18:12:13 CFG: CR 374/699
18:12:13 LOG: SerialLog 2, WebLog 4, MqttLog 0, SysLog 0, LogHost , LogPort 514, TelePeriod 300
18:12:13 HTP: Configuration
18:12:13 CFG: Saved to flash at FA, Count 10, Bytes 4096
18:12:14 WIF: Checking connection...
18:12:16 HTP: Main Menu
18:12:18 HTP: Console
18:12:24 CMD: Display
18:12:24 SRC: WebConsole from 192.168.0.14
18:12:24 CMD: Group 0, Index 1, Command "DISPLAY", Data ""
18:12:24 RSL: stat/tasmota_F4E97B/RESULT = {"Command":"Unknown"}
18:12:34 WIF: Checking connection...
18:12:36 CMD: DisplayModel 2
18:12:36 SRC: WebConsole from 192.168.0.14
18:12:36 CMD: Group 0, Index 1, Command "DISPLAYMODEL", Data "2"
18:12:36 RSL: stat/tasmota_F4E97B/RESULT = {"Command":"Unknown"}

### TO REPRODUCE
Try to set up a tasmota-Display

### EXPECTED BEHAVIOUR

Hm i would like to cofigure the display.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I only selected the I2C ports At GIPO 1 and 2 
I selected the generic modul (18).

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-06-14:
Please, post status 0 output as requested. Thanks.

gamer123, 2020-06-14:
Sorry for that i added the Status 0.

Jason2866, 2020-06-14:
GPIO 1 and 2 are not the best choice. What do you get when you fire the command `i2cscan`

device111, 2020-06-14:
change your GPIO Ports to 4 and 5 as Jason2866 suggested.
If you have a dev Board (NodeMCU, o.s) , the GPIO 2 are almost connected with the On-board LED.

gamer123, 2020-06-15:
hello,

i changed the the port to 4 and 5. 
i used again the tasmota-display 8.3.1 with Tasmota-PyFlasher-1.0

i2cscan 
07:50:40 SRC: WebConsole from 192.168.8.103
07:50:40 CMD: Group 0, Index 1, Command "I2CSCAN", Data ""
07:50:40 RSL: stat/tasmota_F4E98B/RESULT = {"Command":"Error"}

i tried it with and without the display -->same result 

I2cdriver 

08:02:02 CMD: i2cdriver
08:02:02 SRC: WebConsole from 192.168.8.103
08:02:02 CMD: Group 0, Index 1, Command "I2CDRIVER", Data ""
08:02:02 RSL: stat/tasmota_F4E98B/RESULT = {"I2CDriver":"3,4,5,6,47"}

In my point of view i need driver 4, so it should be fine.

i use the wemos D1 mini pro with 4MB

The display is similar to 
https://www.reichelt.de/d1-shield-oled-display-0-66-1-67-cm-d1z-oled-0-66-p266074.html

in my point of view is the big question is why is the comand displaymodel 2 unknown in the webconsole.

device111, 2020-06-15:
Your Display is not found. It is not a Problem with Tasmota programming. Please go to support chat, may some one can help you.

closed 2020-06-15 by gamer123

sandu-bogdan, 2024-04-22:
how you solve? I have same problem...