arendst/Tasmota #8694: Delay on restoring white/CT value after main power cut

Onyxmoon, 2020-06-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The module is installed by the manufacturer in a ceiling light with CT function. This uses PWM CT to set the colour temperature (described here: https://tasmota.github.io/docs/Lights/#pwm-ct). This also works very well so far (I used Xiaomi Philips as a base template). But if you control the lamp with a switch (therefore really disconnect it from the mains), the colour temperature is completely white for a short time after switching on and after about one second the lamp changes back to the original value before switching off. The brightness is immediately restored correctly.

It may be necessary to say that PWM2 is inverted, therefore PWM2i.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Xiaomi Philips
- [X] Tasmota binary firmware version number used: 8.3.1.3
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tuya convert
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
00:04:47 MQT: stat/kuechendecke/RESULT = {"NAME":"LVL 300m Round 24W Ceiling LED","GPIO":[0,0,0,0,0,37,0,0,0,47,0,0,0],"FLAG":0,"BASE":48}
00:04:47 MQT: stat/kuechendecke/RESULT = {"Module":{"0":"LVL 300m Round 24W Ceiling LED"}}
00:04:47 MQT: stat/kuechendecke/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"37":"PWM1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"47":"PWM2i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
00:05:54 MQT: stat/kuechendecke/STATUS = {"Status":{"Module":0,"DeviceName":"Küchendecke","FriendlyName":["Küchendecke"],"Topic":"kuechendecke","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
00:05:54 MQT: stat/kuechendecke/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota-DE.bin","RestartReason":"Power On","Uptime":"0T00:17:47","StartupUTC":"2020-06-14T22:48:07","Sleep":50,"CfgHolder":4617,"BootCount":66,"BCResetTime":"2020-06-14T18:46:12","SaveCount":568,"SaveAddress":"F6000"}}
00:05:54 MQT: stat/kuechendecke/STATUS2 = {"StatusFWR":{"Version":"8.3.1.3(3c4fb6c-tasmota)","BuildDateTime":"2020.06.14 16:06:00","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"417/699"}}
00:05:54 MQT: stat/kuechendecke/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Satinavs Ketten",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008029","2805C8000100060000005A00000000000000","00000000","00000601"]}}
00:05:54 MQT: stat/kuechendecke/STATUS4 = {"StatusMEM":{"ProgramSize":590,"Free":412,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":0,"Features":["00000407","8FDAE797","043683A1","000000CD","010013C0","C000F981","00004024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
00:05:54 MQT: stat/kuechendecke/STATUS5 = {"StatusNET":{"Hostname":"Küchendecke","IPAddress":"10.0.1.166","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"98:F4:AB:C5:87:5C","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
00:05:54 MQT: stat/kuechendecke/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.151","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C5875C","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:05:54 MQT: stat/kuechendecke/STATUS7 = {"StatusTIM":{"UTC":"2020-06-14T23:05:54","Local":"2020-06-15T00:05:54","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:55"}}
00:05:54 MQT: stat/kuechendecke/STATUS10 = {"StatusSNS":{"Time":"2020-06-15T00:05:54"}}
00:05:54 MQT: stat/kuechendecke/STATUS11 = {"StatusSTS":{"Time":"2020-06-15T00:05:54","Uptime":"0T00:17:47","UptimeSec":1067,"Heap":22,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":29,"Color":"0941","White":29,"CT":459,"Channel":[4,25],"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Satinavs Ketten","BSSId":"78:8A:20:71:E5:BA","Channel":6,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Lamp does instantly recover states like with the original firmware.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I know it's a luxury problem, but it's irritating when you turn on the light and it's white for a moment and then it jumps back to warm white.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-06-15:
The way Tasmota works at power up (in theory):
1. First the PWM GPIO are configured as outputs, and the level is set to zero, which means LOW for PMW and HIGH for PWMi.
2. As soon as the event loop fires, the light module computes the correct PWM values and sets both PWM1 and PWM2i with the correct value.

Do you have any Rules of MQTT with Retained messages?

Onyxmoon, 2020-06-15:
Thanks for the quick answer. In theory it sounds really fantastic, I don't know why my lamp doesn't feel in the mood for it.

Unfortunately it behaves like this even if MQTT is deactivated.
I have attached a gif so you can follow the reaction a little bit. 

[image]

Any ideas? 
I mean, the original manufacturer has at least figured it out somehow. Have I done something wrong?
(Product: https://de.aliexpress.com/i/4000115477858.html)

s-hadinger, 2020-06-15:
Thanks for the video. Is the video real-time or slow-motion?

What was the brightness level? And CT level?

Onyxmoon, 2020-06-16:
I checked again.
Apparently, both brightness and CT levels are affected. It starts with 50 percent brightness and coldest color temperature (CT = 53). 

I re-enacted this once:
```08:55:31 RSL: RESULT = {"Time":"2020-06-16T08:55:31","Uptime":"0T00:02:56","UptimeSec":176,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":50,"Color":"8000","White":50,"CT":153,"Channel":[50,0],"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Satinavs Ketten","BSSId":"78:8A:20:71:E5:BA","Channel":6,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}```

This is a very specific behavior. Maybe I will reset and reconfigure the lamp again to eliminate all sources of error on my part.

ascillato2, 2020-06-19:
@Onyxmoon 

Hi, any news on this?

Onyxmoon, 2020-06-19:
I reflashed it and now it behaves as desired. I'd just like to know how that happened.

I'm sorry for wasting your time, and thanks for being so helpful. 😔

Onyxmoon, 2020-06-19:
Okay. I have identified the problem after all. As soon as I activate SetOption82 to make the CT spectrum fit Alexa, the lamp behaves incorrectly as described above.

s-hadinger, 2020-06-19:
Aha. I will take a look. It sounds like an interesting corner case

s-hadinger, 2020-06-20:
Are you able to self-compile a Tasmota version?

I would like to try changing in `xdrv_04_light.ino` line 653:

From:
```
        _ct = changeUIntScale(w, 0, sum, _ct_min_range, _ct_max_range);
```

to
```
        _ct = changeUIntScale(w, 0, sum, CT_MIN, CT_MAX);
```

If not, here is a firmware you can try: https://file.hadinger.fr/by272o

Onyxmoon, 2020-06-24:
Sorry for the late reply. The change or the provided firmware has solved the problem. :)

s-hadinger, 2020-06-24:
Good news. I will submit the PR.

PR #8779 (2020-06-24): Fix wrong CT range when Alexa mode #8694
## Description:

**Related issue (if applicable):** fixes #8694 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core ESP8266 V.2.7.1
  - [x] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The code change must pass CI tests. **Your PR cannot b [... cut]
closed 2020-06-24 by arendst