arendst/Tasmota #8706: double press - update to 8.3.1

Enjoy-Combi, 2020-06-16:
Hi,

this rule was working good before the update (from 8.2.0 to 8.3.1):

"Backlog ButtonTopic 0; SetOption1 1; SetOption11 1; SetOption32 20

Rule1 on button1#state=3 do publish cmnd/lumiere-terrasse/power 2 endon on button1#state=2 do publish cmnd/lumieres-salon-entree/power2 2 endon
 Rule1 1"

Since the update the single and long press are OK, not the double press.
When in the console there is no reaction when I test it, I mean like if I didn't do the physical double press.

Any idea ?

closed 2020-06-16 by close-issue-app[bot]

ascillato, 2020-06-16:
Hi,

From 8.2.0 to 8.3.1 there was a breaking change about that. Now you have more options. Please, check the releases notes, the documentation and do a search in issues. This was addresed before. Thanks.

If you still need help, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

effelle, 2020-06-16:
Buttons have changed on latest 8.3.x. Please head to the [documentation](https://tasmota.github.io/docs/Buttons-and-Switches/#button) to see the new behaviour.

Enjoy-Combi, 2020-06-16:
Many thanks both of you, I will have a look. Sorry for not having noticed the breaking change. And happy to discovery the new options coming with !