arendst/Tasmota #8707: Querying ShutterPosition after blinds movement triggers Power for a second or less

Zer0x00, 2020-06-16:
### PROBLEM DESCRIPTION
If I change the position of the shutters and afterwards query their position either via HTTP or Console they move for a fraction of a second either up or down.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): WF-CS02 Curtain Switch
- [X] Tasmota binary firmware version number used: 8.3.1
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Gitpod
- [X] Flashing tools used: Tuya-Convert
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
15:43:53 CMD: Backlog Template; Module; GPIO 255`
15:43:53 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/RESULT = {"NAME":"WF-CS02 Tuya","GPIO":[157,0,53,11,23,18,0,0,17,21,0,22,0],"FLAG":0,"BASE":18}
15:43:54 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/RESULT = {"Module":{"0":"WF-CS02 Tuya"}}
15:43:54 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/RESULT = {"GPIO0":{"157":"LedLink"},"GPIO1":{"0":"None"},"GPIO2":{"53":"Led2"},"GPIO3":{"11":"Switch3"},"GPIO4":{"23":"Relay3"},"GPIO5":{"18":"Button2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"17":"Button1"},"GPIO13":{"21":"Relay1"},"GPIO14":{"0":"None"},"GPIO15":{"22":"Relay2"},"GPIO16":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
15:44:41 CMD: Backlog Rule1; Rule2; Rule3
15:44:41 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":57,"Free":454,"Rules":"on Power3#State=1 do backlog ShutterStop1; Power3 0 endon"}
15:44:41 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
15:44:41 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
15:45:19 CMD: status 0
15:45:19 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/STATUS = {"Status":{"Module":0,"DeviceName":"schlafzimmer_strasse-jalousie","FriendlyName":["schlafzimmer_strasse-jalousie","schlafzimmer_strasse-jalousie2","schlafzimmer_strasse-jalousie3"],"Topic":"schlafzimmer_strasse-jalousie","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
15:45:19 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"2T13:56:51","StartupUTC":"2020-06-13T23:48:28","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2020-06-07T02:04:57","SaveCount":212,"SaveAddress":"F8000"}}
15:45:19 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/STATUS2 = {"StatusFWR":{"Version":"8.3.1(tasmota)","BuildDateTime":"2020-06-06T23:59:15","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"696/699"}}
15:45:19 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["UPC3D1EAF3_2.4GHz","bueny_2.4GHz"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000E029","2805C8000100060000005A00000000000000","40000000","00000000"]}}
15:45:19 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/STATUS4 = {"StatusMEM":{"ProgramSize":467,"Free":536,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":0,"Features":["00000809","0F800482","040083A1","00000004","010000C0","00000900","00000020"],"Drivers":"1,2,3,4,8,9,10,16,27,29","Sensors":"2,3"}}
15:45:19 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/STATUS5 = {"StatusNET":{"Hostname":"schlafzimmer_strasse-jalousie-7","IPAddress":"192.168.0.153","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"68:C6:3A:E1:FD:EE","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:45:19 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.100","MqttPort":1883,"MqttClientMask":"schlafzimmer_strasse-jalousie","MqttClient":"schlafzimmer_strasse-jalousie","MqttUser":"schlafzimmer_strasse-jalousie","MqttCount":9,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:45:19 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/STATUS7 = {"StatusTIM":{"UTC":"2020-06-16T13:45:19","Local":"2020-06-16T15:45:19","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"03:24","Sunset":"23:46"}}
15:45:19 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/STATUS10 = {"StatusSNS":{"Time":"2020-06-16T15:45:19","Switch3":"ON","Shutter1":{"Position":1,"Direction":0,"Target":1}}}
15:45:19 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/STATUS11 = {"StatusSTS":{"Time":"2020-06-16T15:45:19","Uptime":"2T13:56:51","UptimeSec":223011,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":9,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"UPC3D1EAF3_2.4GHz","BSSId":"2C:91:AB:83:BC:9B","Channel":6,"RSSI":92,"Signal":-54,"LinkCount":8,"Downtime":"0T00:02:53"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

15:39:45 HTP: Console
15:39:48 CMD: ShutterPosition
15:39:48 SRC: WebConsole from 192.168.0.20
15:39:48 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data ""
15:39:48 SHT: Pos. in: payload  (0), payload -99, idx 1, src 9
15:39:48 SHT: Start in dir 0
15:39:48 SRC: Shutter
15:39:48 SHT: Switched relay: 1 by Shutter
15:39:48 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/RESULT = {"POWER1":"ON"}
15:39:48 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/POWER1 = ON (retained)
15:39:48 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/RESULT = {"ShutterPosition1":1}
15:39:48 SHT: Shutter1 Real 3062, Start 3062, Stop 3450, Dir 1, Delay 0, Rtc 0.10 [s], Freq 0
15:39:48 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/RESULT = {"Shutter1":{"Position":1,"Direction":1,"Target":1}}
15:39:48 SRC: Shutter
15:39:48 SHT: Switched relay: 1 by Shutter
15:39:48 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/RESULT = {"POWER1":"OFF"}
15:39:48 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/POWER1 = OFF (retained)
15:39:48 SHT: Shutter1 Real 3462, Start 3062, Stop 3450, Dir 1, Delay 0, Rtc 0.20 [s], Freq 0
15:39:48 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/SHUTTER1 = 1 (retained)
15:39:48 MQT: tasmota/stat/schlafzimmer_strasse-jalousie/RESULT = {"Shutter1":{"Position":1,"Direction":0,"Target":1}}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- Change position of blinds via WebGUI.
- query their position via Console (``ShutterPosition``) or HTTP http://192.168.0.153/cm?cmnd=ShutterPosition
First you'll get a ``{"POWER2":"ON","ShutterPosition1":1}`` back, if you repeat the HTTP query you'll get ``{"Shutter1":{"Position":1,"Direction":0,"Target":1},"ShutterPosition1":1}`` back

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The query should answer with ``{"Shutter1":{"Position":2,"Direction":0,"Target":2},"ShutterPosition1":2}`` directly and not trigger any movements if there aren't specified any position values

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I can't reproduce this evey time I test this but 3 of 5 times it is reproducible.

**(Please, remember to close the issue when the problem has been addressed)**

curzon01, 2020-06-17:
can't reproduce

Preliminary check 
- no retained msg are stored for `tasmota/cmd/schlafzimmer_strasse-jalousie/#` topic, otherwise delete retained msg and eliminate the source of retained msg. Check double if retained are realy deleted..
- temporarly disable Rule1 (Rule1 0) and try reproduce

Zer0x00, 2020-06-18:
- there's no tasmota/cmd topic:
[image]
for a test I've disabled MQTT, here's the console output without MQTT:
```
02:00:28 CMD: ShutterPosition
02:00:28 SHT: Start in dir 0
02:00:28 RSL: RESULT = {"POWER2":"ON"}
02:00:28 RSL: POWER2 = ON
02:00:28 RSL: RESULT = {"ShutterPosition1":3}
02:00:28 SHT: Shutter1 Real 9602, Start 9602, Stop 9202, Dir -1, Delay 0, Rtc 0.05 [s], Freq 0
02:00:28 RSL: RESULT = {"Shutter1":{"Position":3,"Direction":-1,"Target":3}}
02:00:28 RSL: RESULT = {"POWER2":"OFF"}
02:00:28 RSL: POWER2 = OFF
02:00:28 SHT: Shutter1 Real 9214, Start 9602, Stop 9202, Dir -1, Delay 0, Rtc 0.20 [s], Freq 0
02:00:28 RSL: SHUTTER1 = 3
02:00:28 RSL: RESULT = {"Shutter1":{"Position":3,"Direction":0,"Target":3}}
```
- disabling Rule1 also didn't change anything

Thanks!

curzon01, 2020-06-18:
- Pulsetime 0?
`Backlog PulseTime1 0; PulseTime2 0`
- what's your shutter config?
 (use  e.g. [decode-config](https://github.com/tasmota/decode-config):
`decode-config -d "schlafzimmer_strasse-jalousie-7" -p "<webpassword>" -g Shutter`)

Zer0x00, 2020-06-18:
> * Pulsetime 0?
Didn't change anything.

> * what's your shutter config?
```json
{
   "header":{
      "data":{
         "crc":"0x633",
         "crc32":"0xc7b27a01",
         "platform":"ESP82xx",
         "size":4096,
         "template":{
            "crc":"0x828a",
            "crc32":"0xc7b27a01",
            "size":4096,
            "version":"0x8030100"
         },
         "version":"0x8030100"
      },
      "env":{
         "platform":"Windows-10-10.0.19041-SP0",
         "python":"3.6.8",
         "script":"decode-config.py v8.3.1.0 [00066]",
         "system":"Windows AMD64 10 10.0.19041"
      },
      "timestamp":"2020-06-18 12:05:33"
   },
   "shutter_accuracy":1,
   "shutter_button":[
      {
         "enabled":0,
         "mqtt_broadcast_all":0,
         "mqtt_broadcast_double":0,
         "mqtt_broadcast_hold":0,
         "mqtt_broadcast_single":0,
         "mqtt_broadcast_triple":0,
         "press_double":"-",
         "press_hold":"-",
         "press_single":"-",
         "press_triple":"-",
         "shutter":1
      },
      {
         "enabled":0,
         "mqtt_broadcast_all":0,
         "mqtt_broadcast_double":0,
         "mqtt_broadcast_hold":0,
         "mqtt_broadcast_single":0,
         "mqtt_broadcast_triple":0,
         "press_double":"-",
         "press_hold":"-",
         "press_single":"-",
         "press_triple":"-",
         "shutter":1
      },
      {
         "enabled":0,
         "mqtt_broadcast_all":0,
         "mqtt_broadcast_double":0,
         "mqtt_broadcast_hold":0,
         "mqtt_broadcast_single":0,
         "mqtt_broadcast_triple":0,
         "press_double":"-",
         "press_hold":"-",
         "press_single":"-",
         "press_triple":"-",
         "shutter":1
      },
      {
         "enabled":0,
         "mqtt_broadcast_all":0,
         "mqtt_broadcast_double":0,
         "mqtt_broadcast_hold":0,
         "mqtt_broadcast_single":0,
         "mqtt_broadcast_triple":0,
         "press_double":"-",
         "press_hold":"-",
         "press_single":"-",
         "press_triple":"-",
         "shutter":1
      }
   ],
   "shutter_closetime":[
      210,
      0,
      0,
      0
   ],
   "shutter_motordelay":[
      0,
      0,
      0,
      0
   ],
   "shutter_opentime":[
      205,
      0,
      0,
      0
   ],
   "shutter_options":[
      0,
      0,
      0,
      0
   ],
   "shutter_position":[
      3,
      0,
      0,
      0
   ],
   "shutter_set50percent":[
      66,
      0,
      0,
      0
   ],
   "shutter_startrelay":[
      1,
      0,
      0,
      0
   ],
   "shuttercoeff":[
      [
         13200,
         0,
         0,
         0
      ],
      [
         278,
         0,
         0,
         0
      ],
      [
         2918,
         0,
         0,
         0
      ],
      [
         900,
         0,
         0,
         0
      ],
      [
         1000,
         0,
         0,
         0
      ]
   ]
}
```

curzon01, 2020-06-18:
I'm not deep enough into the Shutter module src but my shuttercoeff values are always increased - your are looking very randomly. Tried a new calibration from scratch for this shutter? Reinit [Shutter Mode](https://tasmota.github.io/docs/Blinds-and-Shutters/#shutter-modes) and repeat [calibration](https://tasmota.github.io/docs/Blinds-and-Shutters/#calibration). You can also preliminary reset values by restore your json-part from above using decode-config setting all to 0 and then re-calibrate.

Zer0x00, 2020-06-18:
I'm having this issue with all of my shutters and I've calibrated some of them just a couple of days ago starting with a fresh install. Also I've calibrated it again from scrath after realizing that this behaviour is not triggered by a bug of my homebridge plugin.

curzon01, 2020-06-18:
Can only try to help by try-n-error. My 10+ shutter (all different hardware, Sonoff Dual, Wemos) does not have this, not with 8.2.0, 8.3.1 or the newest 8.3.1.6. Tried the release bin (not self-compiled)?

Zer0x00, 2020-06-18:
No problem, try-n-error should be fine.

The release bin does not have the shutter option included afaik so no, din't tried that.

curzon01, 2020-06-18:
release does have shutter included see [available features](https://github.com/arendst/Tasmota/blob/v8.3.1/BUILDS.md) - just checked tasmota.bin.gz 8.3.1 from [release](https://github.com/arendst/Tasmota/releases), it's in

Zer0x00, 2020-06-18:
Oh, seems like it was changed since I've looked it up!

I'll flash the default image and test with it.

curzon01, 2020-06-21:
any progress with origin fw?

Zer0x00, 2020-06-22:
Sorry for the delay, I've tested with the origin fw with no success so far, same results.

curzon01, 2020-06-22:
Ist Looks Like your steps are not identical to the describes calibration process

Zer0x00, 2020-06-22:
I've tried again with this runbook:

```
Reset 6 (to start with a fresh config)

Set template to:
{"NAME":"WF-CS02 Tuya","GPIO":[157,0,53,11,23,18,0,0,17,21,54,22,52],"FLAG":0,"BASE":18}

SetOption80 1 # Enable shutter support, reset the device after for correct interface to show up

ShutterRelay1 1 # Enable shutter relay 1

Backlog PulseTime1 0; PulseTime2 0 # ShutterMode0

Backlog Interlock 1,2,3; Interlock ON

PowerRetain 1 # Save state for Home Assistant (works only if MQTT is enabled)

Rule1 on Power3#State=1 do backlog ShutterStop1; Power3 0 endon # Use the middle button for stopping shutter movement
Rule1 1

ShutterOpenDuration 20.5
ShutterCloseDuration 21
ShutterSetHalfway 66
```

Am I missing something?

curzon01, 2020-06-23:
Yes, as already wrote you do not follow the [Calibration](https://tasmota.github.io/docs/Blinds-and-Shutters/#calibration) procedure - you follow the first to items of the procudure then yo break it and set ShutterSetHalfway 66 direcly.

Again, 

- try this with an fresh uncalbrated device using origin fw
- follow the linked procedure step by step (you don't need the video)

Zer0x00, 2020-06-24:
I've tried to only set ShutterOpenDuration with ShutterCloseDuration, setting your suggested template and SwitchMode's.
No success so far :/

Thank you any ways for your help!

curzon01, 2020-06-25:
What I'm missing is the weblog 4 output of the complete activity you originaly described including prev. origin shutter possition and WebUI usage). That's why I identified the src position where the micro-movement starts but not why (due to missing previous states).
You posted logs at the moment after you used the WebUI slider and set the command `ShutterPosition` missing the previous task. 

Can you log a complete procedure where this happens:
- set `Weblog 4`
- call `ShutterPosition`
- set new position using GUI/http command
- call `ShutterPosition`

Zer0x00, 2020-06-25:
Here's the log output with the steps described by you:
```
20:01:41 HTP: Console
20:01:44 HTP: Command
20:01:44 SRC: WebCommand from 192.168.0.20
20:01:44 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data ""
20:01:44 SHT: Pos. in: payload  (0), payload -99, idx 1, src 8
20:01:44 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"Shutter1":{"Position":100,"Direction":0,"Target":99}}
20:01:44 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"ShutterPosition1":100}
20:01:45 SRC: WebGui from 192.168.0.20
20:01:45 SRC: WebGui from 192.168.0.20
20:01:45 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data "StopClose"
20:01:45 SHT: Pos. in: payload StopClose (9), payload -99, idx 1, src 7
20:01:45 SHT: Pos. in: payload StopClose (0), payload 0, idx 1, src 7
20:01:45 SHT: lastsource 7:, real 39000, target 0, payload 0
20:01:45 SHT: Start in dir 0
20:01:45 SRC: Shutter
20:01:45 SHT: Switched relay: 2 by Shutter
20:01:45 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"POWER2":"ON"}
20:01:45 MQT: tasmota/stat/wohnzimmer-jalousie/POWER2 = ON
20:01:45 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"ShutterPosition1":0}
20:01:46 SHT: Shutter1 Real 39000, Start 39000, Stop 0, Dir -1, Delay 0, Rtc 0.10 [s], Freq 0
20:01:46 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"Shutter1":{"Position":100,"Direction":-1,"Target":0}}
20:01:46 SHT: Shutter1 Real 37940, Start 39000, Stop 0, Dir -1, Delay 0, Rtc 0.50 [s], Freq 0
20:01:46 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"Shutter1":{"Position":96,"Direction":-1,"Target":0}}
20:01:46 CFG: Saved to flash at FA, Count 122, Bytes 4096
20:01:47 SRC: WebGui from 192.168.0.20
20:01:47 SHT: Switched relay: 2 by I
20:01:47 SHT: Shutter 1: Switch OFF motor. Target: 36880, source: I, powerstate_local 0, Shutter.switched_relay 2, manual change 1
20:01:47 SHT: Shutter 1: Target: 36880, powerstatelocal 0
20:01:47 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"POWER2":"OFF"}
20:01:47 MQT: tasmota/stat/wohnzimmer-jalousie/POWER2 = OFF
20:01:47 SHT: Switched relay: 4 by I
20:01:47 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"POWER3":"ON"}
20:01:47 MQT: tasmota/stat/wohnzimmer-jalousie/POWER3 = ON
20:01:47 RUL: POWER3#STATE=1 performs "backlog ShutterStop1; Power3 0"
20:01:47 SRC: Rule
20:01:47 CMD: Group 0, Index 1, Command "BACKLOG", Data "ShutterStop1; Power3 0"
20:01:47 SHT: Shutter1 Real 36350, Start 39000, Stop 36880, Dir -1, Delay 0, Rtc 1.25 [s], Freq 0
20:01:47 MQT: tasmota/stat/wohnzimmer-jalousie/SHUTTER1 = 90
20:01:47 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"Shutter1":{"Position":90,"Direction":0,"Target":92}}
20:01:47 SRC: Backlog
20:01:47 CMD: Group 0, Index 1, Command "SHUTTERSTOP", Data ""
20:01:47 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"ShutterStop":"Done"}
20:01:47 SRC: Backlog
20:01:47 CMD: Group 0, Index 3, Command "POWER", Data "0"
20:01:47 SHT: Switched relay: 4 by I
20:01:47 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"POWER3":"OFF"}
20:01:47 MQT: tasmota/stat/wohnzimmer-jalousie/POWER3 = OFF
20:01:47 CFG: Saved to flash at F9, Count 123, Bytes 4096
20:01:48 HTP: Command
20:01:48 SRC: WebCommand from 192.168.0.20
20:01:48 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data ""
20:01:48 SHT: Pos. in: payload  (0), payload -99, idx 1, src 8
20:01:48 SHT: Start in dir 0
20:01:48 SRC: Shutter
20:01:48 SHT: Switched relay: 1 by Shutter
20:01:48 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"POWER1":"ON"}
20:01:48 MQT: tasmota/stat/wohnzimmer-jalousie/POWER1 = ON
20:01:48 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"ShutterPosition1":90}
20:01:48 SHT: Shutter1 Real 36350, Start 36350, Stop 36880, Dir 1, Delay 0, Rtc 0.10 [s], Freq 0
20:01:48 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"Shutter1":{"Position":90,"Direction":1,"Target":92}}
20:01:48 SRC: Shutter
20:01:48 SHT: Switched relay: 1 by Shutter
20:01:48 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"POWER1":"OFF"}
20:01:48 MQT: tasmota/stat/wohnzimmer-jalousie/POWER1 = OFF
20:01:48 SHT: Shutter1 Real 36950, Start 36350, Stop 36880, Dir 1, Delay 0, Rtc 0.30 [s], Freq 0
20:01:48 MQT: tasmota/stat/wohnzimmer-jalousie/SHUTTER1 = 92
20:01:48 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"Shutter1":{"Position":92,"Direction":0,"Target":92}}
20:01:48 CFG: Saved to flash at F8, Count 124, Bytes 4096
```

curzon01, 2020-06-25:
and what was your GUI action?

Zer0x00, 2020-06-25:
Click on Button for move down, then stop it by clicking the stop button

curzon01, 2020-06-26:
I was able to reproduce the problem, but not always (e.g. I had to handle very fast and using the slider I was unable to reproduce it, thats why I ask in detail what you have done).

If the target position does not correspond to the current one, ShutterPosition then triggers and switches the motor on for a fraction of second. This is the case with the origin rule when using ButtonDown (Power2) then ButtonStop (Power3) within WebUI:

20:01:48 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data ""
...
20:01:48 SHT: Shutter1 Real 36350, Start 36350, Stop 36880, Dir 1, Delay 0, Rtc 0.10 [s], Freq 0
20:01:48 MQT: tasmota/stat/wohnzimmer-jalousie/RESULT = {"Shutter1":{**"Position":90**,"Direction":1,**"Target":92**}}

The following ShutterPosition query cmnd internaly triggers the short power on/off to match the target position.

I'd not realy found the reasons why current position and target are different in this cases but the rule `Backlog`  might be an issue. I solved it preventing Backlog within stop button rule.

Try the following
```
Pulsetime3 1 # auto disabled Power3 button
Rule1 on Power3#State=1 do ShutterStop1 endon # do not use backlog for stop shutter
```

Zer0x00, 2020-06-30:
@curzon01 Thank you, this worked very well! <3

closed 2020-06-30 by Zer0x00