arendst/Tasmota #8711: Script-language timer imprecise

timoewald, 2020-06-16:
### PROBLEM DESCRIPTION
I am now trying to bring an old rules "script" into the script language. However, I have a problem with the timers.

Here is a sample code just to find the effect and possibly inaccuracy of the timers. Or just to show my possible mistake:

```
>D   ; define all vars here
t:timer1=2
t:timer2=-1

>F   ; Executed every second
if timer1==0
then
=>print "timer1 == 0"
timer1=-1
timer2=10
=>print "timer2 = 10 gesetzt"
endif

if timer2==0
then
=>print "timer2 == 0"
timer2=-1
timer1=7
=>print "timer1 = 7 gesetzt"
endif
```
and the output of the console:
```
23:36:07  "timer1 == 0"
23:36:07  "timer2 = 10 gesetzt"
23:36:24  "timer2 == 0"
23:36:24  "timer1 = 7 gesetzt"
23:36:36  "timer1 == 0"
23:36:36  "timer2 = 10 gesetzt"
23:36:55  "timer2 == 0"
23:36:55  "timer1 = 7 gesetzt"
23:37:09  "timer1 == 0"
23:37:09  "timer2 = 10 gesetzt"
23:37:27  "timer2 == 0"
23:37:27  "timer1 = 7 gesetzt"
23:37:39  "timer1 == 0"
23:37:39  "timer2 = 10 gesetzt"
23:38:00  "timer2 == 0"
```
As you can see from the output in the console, the time information does not match the set timer time.
The actual time intervals are often twice as long. But not consistently but sometimes.

With my rather simple script, only these timers run?
Where's the mistake? Maybe I am making a mistake in the syntax or I have misunderstood the entry.

[https://tasmota.github.io/docs/Scripting-Language/#commands](url) says:
> t:vname
specifies countdown timers, if >0 they are decremented in seconds until zero is reached.
 

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Pow R2
- [x] Tasmota binary firmware version number used: Tasmota 8.3.1.4
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: VisualStudio Code
- [x] Flashing tools used: Espflasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
00:02:27 MQT: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
00:02:27 MQT: stat/sonoff/RESULT = {"Module":{"43":"Sonoff Pow R2"}}
00:02:28 MQT: stat/sonoff/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"145":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"146":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:

- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
00:03:43 CMD: Status 0
00:03:43 MQT: stat/sonoff/STATUS = {"Status":{"Module":43,"DeviceName":"","FriendlyName":[""],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:03:43 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/020300/sonoff-DE.bin","RestartReason":"Software/System restart","Uptime":"0T04:14:01","StartupUTC":"2020-06-16T17:49:42","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2020-05-15T21:06:56","SaveCount":259,"SaveAddress":"F4000"}}
00:03:43 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"8.3.1.4(tasmota)","BuildDateTime":"2020.06.16 11:29:06","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"388/699"}}
00:03:43 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Klieber",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100068000005A00000000000000","00000000","00006000"]}}
00:03:43 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":617,"Free":384,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000407","870AA793","043C83A1","0000008D","010013C0","C000F981","00004024"],"Drivers":"1,2,3,4,5,6,9,10,16,20,21,22,24,27,29,30,35,37","Sensors":"1,2,3,4,6"}}
00:03:43 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4722","IPAddress":"192.168.178.43","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"EC:FA:BC:4F:B2:72","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:03:43 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.30","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4FB272","MqttUser":"Klieber","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:03:43 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"2020-06-16T22:03:43","Local":"2020-06-17T00:03:43","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"05:13","Sunset":"21:35"}}
00:03:43 MQT: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:03:43 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2020-06-17T00:03:43","ENERGY":{"TotalStartTime":"2020-05-15T21:06:56","Total":0.038,"Yesterday":0.014,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
00:03:43 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2020-06-17T00:03:43","Uptime":"0T04:14:01","UptimeSec":15241,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Klieber","BSSId":"E0:28:6D:D2:7E:BD","Channel":6,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_

  Console output here:
It is the same output that I added above as console output. I don´t get any error messages or so, just a wrong(?) time.

### TO REPRODUCE
Use timers in the scripting language the way I did above
=> start a timer and get an output when the time runs out

### EXPECTED BEHAVIOUR
I expected a timer with 10 seconds ends after 10 seconds with timer==0. Or that, as in my script, the output comes after 10 seconds. And not 20 or 16.

gemu2015, 2020-06-17:
strange, will investigate.
however it is inaccurate by design to +1 seconds, but should not be so far off.

better use the new ticker timers (up to 4) which mostly are within milliseconds precision
#define USE_SCRIPT_TIMERS

ts1(ms)  set timer 1 to ms time (ms>=10)

\>ti1
print timer stopped here
; may repeat here
ts1(ms)

gemu2015, 2020-06-17:
could not find such long intervals
try below script to monitor your timers

\>D   ; define all vars here
t:timer1=2
t:timer2=-1

 ; Executed every second
\>S
dp5
print t1=%timer1% - t2=%timer2%

if timer1==0
then
print "timer1 == 0"
timer1=-1
timer2=10
print "timer2 = 10 gesetzt"
endif

if timer2==0
then
print "timer2 == 0"
timer2=-1
timer1=7
print "timer1 = 7 gesetzt"
endif

timoewald, 2020-06-17:
> ´>S
 dp5
 print t1=%timer1% - t2=%timer2%

good idea with displaying the timer! However, it only becomes clear that the timers always take about twice as long.

Even if I do it with `>F`, which executes every 100ms 100ms, exactly the same thing happens. For 1 second to pass, in reality it takes about 2 seconds:
```
10:13:49 t1=-1.00000 - t2=6.00000
10:13:49 t1=-1.00000 - t2=5.00000
10:13:49 t1=-1.00000 - t2=5.00000
10:13:49 t1=-1.00000 - t2=5.00000
10:13:49 t1=-1.00000 - t2=5.00000
10:13:49 t1=-1.00000 - t2=5.00000
10:13:50 t1=-1.00000 - t2=5.00000
10:13:50 t1=-1.00000 - t2=5.00000
10:13:50 t1=-1.00000 - t2=5.00000
10:13:50 t1=-1.00000 - t2=5.00000
10:13:50 t1=-1.00000 - t2=5.00000
10:13:50 t1=-1.00000 - t2=5.00000
10:13:50 t1=-1.00000 - t2=5.00000
10:13:50 t1=-1.00000 - t2=5.00000
10:13:50 t1=-1.00000 - t2=5.00000
10:13:50 t1=-1.00000 - t2=5.00000
10:13:51 t1=-1.00000 - t2=5.00000
10:13:51 t1=-1.00000 - t2=5.00000
10:13:51 t1=-1.00000 - t2=5.00000
10:13:51 t1=-1.00000 - t2=5.00000
10:13:51 t1=-1.00000 - t2=5.00000
10:13:51 t1=-1.00000 - t2=4.00000
10:13:51 t1=-1.00000 - t2=4.00000
```

gemu2015, 2020-06-17:
you should NOT use the \>F section to update the timers, they are decremented directly before the >S section.
With my script above i do not see this double time effect.
very strange that you have no decimals other than zero.
the timer measures the difference between second ticks in milliseconds and decrements the timer with this difference. This is often not exactly 1 second.

09:37:17 "timer1 == 0"
09:37:17 "timer2 = 10 gesetzt"
09:37:18 t1=-1.00000 - t2=9.00000
09:37:19 t1=-1.00000 - t2=7.99600
09:37:20 t1=-1.00000 - t2=6.99400
09:37:21 t1=-1.00000 - t2=5.98800
09:37:22 t1=-1.00000 - t2=4.98500
09:37:23 t1=-1.00000 - t2=4.01800
09:37:24 t1=-1.00000 - t2=3.01800
09:37:25 t1=-1.00000 - t2=1.97100
09:37:26 t1=-1.00000 - t2=1.01500
09:37:27 t1=-1.00000 - t2=0.01400
09:37:28 t1=-1.00000 - t2=0.00000

timoewald, 2020-06-17:
That with >F makes sense.

But it's strange anyway. Even if I use other logging settings, I never get decimal places other than zero. And also with your script, I experience this enormous delay.

gemu2015, 2020-06-17:
this is very strange!
please post your user config overwrite settings. I will try to reproduce this behavior.

timoewald, 2020-06-17:
[user_config_override.txt](https://github.com/arendst/Tasmota/files/4791676/user_config_override.txt)

I hope this works with the file. Had to convert them to a .txt.
And of course I hope that the settings I made basically makes sense ;)

gemu2015, 2020-06-17:
the decimal places being zero is the normal (wrong) behavior. When i started debugging this morning i made a small change that corrected this. Sorry i forgot (was too tired this morning)
BUT this does not change the double time bug you see.
Even with your setup i can't reproduce this.
Probably you should download my fork and test if it works.
If it does we have to figure out what the difference is.

https://github.com/gemu2015/Sonoff-Tasmota

gemu2015, 2020-06-17:
@timoewald 
tried again and this was indeed the reason for the bug, sorry again.
tested this function several times and it worked well, but depending on some timing it just decremented the timer by zero.
this line caused the bug:
was:  float delta=(millis()-script_lastmillis)/1000;
 should be: float delta=(millis()-script_lastmillis)/1000.0;
i will make a pr soon

PR #8714 (2020-06-17): scripter timer fix
## Description:

fixes t: and i: timer missing counts

**Related issue (if applicable):** fixes #8711

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core ESP8266 V.2.7.1
  - [x] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The code change must pass CI tests. **Y [... cut]
closed 2020-06-17 by arendst