arendst/Tasmota #8715: ESP32 MESH Network

kugelkopf123, 2020-06-17:
**Have you looked for this feature in other issues and in the docs?**  
Yes, there is one for the ESP8266.  #7401 But this one is for the ESP32 hast two interfaces, one as client and one as AP. 

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

The world of IoT continues to grow and with it the demand for suitable hardware and of course software. 
In my own house I use the Zigbee protocol most of the time. Because of the MeshNetworking. My Zigbee Gateway is on the first floor of my house and through the mesh of the Zigbee devices here in the house I have reception until the last end point (Osram Plug) which is about 60m away at the end of my garden. For comparison, my Fritzbox (router) is in the same room and just manages to cover the front part of the house.
Espressif has published a tool for its IDF to develop these mesh networks. Maybe that would help Tasmota too. Unfortunately I could not find any examples for the Arduino IDE. Nevertheless, parts of the SDK are already present in the Arduino ESP32Core. 
By the way, besides the possibility to create this MESH via WiFi, a Bluetooth Mesh would also be interesting. 
I'm not interested in carrying my Internet to the last corner of the garden. Rather, it is because I can use the Tasmota devices regardless of the location of a router or repeater. 

Here I have added a few links with information. 

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/mesh.html#tree-topology
https://github.com/espressif/esp-mdf

https://www.espressif.com/en/news/ESP_BLE_MESH_SIG_Certified

Arduino:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/esp32/esp_mesh.h

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
The Same Idea with the ESP32 integrated BT.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-06-18:
Wrong place for this feature request. Tasmota uses network functions from Arduino.
So in first step Mesh  support has to be there first

kugelkopf123, 2020-06-18:
>Wrong place for this feature request. Tasmota uses network functions from Arduino. ...

@Jason2866 Don’t think so. The sdk is already implemented. But there is no example. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Jason2866, 2020-06-18:
Yes, SDK but there is NO Arduino implementation.

kugelkopf123, 2020-06-18:
>Yes, SDK but there is NO Arduino implementation.  ...

@Jason2866 
Oh okay. So this: https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/esp32/esp_mesh.h https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/esp32/esp_mesh.h
Isn’t the right stuff for that? 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Jason2866, 2020-06-18:
Everything in folder SDK is Espressif stuff. The Arduino layer "translates" to Espressif SDK layer
This translation layer part for mesh is missing.

ascillato, 2020-06-18:
Please, open a new issue in the Arduino ESP32 Core for adding Mesh. Thanks.

meingraham, 2020-06-20:
New Broadlink Wi-Fi smart bulbs & plugs - [Paul Hibbert's latest](https://www.youtube.com/watch?v=lzhLFpG71Rc). The No App feature is basically what we already do to configure Tasmota. Yawn. It's the other features that Paul announced Wi-Fi mesh and auto-configure. Unfortunately he didn't actually demonstrate them in the video 😡 The point of my post is more about the new relatively inexpensive Broadlink devices that they claim to have these features. Definitely sounds interesting.

kugelkopf123, 2020-06-23:
I haven't posted anything in the git of the arduino esp32 yet. It seems to me that with >190 open issues they have other things to do than adding features. what do you think? How would you go about it? @meingraham

meingraham, 2020-06-23:
First thing to research is what kind of Wi-Fi SOC is in the Broadlink devices.

ascillato, 2020-06-23:
> I haven't posted anything in the git of the arduino esp32 yet. It seems to me that with >190 open issues they have other things to do than adding features. what do you think?

May be we can ask if it is in their plans to add it.

kugelkopf123, 2020-06-23:
@ascillato Does any of you have contact to the developers? Or should I ask this as an issue?

ascillato, 2020-06-24:
Sorry, no. We should open an issue there.

kugelkopf123, 2020-07-04:
@ascillato 
https://github.com/espressif/arduino-esp32/issues/4132#issuecomment-653254432
Is this perhaps an option?

ascillato, 2020-07-04:
Yes, indeed that is what was needed for adding mesh to Tasmota. Now, to become a reality there is a lot of rework to Tasmota to support that because Tasmota was made for mqtt as its core and with mesh you need to make the nodes to repeat mqtt messages or to develop another protocol for them.

Reopening this issue to see if someone is interested on working on this POC.

kugelkopf123, 2020-07-04:
BTW there is another great Library I found:

https://gitlab.com/painlessMesh/painlessMesh

closed 2020-08-08 by stale[bot]

jnherm, 2021-05-29:
Does someone working on this? I am interested.

Jason2866, 2021-05-29:
@Staars was working on this. He had a working PoC. Stopped working on that. https://gist.github.com/Staars/42f8b41ddc593d25db868b30b8c28721
So the answer is no.