arendst/Tasmota #8720: Web Password too long

tom2199, 2020-06-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I typed a web password in that was 128 chars, now I can't log in into the web ui.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Aisirer RGBW Modul
- [X] Tasmota binary firmware version number used: 8.3.1
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
09:57:13 RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":32,"Free":479,"Rules":"ON Dimmer#Boot DO Power ON ENDON"}
09:57:14 RSL: RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
09:57:14 RSL: RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Set web password over 32 chars

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
That the password would be saved. Of course, if it's too long, don't save it or save the first 32 chars.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I know it was stupid that I didn't read the docs beforehand, but power cycling doesn't work (maybe because of 8.3.1? I know it did work before) so this is my last chance. Is there any way to recover the web password? I have the URL and the full password I typed in.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-06-18:
Just do a serial flash. Dont forget to erase flash before flashing Tasmota.

arendst, 2020-06-18:
A bit longer password should work. I won't duscuss if 125 character is a bit but hey, it's your device.

If you have access to the device by serial connection or mqtt you can simply execute command ``webpassword "`` to erase the password.

tom2199, 2020-06-18:
Sadly I don't have a mqtt setup and i cannot serial flash it without destroying the device more or less. Is there any chance to know which password was saved by tasmota? The password I used was: "WJz1eZnKP9snSj6231Wchm3kkwKtEvDLwgxdCjmNOwG00TKR2IulEtGmMPi8FmpIIUKOwBTwobtpJH9AJcQxR4d05jCntfxVSlR3Q87fACAwA0wyBgBZoxBMWnbsSHGO"

arendst, 2020-06-18:
I just tried your password on one of my devices and it worked!

Are you sure you used username ``admin``?

You might want to try to use the ``decode-config.py`` tool which needs the same password. Using a command line like 
```
python decode-config.py -d your-device-name -o config-dump-file -t bin --ignore-warnings -p WJz1eZnKP9snSj6231Wchm3kkwKtEvDLwgxdCjmNOwG00TKR2IulEtGmMPi8FmpIIUKOwBTwobtpJH9AJcQxR4d05jCntfxVSlR3Q87fACAwA0wyBgBZoxBMWnbsSHGO
```
It dumps the config and verifies a correct password. No real use in your case.

Anyway, the quick power toggles should still work so I suggest you just try again at least 7 quick power toggles.

tom2199, 2020-06-18:
> I just tried your password on one of my devices and it worked!
>
> Are you sure you used username `admin`?
>
> You might want to try to use the `decode-config.py` tool which needs the same password. Using [...]

Yes I used admin as the username, here is the complete URL that was send to my device:
```
http://192.168.178.39/co?t1=%7B%22NAME%22%3A%22Aisirer+RGBW%22%2C%22GPIO%22%3A%5B0%2C0%2C0%2C0%2C37%2C40%2C0%2C0%2C38%2C0%2C39%2C0%2C0%5D%2C%22FLAG%22%3A0%2C%22BASE%22%3A18%7D&wp=WJz1eZnKP9snSj6231Wchm3kkwKtEvDLwgxdCjmNOwG00TKR2IulEtGmMPi8FmpIIUKOwBTwobtpJH9AJcQxR4d05jCntfxVSlR3Q87fACAwA0wyBgBZoxBMWnbsSHGO&dn=Tischlampe&a0=Tischlampe&b2=2&save=
```

The decode-config.py gets me this result:
```
Load data from device '192.168.178.39'
ERROR 401 (@2623): Error on http GET request for http://192.168.178.39/dl - Unauthorized
Premature exit -  #401
```
So as expected. The power toggle doesn't work for some reason, I know it is not part of this issue but I recorded a short video:
https://www.youtube.com/watch?v=-5VXfmDTIWk
The lamp shouldn't turn on the 7th time, because tasmota should reset all the settings, or am I wrong? Am I doing something wrong? I've resetted multiple devices by mistake in the past, and now I can't do it even if I want to.

arendst, 2020-06-18:
So for some reason the password is stored wrong considering the ``Unauthorized`` message.

I think your quick power cycles are too slow. make them a lot quicker pls.

tom2199, 2020-06-18:
> So for some reason the password is stored wrong considering the `Unauthorized` message.
> 
> I think your quick power cycles are too slow. make them a lot quicker pls.

Thank you for your time, I appreciate the effort for trying to help me!
I recoreded it again, am I still doing something wrong? https://www.youtube.com/watch?v=Ec0Zin2GWNQ

arendst, 2020-06-18:
Pls try eight times

tom2199, 2020-06-18:
> Pls try eight times

Here's the video: https://www.youtube.com/watch?v=1SWXyMJ9Ff8
I even tried 9 and 10 times, still nothing unfortunately. Does it have something to do with the rule I've set up?

arendst, 2020-06-18:
No. The rule should not be interfering.

Sorry, out of ideas

tom2199, 2020-06-18:
So there is no option to recover my password? That's a shame, I thought if we could reproduce the steps it would be possible. Thank you for your help.

closed 2020-06-18 by ascillato2