arendst/Tasmota #8725: Configuring two APs / SSIDs, which is selected?

JsBergbau, 2020-06-18:
You can configure 2 APs or SSIDs in parallel. However the FAQ https://tasmota.github.io/docs/FAQ/ doesn't state how the device is behaving.
Is second SSID only for backup? If so when SSID 1 is available again, will it switch to SSID 1? Or is it continoulsy connected with both Wifis, accepting MQTT commands on both?

closed 2020-06-18 by close-issue-app[bot]

meingraham, 2020-06-18:
It is only connected to one AP at any one time. So56 & So57 allow Tasmota to switch which AP it connects to based on best signal.

ascillato, 2020-06-18:
Tasmota connects to one AP only. ESP8266 chips can only connect to one AP, they have just one antenna. Tasmota connects to the first AP you configure. Then, the behaviour depends on what you have selected using command WIFICONFIG. Please, check the docs for more information.

If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

JsBergbau, 2020-06-18:
Ok, I've found out. You have to set WifiConfig to 4 `4 = disable Wi-Fi Manager but retry the other AP without rebooting` then it won't create an AP to connect, but switches to alternate AP. But obviously it connects only back to AP1 when AP2 is gone. 
https://tasmota.github.io/docs/Commands/

chaseadam, 2024-11-25:
for those who find this ticket wanting to query or force a switch to a specific SSID enty (SSID1 or SSID2), there is an (undocumented?) `ap` cmnd which will show which entry is selected and allow you to specify "1" or "2" to connect to the access point.

The last known connected access point is marked as `sta_active` in the configuration and will be restored and used in the default configuration until something triggers the unit to "switch" to the other AP (such as SSID missing).