arendst/Tasmota #8730: QS-WiFi-D01-TRIAC double click feature

lutierigb, 2020-06-19:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
No.

**Describe the solution you'd like**  
In the script used by the dimmer implement a double click/touch function that will quickly set the dimmer to an upper/lower bound. Press the momentary switch twice in one second it will dim to 100%. Press the momentary switch twice in one second again, it jumps to a lower level like 30%.
 
This would make the user to quickly switch between low light and high light and still allow the long press to increase/decrease.

**Describe alternatives you've considered**  
I tried to do it myself but I'm not very handy with scripting.

**Additional context**  
no

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-06-19:
It is already supported using rules on the standard Tasmota precompiled bins.

Please, check the documentation at: https://tasmota.github.io/docs/Rules/#button-single-press-double-press-and-hold and also at https://tasmota.github.io/docs/Buttons-and-Switches/

If you need further assistance on this, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

lutierigb, 2020-06-19:
For this dimmer, rules are not used or required and I believe a double click feature would have to implement as part of the script so it has context and can take the right action. for example, this is how the short press is implemented:

```
>F
cnt=pc[1]
if chg[cnt]>0
; sw pressed
then sw=1
else sw=0
; sw not pressed
endif

; 100ms timer
tmr+=1
; short press
if sw==0
and tmr>shortprl
and tmr<shortpru
...
```

link to the script: https://templates.blakadder.com/qs-wifi_D01_dimmer.html

closed 2020-07-25 by stale[bot]