arendst/Tasmota #8734: SlaveSend resets the device

ghiglie, 2020-06-19:
### PROBLEM DESCRIPTION
On Sonoff SC hardware (ESP8266+ATMega328p, connected via serial with level shifter, so no reset pin connection), console command SlaveSend resets the ESP.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff SC
- [x] Tasmota binary firmware version number used: 8.3.1
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [x] IDE / Compiler used: tasmota-docker
- [X] Flashing tools used: tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
09:37:46 SRC: WebConsole from 192.168.188.150
09:37:46 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
09:37:46 SRC: Backlog
09:37:46 CMD: Group 0, Index 1, Command "RULE", Data ""
09:37:46 RSL: stat/tasmota_F8AF5F/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
09:37:46 SRC: Backlog
09:37:46 CMD: Group 0, Index 2, Command "RULE", Data ""
09:37:46 RSL: stat/tasmota_F8AF5F/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
09:37:47 SRC: Backlog
09:37:47 CMD: Group 0, Index 3, Command "RULE", Data ""
09:37:47 RSL: stat/tasmota_F8AF5F/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
09:38:18 CMD: Backlog Template; Module; GPIO 255
09:38:18 SRC: WebConsole from 192.168.188.150
09:38:18 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
09:38:18 SRC: Backlog
09:38:18 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
09:38:18 RSL: stat/tasmota_F8AF5F/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
09:38:18 SRC: Backlog
09:38:18 CMD: Group 0, Index 1, Command "MODULE", Data ""
09:38:18 RSL: stat/tasmota_F8AF5F/RESULT = {"Module":{"18":"Generic"}}
09:38:18 SRC: Backlog
09:38:18 CMD: Group 0, Index 1, Command "GPIO", Data "255"
09:38:18 RSL: stat/tasmota_F8AF5F/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"184":"Slave TX"},"GPIO5":{"185":"Slave RX"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
09:37:46 CMD: Backlog Rule1; Rule2; Rule3
09:37:46 SRC: WebConsole from 192.168.188.150
09:37:46 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
09:37:46 SRC: Backlog
09:37:46 CMD: Group 0, Index 1, Command "RULE", Data ""
09:37:46 RSL: stat/tasmota_F8AF5F/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
09:37:46 SRC: Backlog
09:37:46 CMD: Group 0, Index 2, Command "RULE", Data ""
09:37:46 RSL: stat/tasmota_F8AF5F/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
09:37:47 SRC: Backlog
09:37:47 CMD: Group 0, Index 3, Command "RULE", Data ""
09:37:47 RSL: stat/tasmota_F8AF5F/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
09:36:09 CMD: Status 0
09:36:09 SRC: WebConsole from 192.168.188.150
09:36:09 CMD: Group 0, Index 1, Command "STATUS", Data "0"
09:36:09 RSL: stat/tasmota_F8AF5F/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_F8AF5F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:36:09 RSL: stat/tasmota_F8AF5F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Exception","Uptime":"0T00:06:13","StartupUTC":"2020-06-19T08:29:56","Sleep":50,"CfgHolder":4617,"BootCount":30,"BCResetTime":"2020-06-18T09:18:31","SaveCount":48,"SaveAddress":"F9000"}}
09:36:09 RSL: stat/tasmota_F8AF5F/STATUS2 = {"StatusFWR":{"Version":"8.3.1.6(sensors)","BuildDateTime":"2020-06-18T22:26:53","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"345/699"}}
09:36:09 RSL: stat/tasmota_F8AF5F/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ghiglierotti",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000"]}}
09:36:09 RSL: stat/tasmota_F8AF5F/STATUS4 = {"StatusMEM":{"ProgramSize":681,"Free":320,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FCAA5B6","0405A1A5","B7FFBFCD","01DA9BC4","64377CC7","0008C052"],"Drivers":"1,2,3,4,5,6,8,9,10,12,14,16,17,24,29,31,34,40","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
09:36:09 RSL: stat/tasmota_F8AF5F/STATUS5 = {"StatusNET":{"Hostname":"tasmota_F8AF5F-3935","IPAddress":"192.168.188.167","Gateway":"192.168.188.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.188.5","Mac":"5C:CF:7F:F8:AF:5F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
09:36:09 RSL: stat/tasmota_F8AF5F/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F8AF5F","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
09:36:09 RSL: stat/tasmota_F8AF5F/STATUS7 = {"StatusTIM":{"UTC":"2020-06-19T08:36:09","Local":"2020-06-19T09:36:09","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:56"}}
09:36:09 RSL: stat/tasmota_F8AF5F/STATUS10 = {"StatusSNS":{"Time":"2020-06-19T09:36:09"}}
09:36:09 RSL: stat/tasmota_F8AF5F/STATUS11 = {"StatusSTS":{"Time":"2020-06-19T09:36:09","Uptime":"0T00:06:13","UptimeSec":373,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"ghiglierotti","BSSId":"10:DA:43:08:D6:EC","Channel":11,"RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:03"}}}
09:36:09 RSL: stat/tasmota_F8AF5F/STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"Exception","EPC":["40216f5c","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4029fd00","40291651","4029316c","40291640","4021700f","4029163c","40291640","4020cb9a","40291638","4023fa5d","40295be0","40237b61","40217131","40234162","40260f24","40259c77","40226c05","40101dd3","40101b34","40000000","40258408","40248b1c","40234280","402341c0","4023e7a1","402448fd","4025a41c","40259add","4021d088","402464d8","4021d088"]}}

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

09:30:01 RSL: tele/tasmota_F8AF5F/INFO3 = {"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["40216f5c","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4029fd00","40291651","4029316c","40291640","4021700f","4029163c","40291640","4020cb9a","40291638","4023fa5d","40295be0","40237b61","40217131","40234162","40260f24","40259c77","40226c05","40101dd3","40101b34","40000000","40258408","40248b1c","40234280","402341c0","4023e7a1","402448fd","4025a41c","40259add","4021d088","402464d8","4021d088"]}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Self compile enabling Arduino OTA + SlaveTasmota feature

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Send/Receive commands from slave 328p

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-06-19:
On the SC the standard ATMEGA328P is connected to GPIO1 and 3.

Apparently you connected a second Atmega to GPIO4 and 5 and made sure the power supply can handle this.

You'll need to provide console logging of steps you take until the exception occurs.

ghiglie, 2020-06-19:
Sorry Theo, I got multiple page 500 errors during submit and I noticed too late it was committing for real. 
I was testing with other GPIOs - I'm putting the right ones back and update here. Thanks for your time!

ghiglie, 2020-06-19:
I've just recompiled using dev branch, same with 8.3.1.6 (Client instead of Slave), my status 0:
`11:23:41 CMD: STATUS 0
11:23:41 RSL: stat/tasmota_F8AF5F/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_F8AF5F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:23:41 RSL: stat/tasmota_F8AF5F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Exception","Uptime":"0T00:00:47","StartupUTC":"2020-06-19T10:22:54","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2020-06-19T11:19:38","SaveCount":9,"SaveAddress":"FB000"}}
11:23:41 RSL: stat/tasmota_F8AF5F/STATUS2 = {"StatusFWR":{"Version":"8.3.1.6(tasmota)","BuildDateTime":"2020-06-19T10:03:06","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"345/699"}}
11:23:41 RSL: stat/tasmota_F8AF5F/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ghiglierotti",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000"]}}
11:23:41 RSL: stat/tasmota_F8AF5F/STATUS4 = {"StatusMEM":{"ProgramSize":588,"Free":412,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE793","043683A1","000000CD","010013C0","C001F981","00004024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,31,35,37","Sensors":"1,2,3,4,5,6"}}
11:23:41 RSL: stat/tasmota_F8AF5F/STATUS5 = {"StatusNET":{"Hostname":"tasmota_F8AF5F-3935","IPAddress":"192.168.188.167","Gateway":"192.168.188.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.188.5","Mac":"5C:CF:7F:F8:AF:5F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:23:41 RSL: stat/tasmota_F8AF5F/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F8AF5F","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:23:41 RSL: stat/tasmota_F8AF5F/STATUS7 = {"StatusTIM":{"UTC":"2020-06-19T10:23:41","Local":"2020-06-19T11:23:41","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:56"}}
11:23:41 RSL: stat/tasmota_F8AF5F/STATUS10 = {"StatusSNS":{"Time":"2020-06-19T11:23:41"}}
11:23:41 RSL: stat/tasmota_F8AF5F/STATUS11 = {"StatusSTS":{"Time":"2020-06-19T11:23:41","Uptime":"0T00:00:47","UptimeSec":47,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"ghiglierotti","BSSId":"10:DA:43:08:D6:EC","Channel":11,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:03"}}}
11:23:41 RSL: stat/tasmota_F8AF5F/STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"Exception","EPC":["4020e7ac","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4027edf9","4027fd00","4027b7fa","4027e0c4","4027b7e8","4020e85f","4027b7e4","4027b7e8","402034b2","4027b7e0","40230715","4022548d","4020ee31","40222152","4024be74","402445ec","40217d40","401019eb","4010174c","40000000","4024444c","40237f70","4022226c","402221ac","4022e009","40233251","402462b8","40245955","40212038","40235120","40212038"]}}`

arendst, 2020-06-22:
If no Client Reset pin is defined the ``ClientSend`` command should not have been accepted.

This is fixed in the latest commit.

So in your case you will need to define a Client reset pin to be able to execute ``ClientSend`` commands

ghiglie, 2020-06-22:
Thanks Theo, so it's not a real issue. I didn't define a reset pin because, on Sonoff SC, the 328p runs on 5V and I'm not confident to bridge it to the ESP8266.

closed 2020-06-22 by ghiglie