arendst/Tasmota #8737: Make possible to build ESP32 versions too using GitPod

nagyrobi, 2020-06-19:
**Have you looked for this feature in other issues and in the docs?**  
Yes.

**Is your feature request related to a problem? Please describe.**  
It's great that [we can use GitPod](https://tasmota.github.io/docs/Gitpod/) to compile custom binaries of Tasmota. But as far as I can see it's only possible to create ESP8266 binaries with that tool, and not ESP32.

**Describe the solution you'd like**  
Please enable ESP32 builds on GiitPod.

**Describe alternatives you've considered**  
It's a pain to set up a local build machine each time.

**Additional context**  
N/A

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-06-19:
It works no change needed. Just rename platformio_override_sample.ini to platformio_override.ini 
No difference in using PlatformIO local or with Gitpod

nagyrobi, 2020-06-19:
OK. I'll make some PRs to extend the documentation to cover ESP32 too.

closed 2020-06-19 by nagyrobi