arendst/Tasmota #8743: Support Multiple IR Transmitters

bdr99, 2020-06-21:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
I connected 2 IR transmitter LEDs and configured them in the Module settings. When I used the `IRSend` command, Tasmota only used the first IR transmitter, and ignored the rest.

**Describe the solution you'd like**  
It would be helpful if the `IRSend` command could transmit on all configured IR transmitters, instead of just the first one. Or alternatively, the `IRSend` command could include an argument to select which transmitter to use, similar to how the `Power1`, `Power2` commands work.

**Describe alternatives you've considered**  
Currently, I'm using a separate Tasmota device for each IR transmitter, but it would be easier if I could use just one.

**Additional context**  
N/A

arendst, 2020-06-21:
Why not connect them to a single gpio as all other ir busters do? A simple transistor can fix this without any software change. No other fix expected anytime soon

device111, 2020-06-21:
so, what is the background to use 2 different IR-Transmitter?
you can adjust the protokoll and all paras with IRsend and use one transmitter for all.

ascillato2, 2020-06-21:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-06-21 by ascillato2

bdr99, 2020-06-22:
> 
> 
> Why not connect them to a single gpio as all other ir busters do? A simple transistor can fix this without any software change. No other fix expected anytime soon

Thanks, this is a great idea and I hadn't thought of it. I'll just connect them both to the same GPIO.

> so, what is the background to use 2 different IR-Transmitter?
> you can adjust the protokoll and all paras with IRsend and use one transmitter for all.

The reason I need 2 transmitters is because I'm trying to control 2 different devices in 2 different locations. If I only use 1 transmitter, I can't get it in range of both devices.