arendst/Tasmota #8745: iBeacon JSON attributes (related to #8693)

Elix-g, 2020-06-21:
**Have you looked for this feature in other issues and in the docs?**  

Yes I did.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

Thanks a lot for the swift implementation of #8693! Everything working fine except two details:

- The MAC address of detected iBeacons is now being omitted and not issued as part of the MQTT message.

- The decision, if a zero RSSI for a disappeared iBeacon is being sent by MQTT is still based on the iBeacon's MAC address. Having an iBeacon that uses MAC randomization, leads to frequent zero RSSI MQTT messages in the moment the beacons switches to another MAC address.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

- Add iBeacon's MAC address as a name/value pair to MQTT JSON string

- Modify condition which decides if zero RSSI is being sent for an iBeacon from disppeared MAC address to disappeared UUID.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2020-06-21:
could you please try this.
if it works for you ill do a pr

[xsns_52_ibeacon.ino.zip](https://github.com/arendst/Tasmota/files/4809498/xsns_52_ibeacon.ino.zip)

Elix-g, 2020-06-21:
Thank you for the amazing response time!

Works like a charm, MAC address is part of the JSON message and no zero RSSI message issued after MAC address change:

[image]

One more detail: Is it easy to implement a defined JSON path? By using IBEACON__uid_ as a JSON name, it's difficult to parse the JSON message. Adding the UID as a name/value pair to a unique parent name improves the possibility to parse a message and would unify the message with those from other sensors. Assuming "IBEACON" as the sensor name:

{"Time":"2020-06-21T14:40:59","IBEACON":{"UID":"20D01D1B6EF64FADAE3DE340F7E98DBE","MAJOR":"000D","MINOR":"0007","MAC":"5348A6E47435","RSSI":-94}}

gemu2015, 2020-06-21:
no problem, next try

[xsns_52_ibeacon.ino.zip](https://github.com/arendst/Tasmota/files/4809589/xsns_52_ibeacon.ino.zip)

Elix-g, 2020-06-21:
Time for pr :-)

[image]

Perfect now, many thanks again!

ascillato2, 2020-06-21:
Closing as the PR has been submitted.

Thanks :+1:

PR #8749 (2020-06-21): ibeacon json path
## Description:

change ibeacon json path
switch on UID instead of MAC

**Related issue (if applicable):** fixes #8745

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core ESP8266 V.2.7.1
  - [x] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The code change must pa [... cut]
closed 2020-06-21 by ascillato2