arendst/Tasmota #8747: Blitzwolf SHP5 crashes on turning off POWER1

Cybolord, 2020-06-21:
### PROBLEM DESCRIPTION
...researched a lot but found no solution so far:
When a 'switch OFF rule' is fired to switch POWER1 OFF or even manualy via webinterface the SHP5 immediately 'crash' and reboots with following reason...

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Blitzwolf SHP5
- [x] Tasmota binary firmware version number used: 8.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
12:45:23 CMD: Backlog Template; Module; GPIO 255
12:45:23 MQT: stat/poolfilter/RESULT = {"NAME":"BlitzWolf SHP5","GPIO":[57,145,56,146,0,22,0,0,0,0,21,0,17],"FLAG":0,"BASE":18}
12:45:24 MQT: stat/poolfilter/RESULT = {"Module":{"0":"BlitzWolf SHP5"}}
12:45:24 MQT: stat/poolfilter/RESULT = {"GPIO0":{"57":"Led2i"},"GPIO1":{"145":"CSE7766 Tx"},"GPIO2":{"56":"Led1i"},"GPIO3":{"146":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"22":"Relay2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"21":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"17":"Button1"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
12:46:03 CMD: Backlog Rule1; Rule2; Rule3
12:46:03 MQT: stat/poolfilter/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
12:46:03 MQT: stat/poolfilter/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
12:46:04 MQT: stat/poolfilter/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
12:46:43 CMD: Status 0
12:46:43 MQT: stat/poolfilter/STATUS = {"Status":{"Module":0,"DeviceName":"Pool Filter","FriendlyName":["Pool Filter","Sonoff2"],"Topic":"poolfilter","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:46:43 MQT: stat/poolfilter/STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:22:12","StartupUTC":"2020-06-21T11:24:31","Sleep":50,"CfgHolder":4617,"BootCount":889,"BCResetTime":"2020-06-11T10:14:01","SaveCount":4019,"SaveAddress":"F8000"}}
12:46:43 MQT: stat/poolfilter/STATUS2 = {"StatusFWR":{"Version":"8.3.1(sensors)","BuildDateTime":"2020-05-18T15:41:26","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"403/699"}}
12:46:43 MQT: stat/poolfilter/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SpeedWireNetworx",""],"TelePeriod":300,"Resolution":"559380C0","SetOption":["00008049","2805C8000100060100005AFF000000000000","00000000","00000000"]}}
12:46:43 MQT: stat/poolfilter/STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAA787","0415A3A4","B7FFBFCD","01DA9BC4","64367CC7","00000072"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67"}}
12:46:43 MQT: stat/poolfilter/STATUS5 = {"StatusNET":{"Hostname":"poolfilter-0746","IPAddress":"192.168.100.106","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.2","Mac":"DC:4F:22:9C:C2:EA","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
12:46:43 MQT: stat/poolfilter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.200","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9CC2EA","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:46:43 MQT: stat/poolfilter/STATUS7 = {"StatusTIM":{"UTC":"2020-06-21T11:46:43","Local":"2020-06-21T12:46:43","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:57"}}
12:46:43 MQT: stat/poolfilter/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:46:43 MQT: stat/poolfilter/STATUS10 = {"StatusSNS":{"Time":"2020-06-21T12:46:43","ENERGY":{"TotalStartTime":"2019-11-03T13:59:19","Total":5.249,"Yesterday":0.106,"Today":0.238,"Power":0.00,"ApparentPower":0.00,"ReactivePower":0.00,"Factor":0.00,"Voltage":232.7,"Current":0.000}}}
12:46:43 MQT: stat/poolfilter/STATUS11 = {"StatusSTS":{"Time":"2020-06-21T12:46:43","Uptime":"0T00:22:12","UptimeSec":1332,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"SpeedWireNetworx","BSSId":"2C:91:AB:9A:99:0B","Channel":6,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
00:00:00 CFG: Loaded from flash at FB, Count 4016
00:00:00 Project tasmota Pool Filter Version 8.3.1(sensors)-2_7_1
00:00:00 SER: Set to 8E1 4800 bit/s
00:00:00 SNS: Hardware Serial
00:00:00 WIF: Connecting to AP1 SpeedWireNetworx Channel 6 BSSId 2C:91:AB:9A:99:0B in mode 11N as poolfilter-0746...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on poolfilter-0746 with IP address 192.168.100.106
12:24:35 MQT: Attempting connection...
12:24:35 MQT: Connected
12:24:35 MQT: tele/poolfilter/LWT = Online (retained)
12:24:35 MQT: cmnd/poolfilter/POWER =
12:24:35 MQT: tele/poolfilter/INFO1 = {"Module":"BlitzWolf SHP5","Version":"8.3.1(sensors)","FallbackTopic":"cmnd/DVES_9CC2EA_fb/","GroupTopic":"cmnd/sonoffs/"}
12:24:35 MQT: tele/poolfilter/INFO2 = {"WebServerMode":"Admin","Hostname":"poolfilter-0746","IPAddress":"192.168.100.106"}
12:24:35 MQT: tele/poolfilter/INFO3 = {"RestartReason":"Hardware Watchdog"}
12:24:35 MQT: stat/poolfilter/RESULT = {"POWER1":"ON"}
12:24:35 MQT: stat/poolfilter/POWER1 = ON
12:24:35 MQT: stat/poolfilter/RESULT = {"POWER2":"ON"}
12:24:35 MQT: stat/poolfilter/POWER2 = ON
...

```

### TO REPRODUCE
switch POWER1 OFF (via MQTT Rule or even via webinterface)

### EXPECTED BEHAVIOUR
POWER1 should stay OFF but it switch ON again

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
played with additional settings but no success:

    SerialLog 0
    SetOption39 1
    ButtonRetain 1

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-06-21:
Hi

Hardware watchdog means that the cpu has hung. When it is a problem in Tasmota is is shown as software watchdog. 

So, let's try reflashing by serial. 

Please, erase all flash using tasmotizer or esptool.py and flash again with the precompiled bins from the development branch (thehackbox.org/tasmota/tasmota.bin)

Cybolord, 2020-06-21:
thank you for your feedback.

this is would be more affort than expected indeed :(

ascillato, 2020-06-21:
Sometimes, a flashing procedure can fail if there was any wire a bit loose or if the flash wasn't erased before flashing.

arendst, 2020-06-22:
Do you have any USB connected to Power2? If so what is the expected load?

Cybolord, 2020-06-22:
Pardon me? what kind of 'load' please? maybe the current power supply load?

Voltage: 237 V
Current 0.336 A
Power 23.36 W

...is it this information you asking for?

arendst, 2020-06-22:
Well in my case there is just a simple router connected as load to the SHP5. Switching works as designed.

If you had a massice inductive load (device, motor etc.) it might interfere with the SHP5 as it is powered from the same voltage as the load. Inductive loads induce voltage peaks in the net which may result in what you experience.

As your load seems to be only 23.36W I don't think that's the case here ;-)

Do you also have something connected to the USB connectors of the SHP5? If so that load may be to much to handle for the SHP5 too.

Cybolord, 2020-06-22:
regarding: _Inductive loads induce voltage peaks_
and yes indeed: it IS a 24Volts Motor connected to the SHP5 at POWER1
and there is also a WEMOS PRO mini board with pH-Sensor (5Volts) connected to the SHP5 USB at POWER2

so you think I could run in trouble by using such a constellation?

arendst, 2020-06-22:
You might want to check what happens if you disconnect the USB device(s)

Cybolord, 2020-06-22:
I will give it a try and come back as soon as possible- thank you.

Cybolord, 2020-06-23:
...unfortunately after disconnecting the USB device the same behaviour can be observed; even if you use the TOGGLE1 button in the Tasmota 8.3.1 webinterface manually the SHP5 crashes like before

arendst, 2020-06-23:
As there are no sensors connected to the SHP5 I you might want to change your binary version from ``sensors`` to just tasmota.bin or tasmota-light.bin

These binaries maybe using less power then the current tasmota-sensors.bin

Cybolord, 2020-06-23:
originaly it was sonoff.bin form v6.x release and for troubleshooting I tried to use tasmota-sensors-precompiled-binary v8.3.1 ...

Cybolord, 2020-06-23:
thus should I erase and flash from scratch again via serial UART?

arendst, 2020-06-23:
It looks like it yes.

Cybolord, 2020-06-23:
do you have some other experience especially with switching SHP5?
 in other words: normaly it should work and thus it is "fully" supported in basic tasmota.bin?

Thank you again.

arendst, 2020-06-23:
Mine works as expected. No reboots.

Cybolord, 2020-06-23:
nice, so I will ersae & flash it again and come back soon. 

thanks for your support!

Cybolord, 2020-06-23:
did two esptool steps with tasmota.bin:
>esptool.py --port /dev/cu.SLAB_USBtoUART --baud 115200 erase_flash
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.6s
Hard resetting via RTS pin...
>esptool.py --port /dev/cu.SLAB_USBtoUART --baud 115200 write_flash -fs 1MB -fm dout 0x00000 tasmota.bin
Compressed 580480 bytes to 399923...
Wrote 580480 bytes (399923 compressed) at 0x00000000 in 35.2 seconds (effective 131.8 kbit/s)...
Hash of data verified.

Result: same behaviour as described before:
00:00:00 CFG: Loaded from flash at F7, Count 29
00:00:00 Project tasmota Pool Filter Version 8.1.0(tasmota)-2_6_1
00:00:00 WIF: Connecting to AP1 SpeedWireNetworx in mode 11N as blitzwolf3-0746...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on blitzwolf3-0746 with IP address 192.168.100.106
17:05:13 MQT: Attempting connection...
17:05:13 MQT: Connected
17:05:13 MQT: tele/blitzwolf3/LWT = Online (retained)
17:05:13 MQT: cmnd/blitzwolf3/POWER = 
17:05:13 MQT: tele/blitzwolf3/INFO1 = {"Module":"SHP5","Version":"8.1.0(tasmota)","FallbackTopic":"cmnd/DVES_9CC2EA_fb/","GroupTopic":"cmnd/tasmotas/"}
17:05:13 MQT: tele/blitzwolf3/INFO2 = {"WebServerMode":"Admin","Hostname":"blitzwolf3-0746","IPAddress":"192.168.100.106"}
**17:05:13 MQT: tele/blitzwolf3/INFO3 = {"RestartReason":"Hardware Watchdog"}**

Cybolord, 2020-06-23:
now the interesting part --> with absolutly **no load** (means empty socket) it works flawlessly !!

17:14:10 MQT: stat/blitzwolf3/POWER1 = OFF
17:14:22 CMD: VoltageSet 231
17:14:22 MQT: stat/blitzwolf3/RESULT = {"VoltageSetCal":3446}
17:14:28 MQT: stat/blitzwolf3/RESULT = {"POWER1":"ON"}
17:14:28 MQT: stat/blitzwolf3/POWER1 = ON
17:14:35 MQT: stat/blitzwolf3/RESULT = {"POWER1":"OFF"}
17:14:35 MQT: stat/blitzwolf3/POWER1 = OFF
17:14:40 MQT: stat/blitzwolf3/RESULT = {"POWER1":"ON"}
17:14:40 MQT: stat/blitzwolf3/POWER1 = ON
17:14:44 MQT: stat/blitzwolf3/RESULT = {"POWER1":"OFF"}
17:14:44 MQT: stat/blitzwolf3/POWER1 = OFF
17:14:47 MQT: stat/blitzwolf3/RESULT = {"POWER1":"ON"}
17:14:47 MQT: stat/blitzwolf3/POWER1 = ON
17:14:59 MQT: stat/blitzwolf3/RESULT = {"POWER1":"ON"}
17:14:59 MQT: stat/blitzwolf3/POWER1 = ON
17:15:15 MQT: stat/blitzwolf3/RESULT = {"POWER1":"OFF"}
17:15:15 MQT: stat/blitzwolf3/POWER1 = OFF
17:15:28 MQT: stat/blitzwolf3/RESULT = {"POWER1":"ON"}
17:15:28 MQT: stat/blitzwolf3/POWER1 = ON

Cybolord, 2020-06-23:
...no more crash/reboot when nothing is pluged in...

arendst, 2020-06-23:
Play with some different loads and see what happens.

If I find time tomorrow I'll have a look too

arendst, 2020-06-24:
Pls try the latest dev version 8.3.1.6

It might contain a solution for your issue.

Cybolord, 2020-06-24:
That's awesome, thank you Theo!
I'll try this out tonight and give you some feedback...

Cybolord, 2020-06-24:
...extend lunchtime to flash with v8.3.1.6 - but negativ:

00:00:00 CFG: Loaded from flash at F9, Count 73
00:00:00 Project tasmota Pool Filter Version 8.3.1.6(b263309-tasmota)-2_7_1
00:00:00 SER: Set to 8E1 4800 bit/s
00:00:00 SNS: Hardware Serial
00:00:00 WIF: Connecting to AP1 SpeedWireNetworx Channel 1 BSSId 34:31:C4:52:35:CB in mode 11N as blitzwolf3-0746...
00:00:03 WIF: Connected
00:00:03 HTP: Web server active on blitzwolf3-0746 with IP address 192.168.100.106
12:44:32 MQT: Attempting connection...
12:44:32 MQT: Connected
12:44:32 MQT: tele/blitzwolf3/LWT = Online (retained)
12:44:32 MQT: cmnd/blitzwolf3/POWER = 
12:44:32 MQT: tele/blitzwolf3/INFO1 = {"Module":"SHP5","Version":"8.3.1.6(b263309-tasmota)","FallbackTopic":"cmnd/DVES_9CC2EA_fb/","GroupTopic":"cmnd/tasmotas/"}
12:44:32 MQT: tele/blitzwolf3/INFO2 = {"WebServerMode":"Admin","Hostname":"blitzwolf3-0746","IPAddress":"192.168.100.106"}
**12:44:32 MQT: tele/blitzwolf3/INFO3 = {"RestartReason":"Hardware Watchdog"}**
12:44:32 MQT: stat/blitzwolf3/RESULT = {"POWER1":"ON"}
12:44:32 MQT: stat/blitzwolf3/POWER1 = ON
12:44:32 MQT: stat/blitzwolf3/RESULT = {"POWER2":"ON"}
12:44:32 MQT: stat/blitzwolf3/POWER2 = ON

arendst, 2020-06-24:
Done some testing with the same version as you do without any error:
```
00:00:00 CFG: Loaded from flash at F9, Count 1647
00:00:00 QPC: Flag 7E
00:00:00 SRC: Restart
00:00:00 Project tasmota Shp5 DLink router Version 8.3.1.6(b263309-tasmota)-2_7_1
00:00:00 SER: Set to 8E1 4800 bit/s
00:00:00 SNS: Hardware Serial
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP2, SSId indebuurt_IoT, Channel 1, BSSId 74:83:C2:2A:F1:AC, RSSI -80, Encryption 1
00:00:04 WIF: Network 1, AP2, SSId indebuurt_IoT, Channel 11, BSSId 18:E8:29:CA:17:C1, RSSI -36, Encryption 1
00:00:04 WIF: Connecting to AP2 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11N as shp5...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on shp5 with IP address 192.168.2.230
16:58:52 NTP: UTC 2020-06-24T14:58:51, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
16:58:52 QPC: Reset
16:58:52 MQT: Attempting connection...
16:58:52 MQT: Connected
16:58:52 MQT: tele/shp5/LWT = Online (retained)
16:58:52 MQT: cmnd/shp5/POWER = 
16:58:52 MQT: Subscribe to cmnd/shp5/#
16:58:52 MQT: Subscribe to cmnd/tasmotas/#
16:58:52 MQT: Subscribe to cmnd/DVES_9CB763_fb/#
16:58:52 MQT: tele/shp5/INFO1 = {"Module":"Blitzwolf SHP5","Version":"8.3.1.6(b263309-tasmota)","FallbackTopic":"cmnd/DVES_9CB763_fb/","GroupTopic":"cmnd/tasmotas/"}
16:58:52 MQT: tele/shp5/INFO2 = {"WebServerMode":"Admin","Hostname":"shp5","IPAddress":"192.168.2.230"}
16:58:52 MQT: tele/shp5/INFO3 = {"RestartReason":"External System"}
16:58:52 MQT: stat/shp5/RESULT = {"POWER1":"OFF"}
16:58:52 MQT: stat/shp5/POWER1 = OFF
16:58:52 MQT: stat/shp5/RESULT = {"POWER2":"OFF"}
16:58:52 MQT: stat/shp5/POWER2 = OFF
16:58:54 APP: Boot Count 141
16:58:55 CFG: Saved to flash at F8, Count 1648, Bytes 4096
16:58:57 MQT: tele/shp5/STATE = {"Time":"2020-06-24T16:58:57","Uptime":"0T00:00:13","UptimeSec":13,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:07"}}
16:58:57 MQT: tele/shp5/SENSOR = {"Time":"2020-06-24T16:58:57","ENERGY":{"TotalStartTime":"2019-06-03T15:18:50","Total":25.504,"Yesterday":0.054,"Today":0.036,"Period":0.00,"Power":0.00,"ApparentPower":0.00,"ReactivePower":0.00,"Factor":0.00,"Voltage":235.6,"Current":0.000}}
16:59:55 APP: Button1 multi-press 1
16:59:55 SRC: Button
16:59:55 MQT: stat/shp5/RESULT = {"POWER1":"ON"}
16:59:55 MQT: stat/shp5/POWER1 = ON
16:59:56 CFG: Saved to flash at F7, Count 1649, Bytes 4096
16:59:57 APP: Button1 multi-press 1
16:59:57 SRC: Button
16:59:57 MQT: stat/shp5/RESULT = {"POWER1":"OFF"}
16:59:57 MQT: stat/shp5/POWER1 = OFF
16:59:58 CFG: Saved to flash at F6, Count 1650, Bytes 4096
17:00:05 HTP: Console
```
You might want to enable more logging with ``weblog 4`` and then do a power off to see if you get something like this:
```
17:01:52 DMP: 55 5A 02 E6 F8 00 06 04 00 3E B2 00 F4 7E 4F E9 08 02 90 D1 71 00 E1 41
17:01:52 DMP: 55 5A 02 E6 F8 00 06 04 00 3E B2 00 F4 7E 4F E9 08 02 90 D1 71 00 E1 41
17:01:52 DMP: 55 5A 02 E6 F8 00 06 04 00 3E B2 00 F3 EF 4F E9 08 02 90 D1 71 00 E2 B2
17:01:52 DMP: 55 5A 02 E6 F8 00 06 04 00 3E B2 00 F3 EF 4F E9 08 02 90 D1 71 00 E3 B3
17:01:52 DMP: 55 5A 02 E6 F8 00 06 04 00 3E B2 00 F3 EF 4F E9 08 02 90 D1 71 00 E3 B3
17:01:53 CMD: power 0
17:01:53 SRC: WebConsole from 192.168.2.1
17:01:53 CMD: Group 0, Index 1, Command "POWER", Data "0"
17:01:53 MQT: stat/shp5/RESULT = {"POWER1":"OFF"}
17:01:53 MQT: stat/shp5/POWER1 = OFF
17:01:53 DMP: 55 5A 02 E6 F8 00 06 04 00 3E B2 00 F3 EF 4F E9 08 02 90 D1 71 00 E4 B4
17:01:53 DMP: 55 5A 02 E6 F8 00 06 04 00 3E B2 00 F3 EF 4F E9 08 02 8B 91 71 00 E4 6F
17:01:53 CFG: Saved to flash at F9, Count 1655, Bytes 4096
17:01:53 DMP: 55 5A 02 E6 F8 00 06 04 00 3E B2 00 F3 EF 4F E9 08 02 8B 91 71 00 E5 70
17:01:53 DMP: 55 5A 02 E6 F8 00 06 04 00 3E B2 00 F3 EF 4F E9 08 02 8B 91 71 00 E5 70
17:01:53 DMP: 55 5A 02 E6 F8 00 06 04 00 3E B2 00 F3 EF 4F E9 08 02 8B 91 71 00 E5 70
```

Cybolord, 2020-06-24:
0:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:03 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E 90 71 EE C3 2C
00:00:03 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E 90 71 EE C3 2C
00:00:03 HTP: Web server active on blitzwolf3-0746 with IP address 192.168.100.106
00:00:03 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E 90 71 EE C4 2D
00:00:03 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E 90 71 EE C4 2D
00:00:03 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E 90 71 EE C5 2E
17:18:22 NTP: UTC 2020-06-24T16:18:21, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
17:18:22 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E 90 71 EE C5 2E
17:18:22 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E 90 71 EE C6 2F
17:18:22 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E 90 71 EE C6 2F
17:18:22 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E 90 71 EE C7 30
17:18:22 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E 90 71 EE C7 30
17:18:22 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E 90 71 EE C8 31
17:18:22 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E D1 71 EE C8 72
17:18:22 MQT: Attempting connection...
17:18:22 MQT: Connected
17:18:22 MQT: tele/blitzwolf3/LWT = Online (retained)
17:18:22 MQT: cmnd/blitzwolf3/POWER = 
17:18:22 MQT: Subscribe to cmnd/blitzwolf3/#
17:18:22 MQT: Subscribe to cmnd/tasmotas/#
17:18:22 MQT: Subscribe to cmnd/DVES_9CC2EA_fb/#
17:18:22 MQT: tele/blitzwolf3/INFO1 = {"Module":"SHP5","Version":"8.3.1.6(b263309-tasmota)","FallbackTopic":"cmnd/DVES_9CC2EA_fb/","GroupTopic":"cmnd/tasmotas/"}
17:18:22 MQT: tele/blitzwolf3/INFO2 = {"WebServerMode":"Admin","Hostname":"blitzwolf3-0746","IPAddress":"192.168.100.106"}
17:18:22 MQT: tele/blitzwolf3/INFO3 = {"RestartReason":"Hardware Watchdog"}
17:18:22 MQT: stat/blitzwolf3/RESULT = {"POWER1":"ON"}
17:18:22 MQT: stat/blitzwolf3/POWER1 = ON
17:18:22 MQT: stat/blitzwolf3/RESULT = {"POWER2":"ON"}
17:18:22 MQT: stat/blitzwolf3/POWER2 = ON
17:18:22 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E D1 71 EE C9 73
17:18:22 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E D1 71 EE C9 73
17:18:23 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E D1 71 EE CA 74
17:18:23 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 1A 4A 34 18 03 5E D1 71 EE CA 74
17:18:23 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E D1 71 EE CB CA
17:18:23 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E D1 71 EE CB CA
17:18:23 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E D1 71 EE CC CB
17:18:23 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E D1 71 EE CC CB
17:18:23 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E D1 71 EE CC CB
17:18:23 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E D1 71 EE CD CC
17:18:23 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E D1 71 EE CD CC
17:18:23 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E D1 71 EE CE CD
17:18:24 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E EF 71 EE CE EB
17:18:24 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E EF 71 EE CF EC
17:18:24 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E EF 71 EE CF EC
17:18:24 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E EF 71 EE D0 ED
17:18:24 QPC: Reset
17:18:24 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E EF 71 EE D0 ED
17:18:24 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E EF 71 EE D1 EE
17:18:24 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E EF 71 EE D1 EE
17:18:24 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E EF 71 EE D2 EF
17:18:24 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E EF 71 EE D2 EF
17:18:25 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E EF 71 EE D3 F0
17:18:25 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E EF 71 EE D3 F0
17:18:25 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5E EF 71 EE D4 F1
17:18:25 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5F 0E 71 EE D4 11
17:18:25 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5F 0E 71 EE D5 12
17:18:25 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5F 0E 71 EE D5 12
17:18:25 DMP: 55 5A 02 C6 28 00 05 F2 00 3B 8D 00 BA 6F 4A 34 18 03 5F 0E 71 EE D6 13

Cybolord, 2020-06-24:
set "weblog 4" is this helpful ?

arendst, 2020-06-24:
It is if you use it around your power off moment don't you think? The previous is useless indeed.

Cybolord, 2020-06-24:
i did exactly when sending the POWER1 OFF command via MQTT but the console is gone in that special moment, SHP5 device is not responding anymore... so, how to catch the debug?

arendst, 2020-06-24:
I guess we don't.

I'm afraid it's time to buy a new one....

arendst, 2020-06-24:
A last resort is removing power measurement and see if you can power toggle. To do this edit the template and set both gpio1 and 3 to 0

Cybolord, 2020-06-24:
ohhhh, that means you really think it is an hardware issue, indeed?
did you test the same device from Gosund? I would like to get those, maybe they also work well...

Thank you very much for your effort Theo!

arendst, 2020-06-24:
You missed one suggestion above

Cybolord, 2020-06-24:
you were faster ;) i try that first

Cybolord, 2020-06-24:
unfortunately it's the same :(

arendst, 2020-06-24:
Then it must be a hardware problem.

Cybolord, 2020-06-25:
...flashed a brand new Gosund SP112 (newer HW Rev. 2.8) today and guess what: same results!

00:00:00 CFG: Loaded from flash at F8, Count 31
00:00:00 Project tasmota Pool Filter Version 8.3.1.6(ad46af6-tasmota)-2_7_1
00:00:00 SER: Set to 8E1 4800 bit/s
00:00:00 SNS: Hardware Serial
00:00:00 WIF: Connecting to AP1 SpeedWireNetworx Channel 1 BSSId 2C:91:AB:9A:99:0B in mode 11N as blitzwolf3-7931...
00:00:01 WIF: Connected
00:00:01 HTP: Web server active on blitzwolf3-7931 with IP address 192.168.100.106
00:00:01 RSL: INFO1 = {"Module":"SHP5","Version":"8.3.1.6(ad46af6-tasmota)","FallbackTopic":"cmnd/DVES_BB1EFB_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:01 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"blitzwolf3-7931","IPAddress":"192.168.100.106"}
00:00:01 RSL: INFO3 = {"RestartReason":"Hardware Watchdog"}
00:00:01 RSL: RESULT = {"POWER1":"ON"}
00:00:01 RSL: POWER1 = ON
00:00:01 RSL: RESULT = {"POWER2":"ON"}
00:00:01 RSL: POWER2 = ON
16:37:10 RSL: STATE = {"Time":"2020-06-25T16:37:10","Uptime":"0T00:00:08","UptimeSec":8,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"SpeedWireNetworx","BSSId":"2C:91:AB:9A:99:0B","Channel":1,"RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:03"}}
16:37:10 RSL: SENSOR = {"Time":"2020-06-25T16:37:10","ENERGY":{"TotalStartTime":"2020-06-25T16:15:22","Total":0.003,"Yesterday":0.000,"Today":0.003,"Period":0,"Power":23,"ApparentPower":75,"ReactivePower":71,"Factor":0.31,"Voltage":229,"Current":0.325}}

so is it worth it to go ahead from your point of view?

arendst, 2020-06-25:
I'm out of ideas.

Cybolord, 2020-06-25:
I understand - thank you very much again for your effort Theo! 

...meanwhile I tried many different power consumers:
e.g.
- a 29W non-eletronic bulb
- a 100W jighsaw
- a MacBookPro power supply 16.5V 6.25A out

but indeed I am trying to switch an INTEX Pool Pump 604G behind a Skyish power supply 12V 5A out...

maybe there is someone else with a BLITZWOLF Switch AND an INTEX Pool pump and would be so nice to test this out?!

...for some reason we may have an inductive issue

kugelkopf123, 2020-06-26:
Perhaps a eq3 esb1 inrush current Protection could help you with this. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Cybolord, 2020-06-26:
> Perhaps a eq3 esb1 inrush current Protection could help you with this.
> 
> Sent with [GitHawk](http://githawk.com)

would you be so kind to specifiy "eq3 esb1 inrush" a little bit more in detail, please?

Cybolord, 2020-06-26:
I will close this case again because it cannot be a tasmota issue... possibly caused by so called "china crap" ;)

closed 2020-06-26 by Cybolord