arendst/Tasmota #8748: mi desk lamp dont update brightness

dabos1, 2020-06-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (mi desk lamp): _____
- [ ] Tasmota binary firmware version number used: _8.3.1____
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [ ] IDE / Compiler used: __PIO___
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

For a few days ago I flashed my new mi desk lamp with tasmota. and it works most time. But when I adjust the brightness with the rotary the brightness wont update to tasmota.

is there a way to fix this issue? 

thank you for help

dabos1, 2020-06-21:
> ### PROBLEM DESCRIPTION
> _A clear and concise description of what the problem is._
> 
> ### REQUESTED INFORMATION
> _Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_
> 
> * [x]  Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
> * [x]  Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
> * [x]  Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
> * [x]  Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
> * [x]  Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
> * [x]  Device used (mi desk lamp): _____
> * [x]  Tasmota binary firmware version number used: _8.3.1____
>   
>   * [ ]  Pre-compiled
>   * [x ] Self-compiled
>     
>     * [x]  IDE / Compiler used: **PIO**_
> * [x]  Flashing tools used: _**tasmotizer**
> * [x]  Provide the output of command: `Backlog Template; Module; GPIO 255`:
> 
> ```
>   Configuration output here:
> 15:08:47 CMD: Backlog Template; Module; GPIO 255
> 15:08:48 MQT: stat/tasmota_79E588/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
> 15:08:48 MQT: stat/tasmota_79E588/RESULT = {"Module":{"66":"Mi Desk Lamp"}}
> 15:08:48 MQT: stat/tasmota_79E588/STATUS = {"Status":{"Module":66,"DeviceName":" Schreibtischlicht ","FriendlyName":[" Schreibtischlicht "],"Topic":"tasmota_79E588","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
> ```
> 
> * [ ]  If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
> 
> ```
>   Rules output here:
> ```
> 
> * [ ]  Provide the output of this command: `Status 0`:
> 
> ```
>   STATUS 0 output here:
> ```
> 
> * [ ]  Provide the output of the Console log output when you experience your issue; if applicable:
>   _(Please use_ `weblog 4` _for more debug information)_
> 
> ```
>   Console output here:
> ```
> 
> ### TO REPRODUCE
> _Steps to reproduce the behavior:_
> 
> ### EXPECTED BEHAVIOUR
> _A clear and concise description of what you expected to happen._
> 
> ### SCREENSHOTS
> _If applicable, add screenshots to help explain your problem._
> 
> ### ADDITIONAL CONTEXT
> _Add any other context about the problem here._
> 
> **(Please, remember to close the issue when the problem has been addressed)**
> 
> For a few days ago I flashed my new mi desk lamp with tasmota. and it works most time. But when I adjust the brightness with the rotary the brightness wont update to tasmota.
> 
> is there a way to fix this issue?
> 
> thank you for help

ascillato, 2020-06-21:
Template incomplete. Please complete it in order to have more debug information in order to help you. Thanks

dabos1, 2020-06-21:
> Template incomplete. Please complete it in order to have more debug information in order to help you. Thanks

I use the module (66 mi desk lamp) I haven’t changed the Template. The Template isn’t active

ascillato2, 2020-06-21:
It is not supported. There is an open feature request for that at https://github.com/arendst/Tasmota/issues/8263

ascillato2, 2020-06-21:
Please, share the output of `status 0` command

ascillato2, 2020-06-21:
Have you followed the instructions of https://templates.blakadder.com/mi_desk_lamp.html ?

You need to use the template

dabos1, 2020-06-21:
> Have you followed the instructions of https://templates.blakadder.com/mi_desk_lamp.html ?
> 
> You need to use the template for this device.

I followed this instruction https://tasmota.github.io/docs/devices/Xiaomi-Mi-Desk-Lamp/

dabos1, 2020-06-21:
> Please, share the output of `status 0` command

I posted it in the requsted information

ascillato, 2020-06-21:
Please, also share (as requested in the issue) the output of the console when you use the rotary. Please, before that, set weblog to 4 in order to display more debug information. Thanks

dabos1, 2020-06-21:
> Please, also share (as requested in the issue) the output of the console when you use the rotary. Please, before that, set weblog to 4 in order to display more debug information. Thanks

This is the output in the console 

18:47:55 MQT: stat/tasmota_79E588/STATUS = {"Status":{"Module":66,"DeviceName":" Schreibtischlicht ","FriendlyName":[" Schreibtischlicht "],"Topic":"tasmota_79E588","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:47:55 MQT: stat/tasmota_79E588/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"2T02:24:57","StartupUTC":"2020-06-19T15:22:58","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2020-06-18T16:22:45","SaveCount":862,"SaveAddress":"F6000"}}
18:47:55 MQT: stat/tasmota_79E588/STATUS2 = {"StatusFWR":{"Version":"8.3.1.3(tasmota)","BuildDateTime":"2020-06-15T17:12:04","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"406/699"}}
18:47:55 MQT: stat/tasmota_79E588/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00000000"]}}
18:47:55 MQT: stat/tasmota_79E588/STATUS4 = {"StatusMEM":{"ProgramSize":590,"Free":412,"Heap":23,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","043683A1","000000CD","010013C0","C000F981","00004024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
18:47:55 MQT: stat/tasmota_79E588/STATUS5 = {"StatusNET":{"Hostname":"tasmota_79E588-1416","IPAddress":"192.168.0.33","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.2","Mac":"5C:CF:7F:79:E5:88","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:47:55 MQT: stat/tasmota_79E588/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.3","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_79E588","MqttUser":"daboss","MqttCount":5,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
18:47:55 MQT: stat/tasmota_79E588/STATUS7 = {"StatusTIM":{"UTC":"2020-06-21T17:47:55","Local":"2020-06-21T18:47:55","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:57"}}
18:47:55 MQT: stat/tasmota_79E588/STATUS10 = {"StatusSNS":{"Time":"2020-06-21T18:47:55"}}
18:47:55 MQT: stat/tasmota_79E588/STATUS11 = {"StatusSTS":{"Time":"2020-06-21T18:47:55","Uptime":"2T02:24:57","UptimeSec":181497,"Heap":21,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":5,"POWER":"ON","Dimmer":54,"Color":"102,36","White":54,"CT":244,"Channel":[40,14],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"WLAN","BSSId":"FC:94:E3:0B:68:F3","Channel":6,"RSSI":82,"Signal":-59,"LinkCount":3,"Downtime":"0T00:03:07"}}}
18:47:55 CFG: Saved to flash at F5, Count 863, Bytes 4096
18:47:56 CFG: Saved to flash at F4, Count 864, Bytes 4096
18:47:57 CFG: Saved to flash at FB, Count 865, Bytes 4096
18:47:58 CFG: Saved to flash at FA, Count 866, Bytes 4096
18:47:59 CFG: Saved to flash at F9, Count 867, Bytes 4096

arendst, 2020-06-28:
Try latest possible fix and report back. Pls enable weblog 4 too.

dabos1, 2020-06-29:
> weblog 4

> Try latest possible fix and report back. Pls enable weblog 4 too.

I download the latest development version 8.3.1.6. And I Won’t update the brightness the bug is still there. I published the output of the console in the requsted information.

arendst, 2020-06-29:
> I published the output of the console in the requsted information.

That's why I asked to do the same with weblog 4 as the 8.3.1.6 provides more information. So provide the info again.

arendst, 2020-06-29:
Oops. Ok I just saw it.

arendst, 2020-06-29:
Pls retry again.

dabos1, 2020-06-29:
> Pls retry again.

now I cannot adjust the brightness with the rotary. The mi desk lamp will toggel the led and after that the lamp will crash. Here you can see what the lamp will do: https://1drv.ms/u/s!AupNOCRYxCCFm2HxNWvc04p5ac-Q?e=bogXde

arendst, 2020-06-29:
I expected something like that. Too many rotations with too many calls to dimmer resulting in crashes.

I ordered both the Mi Desk Lamp and rotary switches to solve this issue. 

So stay tuned until I have them and can finally fix this,

dabos1, 2020-06-29:
> I expected something like that. Too many rotations with too many calls to dimmer resulting in crashes.
> 
> I ordered both the Mi Desk Lamp and rotary switches to solve this issue.
> 
> So stay tuned until I have them and can finally fix this,

okay thank you, for trying to help me

arendst, 2020-06-29:
Another possible fix. Pls retry.

dabos1, 2020-06-29:
> Another possible fix. Pls retry.

Good job it works! But I need much more rotation with the rotary to adjust the brightness than before

arendst, 2020-06-29:
Ha. I'll make final fix once I have mine.

The current fix stalls the rotary movements until the dimmer change is complete. Some optimization is possible.

closed 2020-06-30 by ascillato2

arendst, 2020-06-30:
A new fix which should solve the many rotations needed.

It works on my test rig ;-)

[image]