arendst/Tasmota #8755: Tasmota SetOption65

uviniciusc, 2020-06-22:
Hi everyone.
My question is about pre config this option before the flash flash.
Is it possible to set '1' on SetOption65 by default before the flash? 
I need to set on my_user_config.h

Thanks

**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-06-22:
See below:
```
#define APP_DISABLE_POWERCYCLE false             // [SetOption65] Disable fast power cycle detection for device reset
```

uviniciusc, 2020-06-22:
Why on console the 0 like 'Off' and 1 like 'On'. (on console after flash)

Here, https://tasmota.github.io/docs/Commands/ we have the 0 like a enable and 1 like disable.

Is it correct?

On console 0 like a Off and 1 Like On to SetOption65.
What it realy means?

Tnks for fast answer Theo.

arendst, 2020-06-22:
It's a matter of how you name it. A bridge can be open or closed. For a car it may be open while at the same time for a boat it is closed. So what's in a name.

All you need to do is check if a function is active on 0 or 1. If you find out what it is then a 0 = Off = false, a 1 = On = true.

Simple ;-)

uviniciusc, 2020-06-22:
Nice. Tnks.

closed 2020-06-22 by ascillato2

uviniciusc, 2020-06-24:
Hi Ascillato2,

I can't open a new problem. Some problem with tag but i can see only 4 option to choose on tag and when i select the tag 'Problem report' the system close my problem.

Could you help me?

uviniciusc, 2020-06-24:
The image of my closed problems
[image]

ascillato, 2020-06-24:
When opening an issue, you have a template with standard questions for debugging your problem. If you delete the template, the bot closes it. 

If you have a question about Tasmota usage, you can also address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/#/) for more information.

ascillato2, 2020-06-24:
You have opened 4 times the same issue without the template. 

Your answer to your question is on https://github.com/arendst/Tasmota/pull/8766

Just uncomment line 56 of file my_user_config.h and set there the module template you want.

uviniciusc, 2020-06-24:
Sorry about my ignorance.
realy thanks

uviniciusc, 2020-06-25:
Hi Arendst,
I tried this config on my_user_config.h 
#define APP_DISABLE_POWERCYCLE true             // [SetOption65] Disable fast power cycle detection for device reset

After flash and when i open the console the Setoption65 always keeping Off.

What is wrong?

Thank You.

uviniciusc, 2020-06-25:
Should I use #define APP_ENABLE_POWERCYCLE true ?

uviniciusc, 2020-06-26:
Ineed this result on console:
SetOption65 'On'