arendst/Tasmota #8757: Multi-level Rule Trigger Crashes Device

meingraham, 2020-06-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Tasmota reboots when a rule triggers that uses a multi-level JSON trigger.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic
- [x] Tasmota binary firmware version number used: 8.3.1.6 tasmota.bin
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA migration
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

{"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":44,"Free":467,"Rules":"ON StatusNET#IPAddress DO Var1 %value% ENDON"}
{"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
{"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"sonoff-lr_chairs","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Exception","Uptime":"0T01:11:58","StartupUTC":"2020-06-22T13:28:08","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2020-06-22T01:27:13","SaveCount":30,"SaveAddress":"F6000"}}
STATUS2 = {"StatusFWR":{"Version":"8.3.1.6(428b6dd-tasmota)","BuildDateTime":"2020-06-21T22:02:15","Boot":31,"Core":"2_7_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"340/699"}}
STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mySSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008108","2805C8000100060000005A00000000000000","000000C8","00006000"]}}
STATUS4 = {"StatusMEM":{"ProgramSize":591,"Free":412,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","043683A1","000000CD","010013C0","C000F981","00004024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
STATUS5 = {"StatusNET":{"Hostname":"myTasmota-1234","IPAddress":"192.168.1.123","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"12:34:56:78:90:AB","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
STATUS6 = {"StatusMQT":{"MqttHost":"openhabianpi","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7890AB","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
STATUS7 = {"StatusTIM":{"UTC":"2020-06-22T14:40:06","Local":"2020-06-22T10:40:06","StartDST":"2020-03-08T02:00:00","EndDST":"2020-11-01T02:00:00","Timezone":99,"Sunrise":"early","Sunset":"late"}}
STATUS10 = {"StatusSNS":{"Time":"2020-06-22T10:40:06"}}
STATUS11 = {"StatusSTS":{"Time":"2020-06-22T10:40:06","Uptime":"0T01:11:58","UptimeSec":4318,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"mySSID","BSSId":"CD:EF:12:34:56:78","Channel":6,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:09"}}}
STATUS12 = {"StatusSTK":{"Exception":9,"Reason":"Exception","EPC":["4024d265","00000000","00000000"],"EXCVADDR":"4027db1d","DEPC":"00000000","CallChain":["4027db12","4024d004","40000f49","40000f49","40000e19","40105a39","40105a3f","4010000d","402674a8","40267459","4024c5b9","40101cf5","40101cf5","40245a6e","402141fc","40216d94","401019d8","40101c77","402459c5","402459e4","402173ba","4024a410","4024ce1a","4027db11","4027db11","40101090","40102ab1","40102680","4010591b","40104af6","40104f67"]}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
```
Rule1 ON StatusNET#IPAddress DO Var1 %value% ENDON

Rule1
```
When `Status 5` is executed, 
```
CMD: Status 5
MQT: stat/myTasmota/STATUS5 = {"StatusNET":{"Hostname":"myTasmota-1234","IPAddress":"192.168.1.21","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"12:34:56:78","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
```
Tasmota reboots.

I also tried `STATUS5#StatusNET#IPAddress`.  In this case it doesn't reboot but the rule also does not trigger. So I think this trigger is not correctly specified.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Return the value of the specified trigger in `Var1` and not restart.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-06-22:
After the restart, on `Version 8.3.1.3(2b327c9-tasmota)-2_7_1` I see:
`{"RestartReason":{"Exception":0,"Reason":"Exception","EPC":["4027d52b","00000000","00000000"],"EXCVADDR":"4027d535","DEPC":"00000000","CallChain":["4027d529","4027d52a","4024cb08","40000e19","40105a35","40105a3b","4010000d","40266f8c","40266f3d","4010108c","401061d5","4000050c","40248db4","40248db4","40101cf1","40245576","402140bc","40216c60","401019d4","40101c73","402454cd","402454ec","40217282","40249f10","4024c91e","4027d529","4027d529","4010108c","40102aad","4010267c","40105917"]}}`

arendst, 2020-06-22:
Exception 0 is normally a power issue.
Exception 9 is an alignment issue either caused by tuya flash or bad compile option. There have been made some changes to the alignment recently to save flash space. Disabling this compile option might solve your exception 9 issue.

Other than that I cannot reproduce as mine just keeps on running:
```
18:03:36 CMD: rule1
18:03:36 SRC: Serial
18:03:36 CMD: Group 0, Index 1, Command "RULE", Data ""
18:03:36 MQT: stat/wemos4/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":44,"Free":467,"Rules":"ON StatusNET#IPAddress DO Var1 %value% ENDON"}
18:03:42 CMD: status 5
18:03:42 SRC: Serial
18:03:42 CMD: Group 0, Index 1, Command "STATUS", Data "5"
18:03:42 MQT: stat/wemos4/STATUS5 = {"StatusNET":{"Hostname":"wemos4","IPAddress":"192.168.2.223","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"5C:CF:7F:83:BB:10","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
18:03:42 RUL: STATUSNET#IPADDRESS performs "Var1 192.168.2.223"
18:03:42 SRC: Rule
18:03:42 CMD: Group 0, Index 1, Command "VAR", Data "192.168.2.223"
18:03:42 MQT: stat/wemos4/RESULT = {"Var1":"192.168.2.223"}
```

meingraham, 2020-06-22:
Mine is a Sonoff Basic that I flashed long ago with 6.x and migrated recently by `Reset 6`, flashing 7.2, `Reset 6`, and then 8.1 via file upload (with intermediate minimal for each), `Reset 6` again. Then updating to 8.3.1.6 via web OTA, then restoring the configuration manually.

I was using it today as a testing device. But it normally runs in my house on a lamp and has been in service for almost 2 years.

Without the rule active, I can send `Status 5` with no issue. In fact, my openHAB does this once a day to all my devices to capture network information. The crash only occurs when the rule is enabled... so when the StatusSNS triggers the rule.

@sfromis - what kind of device are you using? And how was it flashed?

sfromis, 2020-06-22:
My test device was a Sonoff S20, flashed a long time ago, and repeatedly OTA upgraded.

Just did the same test on a NodeMCU, flashed using the builtin USB/serial interface, and also upgraded a few times. Again:
`INFO3 = {"RestartReason":{"Exception":9,"Reason":"Exception","EPC":["4024cd69","00000000","00000000"],"EXCVADDR":"4027d535","DEPC":"00000000","CallChain":["4027d52a","4024cb08","40000f49","40000f49","40000e19","40105a35","40105a3b","40238d3c","4010000d","40243f18","402453cc","40246718","40266f8c","40266f3d","4024c0bd","40101cf1","40101cf1","40245576","402140bc","40216c60","401019d4","40101c73","402454cd","402454ec","40217282","40249f10","4024c91e","4010108c","401061d5","4000050c","4027d52a"]}}`

Does not seem like a power issue is among the more likely explanations.
Used a precompiled build `Version 8.3.1.3(2b327c9-tasmota)-2_7_1` Ok options?
Flash Chip Id: `0x16405E` (not sure how to verify Puya or not)

sfromis, 2020-06-22:
The logo on the flash chip of the S20 looks like "XTX", id `0x14405E`

meingraham, 2020-06-22:
Same flash chip ID on my Sonoff Basic.

arendst, 2020-06-22:
Just reproduced on a sonoff basic too ending in exception 0. Must be a power issue while scanning the rule. Nice issue!

meingraham, 2020-06-22:
Also, the crash is immediate. I turned on `WebLog 4` and then entered `Status 5`. The command is not echoed and there is no log output. I even had a second browser window open to the Console to monitor the output. Nothing until the restart messages began logging in the window,

sfromis, 2020-06-22:
Here's what I see with `SerialLog 4` on the NodeMCU:
```18:49:38 CMD: status 5
18:49:38 SRC: WebConsole from 192.168.86.39
18:49:38 CMD: Group 0, Index 1, Command "STATUS", Data "5"
18:49:38 MQT: t155nmcu/stat/STATUS5 = {"StatusNET":{"Hostname":"t155nmcu","IPAddress":"192.168.86.155","Gateway":"192.168.86.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.86.1","Mac":"84:0D:8E:A5:79:20","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:49:38 RUL: STATUSNET#IPADDRESS performs "Var1 192.168.86.155"
18:49:38 SRC: Rule
18:49:38 CMD: Group 0, Index 1, Command "VAR", Data "192.168.86.155"
18:49:38 MQT: t155nmcu/stat/VAR = {"Var1":"192.168.86.155"}

Exception (9):
epc1=0x4024cd69 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4027d535 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffedb0 end: 3fffffb0 offset: 0190
3fffef40:  00000000 4027d52a 3fffeff0 4024cb08  
3fffef50:  40000f49 40000f49 3fffdab0 40000f49  
3fffef60:  40000e19 00000002 0030209c 00000000  
3fffef70:  00000000 aa55aa55 000000ed 40105a35  
3fffef80:  40105a3b 0009359c 00000000 0c5e2446  
3fffef90:  4010000d ad745fb2 f0d12a64 b15556f7  
3fffefa0:  40266f8c 3fffef3c 40266f3d 3ffffd68  
3fffefb0:  3ffff0c0 3ffff0b0 0000000c 3ffe834c  
3fffefc0:  feefeffe 00000001 00000000 feefeffe  
3fffefd0:  00003210 feefeffe feefeffe ffffffff  
3fffefe0:  3ffff5a8 3ffe834c 00000003 4024c0bd  
3fffeff0:  3ffff0ed 00000020 00000001 ffff0208  
3ffff000:  3ffff0ec 00000002 feefeffe 00000000  
3ffff010:  3fff350c 00000020 00000000 40101cf1  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe 00000000 40101cf1  
3ffff040:  feefeffe feefeffe feefeffe 00000000  
3ffff050:  00000000 3ffff14d 3ffff5a8 40245576  
3ffff060:  feefeffe 00000000 3ffff5b4 3ffff140  
3ffff070:  3ffff0a0 3ffff148 3ffff140 402140bc  
3ffff080:  feefeffe feefeffe feefeffe 3ffff140  
3ffff090:  3ffff148 3ffff584 3ffff580 40216c60  
3ffff0a0:  00000009 3ffff14d 0000000c 3ffff610  
3ffff0b0:  000031e8 0000063d 0000063d 401019d4  
3ffff0c0:  00000000 00000009 3ffff580 00000000  
3ffff0d0:  00000000 3ffff130 00000020 40101c73  
3ffff0e0:  0000000a 3ffff584 3ffff5a8 ffffffec  
3ffff0f0:  7fffffff 3ffff590 3ffff5b4 402454cd  
3ffff100:  3ffff580 3ffff130 00000000 402454ec  
3ffff110:  3ffff580 3ffff130 00000000 40217282  
3ffff120:  40249f10 3ffff130 00000400 0000004c  
3ffff130:  3ffff120 3ffff16c 31524156 feefef00  
3ffff140:  3ffff120 3ffff15c 41544144 32393100  
3ffff150:  3836312e 2e36382e 00353531 00000000  
3ffff160:  3ffff148 00000002 3ffff14d 00000000  
3ffff170:  3ffff138 00000007 3ffff140 feefeffe  
3ffff180:  feefeffe feefeffe feefeffe feefeffe  
3ffff190:  feefeffe feefeffe feefeffe feefeffe  
3ffff1a0:  feefeffe feefeffe feefeffe feefeffe  

3ffff1b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1c0:  feefeffe feefeffe feefeffe 00000001  
3ffff1d0:  00000000 00000001 3ffff2b0 4024c91e  
3ffff1e0:  feefeffe 4027d529 feefeffe feefeffe  
3ffff1f0:  feefeffe feefeffe feefeffe 4027d529  

3ffff200:  00000000 4027d52a 3ffff2b0 4024cb08  
3ffff210:  feefeffe feefeffe feefeffe feefeffe  
3ffff220:  feefeffe 00000001 fe3020fe feefeffe  
3ffff230:  feefeffe feefeffe feefeffe feefeffe  
3ffff240:  feefeffe feefeffe feefeffe feefeffe  
3ffff250:  feefeffe feefeffe feefeffe feefeffe  
3ffff260:  feefeffe feefeffe feefeffe feefeffe  
[... 202 lines cut ...]
3ffffee0:  3fff3afc 00000000 4bc6a7f0 00000001  
3ffffef0:  00000001 3fff454c 40101506 6353f7ce  
3fffff00:  00000000 3fff5b74 3fff3abc 3fff33c4  
3fffff10:  00000001 3fff3ae0 3fff3abc 402355c7  
3fffff20:  4024a0a0 00000000 00001388 00000000  
3fffff30:  00000000 3fff5b74 3fff34d4 00000000  
3fffff40:  00000005 00000000 00000005 40235666  
3fffff50:  00000000 00000000 4bc6a7f0 4023568c  
3fffff60:  007a1200 55b85312 40101506 39db22d0  
3fffff70:  00000000 00000030 00000031 4020e875  
3fffff80:  3fffdad0 00000000 0001147e 4022faf6  
3fffff90:  3fffdad0 00000000 3fff3384 3fff33c4  
3fffffa0:  3fffdad0 00000000 3fff3384 40246b78  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
00:00:00 CFG: Loaded from flash at F4, Count 39
```

arendst, 2020-06-22:
Great fun! In addition to exception 9 I now also see Hardware Watchdogs on both sonoff and wemos. 

My my my how did we get here?

meingraham, 2020-06-22:
Interesting that the rule does what's intended (per both of your logs) and then falls off a cliff.

I wonder if the string (the 4 octet period delimited value) assigned to the Var has anything to do with the behavior? I'm just grasping at straws obviously.

arendst, 2020-06-22:
Any straw is welcome. Personally I think it has someting to do with recently introduced possibility to have a rule trigger on a status line.

Investigating...

arendst, 2020-06-22:
I think I've solved it. Give it a try!

sfromis, 2020-06-22:
My NodeMCU no longer crashed on `8.3.1.6(610f1ce-tasmota)` :+1:

meingraham, 2020-06-22:
My Sonoff Basic on `8.3.1.6(610f1ce-tasmota)` works as expected.

Thanks for the quick repair ;-)

Mike

closed 2020-06-22 by meingraham