arendst/Tasmota #8758: Building Tasmota without USE_RULES leads to compile error

eku, 2020-06-22:
### PROBLEM DESCRIPTION
I am trying to compile a firmware for sensors with the minimum required modules. This causes the software not to translate because a module calls function RulesProcessEven without enclosing the call with ```#ifdef```.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [ ] Tasmota binary firmware version number used: dev-branch HEAD
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: platformio CLI
- [ ] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Use the provides ```user_config_override.h``` and compile.
[user_config_override.zip](https://github.com/arendst/Tasmota/files/4814042/user_config_override.zip)

### EXPECTED BEHAVIOUR
Compiles fine.

### SCREENSHOTS
n/a

### ADDITIONAL CONTEXT
```
Linking .pioenvs/tasmota-sensors/firmware.elf
.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/tasmota-sensors/src/tasmota.ino.cpp.o:(.text._Z14HM10StatusInfov+0x4): undefined reference to `RulesProcessEvent(char*)'
.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/tasmota-sensors/src/tasmota.ino.cpp.o: in function `HM10StatusInfo()':
tasmota.ino.cpp:(.text._Z14HM10StatusInfov+0x38): undefined reference to `RulesProcessEvent(char*)'
collect2: error: ld returned 1 exit status

```

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-06-22:
Both xsns_61 and xsns_62 make illegal direct calls to RulesProcessEvent. I'll bind them in defines for now as a quick fix.

ascillato2, 2020-06-22:
Closing this issue as it has been fixed by Theo. Thanks for reporting.

closed 2020-06-22 by ascillato2