arendst/Tasmota #8761: Sonoff MINI - tasmota.bin too big

RomanSladecek, 2020-06-22:
Hi, I cannot flash tasmota.bin (reguler one) to Sonoff Mini switch... it's too big... maximum size for FW is 508kb I think so...
I tried to flash there tasmota.bin.gz (reduced size), but now my sonoff is bricked (only blink of led for 0.1s, and nothing else)...
I also tried flash there sonoff.bin (very old one, which has only 4xx kb), but also now my swich is bricked - same behaviour as previous one...

the only flashable is tasmota-basic.bin, but there is not domoticz configuration, and I need it :)

Any ideas or workaround?

closed 2020-06-22 by close-issue-app[bot]

javier6d2, 2020-06-24:
> Hi, I cannot flash tasmota.bin (reguler one) to Sonoff Mini switch... it's too big... maximum size for FW is 508kb I think so...
> I tried to flash there tasmota.bin.gz (reduced size), but now my [...]

If you are trying to upload using OTA. Try first to update using tasmota-minimal.bin version before upload the firmware version you want. Minimal is just an interim version used to free flash memory for the full version. You need to upgrade to the full version.

meingraham, 2020-06-24:
DISREGARD previous suggestion. Flash tasmota-lite.bin. Flashing tasmota-minimal as the first Tasmota variant loaded onto the device after the Itead/Sonoff firmware will brick your device and require that you solder contacts onto the serial interface (very difficult) in order to flash Tasmota using the "traditional" method.

javier6d2, 2020-06-24:
I assume that you have a previous tasmota firmware uploaded before made OTA. He said have already tasmota-basic installed. What´s the problem?

meingraham, 2020-06-24:
@javier6d2 your assumption is incorrect.

OP
> Hi, I cannot flash tasmota.bin (reguler one) to Sonoff Mini switch... it's too big

tasmota-lite is the variant to use.

Then, once Tasmota is on the device, use web OTA to update to the needed variant. It is the easiest means to update (Tasmota takes care of the intermediate tasmota-minimal step if required to make space in program memory for the final desired binary).

RomanSladecek, 2020-06-25:
Well. I flashed tasmota-basic over the air with diy mode. Then just changed template to

`{"NAME":"Sonoff Mini","GPIO":[17,0,0,0,9,0,0,0,21,56,0,0,255],"FLAG":0,"BASE":1}`

to make S0, S1 working for Mini

But yeah, i havent tried yet to reflash it through tasmota fw updatet to normal tasmota.bin then. I bricked 2 minis before with direct flashing it and i have no more minis now to try it :D let me know its working correctly. Anyway i have domoticz home system and i connected tasmoted minis with basic mqtt protocol. So i dont need to have newest fw with domoticz plugin now