arendst/Tasmota #8762: Ipadress rule in oled

mtasilva, 2020-06-22:
### PROBLEM DESCRIPTION
I try put ipadress in my oled line1.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____
- [x ] Tasmota binary firmware version number used: _____
  - [ x] Pre-compiled
  - [x ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: _____
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

![image](https://user-images.githubusercontent.com/39156930/85342503-ae69cf80-b4e2-11ea-891a-76f80d598fa6.png)

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

_C9AFAD/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":287,"Free":224,"Rules":"ON Wifi#Connected DO Backlog Delay 30;Status 5 ENDON ON StatusNET#IPAddress DO var1 %value% ENDON ON Power1#State=1 DO DisplayText [z0a2f0s2c1l1] %Var1% ENDON ON ENERGY#Power DO DisplayText [a4f0s2c1l3] %value%W ENDON ON ENERGY#Current DO DisplayText [a4f0s2c1l4] %value%A ENDON  Rule1 1"}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  00:00:00 CFG: Lido da flash em F7, Contagem 498
00:00:00 APP: Some settings have been reset (2)
00:00:00 I2C: SSD1306 encontrado em 0x3c
00:00:00 Projeto aplisolution Disjuntor Iluminação Versão 8.3.1.2(tasmota)-2_7_1
00:00:00 WIF: Ligando ao AP1 xxxx Channel 11 BSSId 00:06:91:75:BF:70 em modo 11N como tasmota_C9AFAD-4013...
00:00:01 WIF: Ligado
00:00:01 HTP: Servidor WEB ativo em tasmota_C9AFAD-4013 com o endereço IP 192.168.1.193
23:40:41 RUL: WIFI#CONNECTED performs "Backlog Delay 30;Status 5"
23:40:41 RSL: stat/tasmota_C9AFAD/RESULT = {"Delay":30}
23:40:42 RSL: tele/tasmota_C9AFAD/INFO1 = {"Module":"Disjuntor 1","Version":"8.3.1.2(tasmota)","FallbackTopic":"cmnd/DVES_C9AFAD_fb/","GroupTopic":"cmnd/tasmotas/"}
23:40:42 RSL: tele/tasmota_C9AFAD/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_C9AFAD-4013","IPAddress":"192.168.1.193"}
23:40:42 RSL: tele/tasmota_C9AFAD/INFO3 = {"RestartReason":"Hardware Watchdog"}
23:40:42 RSL: stat/tasmota_C9AFAD/RESULT = {"POWER":"ON"}
23:40:42 RSL: stat/tasmota_C9AFAD/POWER = ON
23:40:42 RUL: ENERGY#POWER performs "DisplayText [a4f0s2c1l3] W"
23:40:42 RSL: stat/tasmota_C9AFAD/RESULT = {"DisplayText":"[a4f0s2c1l3] W"}
23:40:42 RUL: ENERGY#CURRENT performs "DisplayText [a4f0s2c1l4] A"
23:40:42 RSL: stat/tasmota_C9AFAD/RESULT = {"DisplayText":"[a4f0s2c1l4] A"}
23:40:42 RSL: stat/tasmota_C9AFAD/STATUS5 = {"StatusNET":{"Hostname":"tasmota_C9AFAD-4013","IPAddress":"192.168.1.193","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"50:02:91:C9:AF:AD","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:40:42 RUL: STATUSNET#IPADDRESS performs "var1 192.168.1.193"
23:40:42 RSL: stat/tasmota_C9AFAD/RESULT = {"Var1":"192.168.1.193"}
23:40:43 RUL: ENERGY#POWER performs "DisplayText [a4f0s2c1l3] W"
23:40:43 RSL: stat/tasmota_C9AFAD/RESULT = {"DisplayText":"[a4f0s2c1l3] W"}
23:40:43 RUL: ENERGY#CURRENT performs "DisplayText [a4f0s2c1l4] A"
23:40:43 RSL: stat/tasmota_C9AFAD/RESULT = {"DisplayText":"[a4f0s2c1l4] A"}
23:40:43 RUL: ENERGY#POWER performs "DisplayText [a4f0s2c1l3] W"
23:40:43 RSL: stat/tasmota_C9AFAD/RESULT = {"DisplayText":"[a4f0s2c1l3] W"}
23:40:43 RUL: ENERGY#CURRENT performs "DisplayText [a4f0s2c1l4] A"
23:40:43 RSL: stat/tasmota_C9AFAD/RESULT = {"DisplayText":"[a4f0s2c1l4] A"}
23:40:44 RUL: ENERGY#POWER performs "DisplayText [a4f0s2c1l3] W"
23:40:44 RSL: stat/tasmota_C9AFAD/RESULT = {"DisplayText":"[a4f0s2c1l3] W"}
[... 67 lines cut ...]
23:40:54 RSL: stat/tasmota_C9AFAD/RESULT = {"DisplayText":"[a4f0s2c1l4] 0.000A"}
23:40:54 RUL: ENERGY#POWER performs "DisplayText [a4f0s2c1l3] 0W"
23:40:54 RSL: stat/tasmota_C9AFAD/RESULT = {"DisplayText":"[a4f0s2c1l3] 0W"}
23:40:54 RUL: ENERGY#CURRENT performs "DisplayText [a4f0s2c1l4] 0.000A"
23:40:54 RSL: stat/tasmota_C9AFAD/RESULT = {"DisplayText":"[a4f0s2c1l4] 0.000A"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Use rule and restart device

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

show ipadress in line 1

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2020-06-22:
@mtasilva 

Please explain when you expect `Power1#State=1` to trigger?

meingraham, 2020-06-22:
Also, you did not post all the information requested. Please complete.

meingraham, 2020-06-22:
Please address any further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://tasmota.github.io/docs/#/Home) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-06-22 by meingraham

mtasilva, 2020-06-23:
Power1 # state = 1 is the button to turn the oled on and off.
But the line "DisplayText [z0a4f0s2c1l4] z0 = already calls when it goes down

mtasilva, 2020-06-23:
I don't know what else I can present

meingraham, 2020-06-23:
As requested when you opened the issue. I also requested a subset of this information in our dialog on Google Groups.

`Backlog Template; Module; GPIO 255`
`Backlog Rule1; Rule2; Rule3`
`Status 0`

mtasilva, 2020-06-23:
ok the problem is resolved but have another problem:
ipaddress is large and does not fit on the screen, if you reduce the photo the text is imperceptible.
Can i get last digits, like : 192.168.1.154 -> only the last .154 , like split?

meingraham, 2020-06-23:
Not with Rules. Perhaps Tasmota scripting.

GitHub issues are for bugs and feature requests. As your's is neither, please direct further inquiries to the support forum.

mtasilva, 2020-06-23:
Thank you for the patience. You are a saint