arendst/Tasmota #8778: NFC PN532 works does not publish card id to MQTT - Iobroker

ChrisVonFrosta, 2020-06-24:
### PROBLEM DESCRIPTION
I added pn532 to ESP2866. See picture below.
I has been detected by Tasmota and read nfc tags (show the readed card id in console).
But i got no new states in iobroker.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): NodeMcu V3
- [ x] Tasmota binary firmware version number used: 8.2.0(sensors)
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ X] Provide the output of this command: `Status 0`:
```
19:54:40 CMD: Status 0
19:54:40 MQT: stat/tasmota_5FB8B8/STATUS = {"Status":{"Module":18,"FriendlyName":["SMART-Lichtsensor"],"Topic":"tasmota_5FB8B8","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:54:40 MQT: stat/tasmota_5FB8B8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:40","StartupUTC":"2020-06-24T18:53:00","Sleep":200,"CfgHolder":4617,"BootCount":264,"BCResetTime":"2020-04-27T23:34:13","SaveCount":443,"SaveAddress":"FA000"}}
19:54:40 MQT: stat/tasmota_5FB8B8/STATUS2 = {"StatusFWR":{"Version":"8.2.0(sensors)","BuildDateTime":"2020-03-20T14:48:25","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"442/699"}}
19:54:40 MQT: stat/tasmota_5FB8B8/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Phradio","WashTeam"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00040000","00000000"]}}
19:54:40 MQT: stat/tasmota_5FB8B8/STATUS4 = {"StatusMEM":{"ProgramSize":597,"Free":404,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAA387","0415A3A4","B7FFBFCD","01DA9BC4","24367CC7","00000002"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64"}}
19:54:40 MQT: stat/tasmota_5FB8B8/STATUS5 = {"StatusNET":{"Hostname":"SMART-Lichtsensor","IPAddress":"192.168.168.177","Gateway":"192.168.168.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.168.1","Mac":"EC:FA:BC:5F:B8:B8","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:54:40 MQT: stat/tasmota_5FB8B8/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.168.167","MqttPort":1883,"MqttClientMask":"SMART-Lichtsensor","MqttClient":"SMART-Lichtsensor","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:54:40 MQT: stat/tasmota_5FB8B8/STATUS7 = {"StatusTIM":{"UTC":"2020-06-24T18:54:40","Local":"2020-06-24T19:54:40","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"04:47","Sunset":"20:57"}}
19:54:40 MQT: stat/tasmota_5FB8B8/STATUS10 = {"StatusSNS":{"Time":"2020-06-24T19:54:40","ANALOG":{"Illuminance":200},"AM2301":{"Temperature":null,"Humidity":null,"DewPoint":0.0},"TempUnit":"C"}}
19:54:40 MQT: stat/tasmota_5FB8B8/STATUS11 = {"StatusSTS":{"Time":"2020-06-24T19:54:40","Uptime":"0T00:01:40","UptimeSec":100,"Heap":22,"SleepMode":"Dynamic","Sleep":200,"LoadAvg":24,"MqttCount":1,"Wifi":{"AP":2,"SSId":"WashTeam","BSSId":"44:4E:6D:14:7C:5E","Channel":5,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:10"}}}
19:54:46 MQT: tele/tasmota_5FB8B8/STATE = {"Time":"2020-06-24T19:54:46","Uptime":"0T00:01:46","UptimeSec":106,"Heap":24,"SleepMode":"Dynamic","Sleep":200,"LoadAvg":20,"MqttCount":1,"Wifi":{"AP":2,"SSId":"WashTeam","BSSId":"44:4E:6D:14:7C:5E","Channel":5,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:10"}}
19:54:46 MQT: tele/tasmota_5FB8B8/SENSOR = {"Time":"2020-06-24T19:54:46","ANALOG":{"Illuminance":203},"AM2301":{"Temperature":null,"Humidity":null,"DewPoint":0.0},"TempUnit":"C"}
19:54:56 MQT: tele/tasmota_5FB8B8/STATE = {"Time":"2020-06-24T19:54:56","Uptime":"0T00:01:56","UptimeSec":116,"Heap":24,"SleepMode":"Dynamic","Sleep":200,"LoadAvg":20,"MqttCount":1,"Wifi":{"AP":2,"SSId":"WashTeam","BSSId":"44:4E:6D:14:7C:5E","Channel":5,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:10"}}
19:54:56 MQT: tele/tasmota_5FB8B8/SENSOR = {"Time":"2020-06-24T19:54:56","ANALOG":{"Illuminance":203},"AM2301":{"Temperature":null,"Humidity":null,"DewPoint":0.0},"TempUnit":"C"}
19:55:06 MQT: tele/tasmota_5FB8B8/STATE = {"Time":"2020-06-24T19:55:06","Uptime":"0T00:02:06","UptimeSec":126,"Heap":24,"SleepMode":"Dynamic","Sleep":200,"LoadAvg":24,"MqttCount":1,"Wifi":{"AP":2,"SSId":"WashTeam","BSSId":"44:4E:6D:14:7C:5E","Channel":5,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:10"}}
19:55:06 MQT: tele/tasmota_5FB8B8/SENSOR = {"Time":"2020-06-24T19:55:06","ANALOG":{"Illuminance":196},"AM2301":{"Temperature":null,"Humidity":null,"DewPoint":0.0},"TempUnit":"C"}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
20:02:07 CMD: weblog 4
20:02:07 MQT: stat/tasmota_5FB8B8/RESULT = {"WebLog":4}
20:02:08 CFG: Saved to flash at F6, Count 447, Bytes 4096
20:02:09 MQT: tele/tasmota_5FB8B8/SENSOR = {"Time":"2020-06-24T20:02:09","PN532":{"UID":"E3629A0C"}}
20:02:13 MQT: tele/tasmota_5FB8B8/STATE = {"Time":"2020-06-24T20:02:13","Uptime":"0T00:04:26","UptimeSec":266,"Heap":24,"SleepMode":"Dynamic","Sleep":200,"LoadAvg":25,"MqttCount":1,"Wifi":{"AP":2,"SSId":"WashTeam","BSSId":"44:4E:6D:14:7C:5E","Channel":5,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:10"}}
20:02:13 MQT: tele/tasmota_5FB8B8/SENSOR = {"Time":"2020-06-24T20:02:13","ANALOG":{"Illuminance":140}}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
expact that the last red card id (and something else) would be send to mqtt server (iobroker object),

### SCREENSHOTS
[image]
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-06-24:
Hi,

Seems to be an issue on IOBroker's side.

Tasmota is sending by MQTT the ID of the RF CARD to the topic `tele/tasmota_5FB8B8/SENSOR` with payload `{"Time":"2020-06-24T20:02:09","PN532":{"UID":"E3629A0C"}}`

Please, try to make IOBROKER to listen to `tele/tasmota_5FB8B8/SENSOR`. Also you can try MQTTExplorer from your phone to check if it is really being sent.
Also, you can try for testing other MQTT broker like Mosquitto.

ChrisVonFrosta, 2020-06-24:
Thx. that was the impulse that let check me iobroker site again.
i activate "create objects automaticly".
And the object comes up.

> ...Also you can try MQTTExplorer from your phone to check if it is really being sent.

nice hint, i take in mind.

closed 2020-06-24 by ChrisVonFrosta

ascillato, 2020-06-24:
Thanks for full completing the template. That helps a lot. And also thanks for sharing your solution. Very apprecited! :+1: