arendst/Tasmota #8781: Add support for Shelly button1

matrix200, 2020-06-24:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
There is no Tasmota available for this device

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Add support for this device.
https://shop.shelly.cloud/shelly-button1-wifi-smart-home-automation#369

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
None

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-06-24:
Hi,

Please, post photos of the internal board.

If this device contains an ESP8266 chip, may be it is already supported without any modification in Tasmota.

What it is needed is just to open it and to follow the traces of the GPIOs and then flash Tasmota and test those GPIOs using the selection of Generic Module. Then that information is the template.

Remember to backup the factory firmware first.

Thanks.

matrix200, 2020-06-25:
Oh ok.
I have ordered it, will post the photos once I get the devices.
Its a bit different from other devices, as it is just a push button, it has no relays or anything of that nature.
So supposing it has an ESP8266, I would want it to recognize a button press and do some action in response.
Something like access domoticz api on a given URL.

crispy78, 2020-07-03:
If it’s the device described on https://shelly.cloud/products/shelly-button-smart-home-automation-device/ , then it’s probably an ordinary Shelly 1, Shelly 1 PM or Shelly Dimmer as the site says: “Use the Shelly Button as a safe way to use Shelly 1, Shelly 1 PM, and Shelly Dimmer out of the wall.” it’s just a plastic casing for these devices with a button.

matrix200, 2020-07-06:
Got the goods
[image]

What do you say?

matrix200, 2020-07-06:
Here is a better res photo.
What do you think about the pinout?
[image]
[image]

philippsandhaus, 2020-07-09:
I Suspect the following (starting with port closest to the USB-Port): Either GND, TX, RX, VCC (3,3V) or GND, RX, TX, VCC (3,3V)

matrix200, 2020-07-09:
Hmm
What about GPIO0?
I guess, its the next one in line?

matrix200, 2020-07-09:
So after some trial and error.
GND,GPIO0, NC, VCC,RX,TX
Tasmota flashed.
I will try to play with the Generic template and write back.

matrix200, 2020-07-09:
So now I am a bit worried how to get it to work.
Right now its in initial mode so I can see the AP of Tasmota.
If I login (but don't set it up), it turns off after a minute or so.
I suppose, if I do set it up, it will likely remain permanently on, until it completely depletes the battery, unless I figure out, how to to turn it off?
Also, how would I go about setting it in Tasmota, so it just wakes up/turns on, joins wifi, does an http call and then turns off?
Is it even possible currently in Tasmota?

matrix200, 2020-07-09:
After some deliberation I decided to take a chance and set it up to join the network.
The device works the following way (don't even need to program it).
It boots tasmota normally, joins wifi and after something like a minute or maybe less (I need to check with a stopwatch) turns off automatically.
Apparently this is not done in software, as it requires no gpio setup.
What needs to be done:
1. Try to figure out how to read battery charge level (original firmware supports that)
2. Optionally figure out, how to turn the device off using software (for battery conservation).
3. Detect the button press, and make a rule to trigger some action.

matrix200, 2020-07-09:
OK, I played some more and now I am totally confused.
The time it takes for the device to turn off appears to be rather random.
Sometimes it seems, the device won't turn off (unless you press the buttton and then it does after a few seconds).
Sometimes it seems to turn off, after having made a change in config and rebooted (again after a few seconds).
And sometimes it seems to turn off just "after a while" which is a few minutes.

matrix200, 2020-08-01:
Looks like it will be hard (or impossible) to support.
There is an extra mcu actually controlling the button/lights/power with the 8266 likely providing WIFI only.
Now the question is, how would I go about flashing back the original firmware?
I got the files from their site, but unsure which bin to flash.
Tried one which is about 600K, but there is no SSID appearing.

EdnEddy, 2020-08-08:
> There is an extra mcu actually controlling the button

Hi, yes, and the second MCU is controlling the deep sleep to enable battery supply.

> how would I go about flashing back the original firmware?

You will need a 2MB original backup or  you can try to generate one here:
https://github.com/ioprev/shelly-firmware

but didn't work for me (I need a Shelly I3 backup, if anyone can help 😃  ).

matrix200, 2020-08-08:
Thanks, I wasn't aware of that tool.
Did you try providing it the original file?
Because I was able to get the file from Shelly's ftp.

EdnEddy, 2020-08-10:
> Did you try providing it the original file?

Yes, the script takes the original Shelly .zip files 
and tries to 'glue' working esp8266 2MB firmware.bin images from it.

I could create it for you with http://api.shelly.cloud/firmware/SHBTN-1_build.zip
but I'm pretty sure it won't work.

I have started an issue: https://github.com/ioprev/shelly-firmware/issues/4

matrix200, 2020-08-10:
Hmm.
Lets wait what the response on your issue will be.

closed 2020-09-11 by stale[bot]

jeroenst, 2021-10-12:
It seems some reverse engineering is needed in here. It's a pity shelly doesn't help there power users here with documentation about the protocols between the esp8266 and their microcontrollers (same for shelly dimmer 2). So I'm thinking of buying this device and connect a serial sniffer on the serial lines. 

I have my own firmware which I could adapt (faster way for me), but I would like to use tasmota instead so it's public available.

When I figure out the serial protocol, how do I adapt the tasmota code? Is there someone who can help me with that?

barbudor, 2021-10-12:
Hi @jeroenst Sure, just open a Discussion on Github with your findings and we can support you for adding the protocol.
Some similar works has already been done with Door sensors or PIR sensors where the ESP power is under control of another MCU.
The hardest part is often to get the ESP flashed before the MCU turns it off. It may requires to remove some part or lift some pins.
A good idea is also to have 2 devices so you can keep one with the original firmware for easy reference, and one for tinkering

Zumili, 2023-10-25:
Hi guys, 2 years later, but I really want to know if someone of you already sniffed that lines and got some useful information about the data send between the two MCUs? I am facing similar problems right now. I what to use some modified Tasmota or some own Code on the "**Shelly Button 1**", so it can connect to more than just 2 predefined access points. Does Tasmota maybe already fully support this "Shelly Button 1" and I missed something?

sfromis, 2023-10-25:
If it was supported, it should have an entry in https://templates.blakadder.com/
The amount of work likely needed to get it supported may make it a "less viable" direction.

jeroenst, 2023-10-25:
Use a Tasmota zigbee bridge instead and use something like:

https://a.aliexpress.com/_EINe9rh

ramponis, 2024-05-28:
I think this device has several points in common with shelly button1

https://templates.blakadder.com/TYMC-1.html

Did Someone try to use the same approc?

sfromis, 2024-05-28:
I see few "points in common", when it is a very different device Like when the link you posted is using TuyaMCU, which the Shelly device certainly is not. Still, with the Shelly Button 1 using an ESP8266, it would be possible to flash Tasmota, and then explore it from there. Beware that Tasmota and battery powered devices is not a good match from the start....

barbudor, 2024-05-28:
Battery powered device like this one have the ESP in reset or sleep which means a ~10 sec latency from action to wifi connection and sending command.
While this latency is acceptable for a door sensor, it is not for a control button

Zigbee buttons or remote are great when you want a battery powered button

sfromis, 2024-05-28:
This specific one from Shelly can also be USB powered, but it does limit use cases. Still would take a bit of exploration to find out details of how it works, like around that extra chip on the PCB.